lvgl-micropython/lvgl_micropython #314: How to integrate USER_C_MODULES into this project

hcwang, 2025-03-19:
I would like to integrate https://github.com/cnadler86/micropython-camera-API into this project, but I don't  know how to do this. Would you please help me? Thanks.

1. how to add components into ESP-IDF? 
> Setting up the build environment (DIY method)
> 
> To build the project, follow these instructions:
> 
> [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32/get-started/index.html): I used version 5.2.3, but it might work with other versions (see notes).
> Clone the micropython repo and this repo in a folder, e.g. "MyESPCam". MicroPython version 1.24 or higher is required (at least commit 92484d8).
> You will have to add the ESP32-Camera driver (I used v2.0.15). To do this, add the following to the respective idf_component.yml file (e.g. in micropython/ports/esp32/main_esp32s3/idf_component.yml):
>   espressif/esp32-camera:
>     git: https://github.com/espressif/esp32-camera.git
> Alternatively, you can clone the https://github.com/espressif/esp32-camera repository inside the esp-idf/components folder instead of altering the idf_component.yml file.

2. How to add camera configurations to your board ?

> This project supports various boards with camera interface out of the box. You typically only need to add a single line to your board config file ("mpconfigboard.h). Example (don't forget to add the empty line at the bottom):
> 
>  `#define` MICROPY_CAMERA_MODEL_WROVER_KIT       1`
> 

3. How to build the project with USER_C_MODULES?

> To build the project, you could do it the following way:
> 
> . <path2esp-idf>/esp-idf/export.sh
> `cd MyESPCam/micropython/ports/esp32`
> `make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=<Your-Board> clean`
> `make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=<Your-Board> submodules`
> `make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=<Your-Board> all`
> 

Thanks for your help, and your project make my life easier, thanks.

kk8900, 2025-06-30:
Clone `esp32-camera` into `lvgl_micropython/lib/esp-idf/components`:

```bash
cd lvgl_micropython/lib/esp-idf/components
git clone https://github.com/cnadler86/esp32-camera.git
```

Clone `micropython-camera-API` into `lvgl_micropython/ext_mod`:

```bash
cd lvgl_micropython/ext_mod
git clone https://github.com/cnadler86/micropython-camera-API.git
```

Modify the `micropython.cmake` file in `ext_mod` to add the following line:

```bash
include(${CMAKE_CURRENT_LIST_DIR}/micropython-camera-API/src/micropython.cmake)
```

No need to modify the compilation command, for example:

```bash
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=ili9341
```

kdschlosser, 2025-07-01:
You should be able to simply clone the camera repo into any folder you like and then add `USER_C_MODULE=absolute/path/to/micropython.cmake` to your build command. The build script should do all of the legwork needed to add the module. You can use the `USER_C_MODULE` build command multiple times if you need to add more than one C extension module.