lvgl-micropython/lvgl_micropython #319: The display only works after the crash.

VsevolodVasin, 2025-04-01:
**Describe the bug**
After power supply, the screen is off

After starting one program, an artifact like an old TV appears on the screen for a split second. Then a black picture and no image

After the second start, there is a
output

1
2
3
4

```shell
0x420e9b22: mp_obj_instance_store_attr at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/objtype.c:808
0x420e9b22: mp_obj_instance_attr at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/objtype.c:817
0x40383f60: mp_execute_bytecode at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/vm.c:1042
0x42104f94: mp_hal_stdout_tx_strn at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/ports/esp32/mphalport.c:179
0x4038390a: mp_execute_bytecode at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/vm.c:285
0x420e2ff8: fun_bc_call at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/objfun.c:270
0x420ea336: mp_obj_instance_make_new at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/objtype.c:356
0x420ebf00: mp_load_method at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/runtime.c:1229
0x40383f60: mp_execute_bytecode at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/vm.c:1042
0x42117778: mp_compile_to_raw_code at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/compile.c:3658
0x42117778: mp_compile at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/compile.c:3673
0x4038390a: mp_execute_bytecode at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/vm.c:285
0x420e2ff8: fun_bc_call at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/py/objfun.c:270
0x4212baa0: parse_compile_execute at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:120
0x4212b96a: mp_reader_stdin_readbyte at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:203
0x4212bba4: mp_reader_stdin_close at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:234
0x4212ba1a: parse_compile_execute at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:78
0x4212bcd0: pyexec_raw_repl at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:510
0x4212b96a: mp_reader_stdin_readbyte at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:203
0x4212bba4: mp_reader_stdin_close at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:234
0x42104000: st_word at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/lib/oofatfs/ff.c:621
0x42104000: f_mkfs at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/lib/oofatfs/ff.c:5723
0x42104000: st_word at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/lib/oofatfs/ff.c:621
0x42104000: f_mkfs at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/lib/oofatfs/ff.c:5723
0x4210477a: vprintf_null at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/ports/esp32/main.c:101
0x42104932: mp_task at /home/vsevolodvasin/last/lvgl_micropython/lib/micropython/ports/esp32/main.c:171
```

The program apparently runs successfully and hello world already remains on the screen. What am I doing wrong? I need help

**Expected behavior**
Hello World appears on the screen the first time and does not disappear

**If your problem is running the binding please past the code you are using below.***
I need to see ALL of the setup code exactly as it is being ran.

```python
import lvgl as lv
import gc9a01
import lcd_bus
from machine import SPI, Pin
from micropython import const
import task_handler  # NOQA
import time  # NOQA
import gc

# display settings

WIDTH = const(240)
HEIGHT = const(240)

SPI_HOST = const(1)  # SPI2_HOST
SPI_SCK = const(1)   # GPIO1 (SCLK)
SPI_MOSI = const(0)  # GPIO0 (MOSI)
SPI_MISO = const(-1) # Not connected

LCD_FREQ = const(80000000)
LCD_DC = const(4)
LCD_CS = const(10)
LCD_RST = const(2)
LCD_BACKLIGHT = const(8)

LCD_PIXEL_CLOCK_HZ    = const (20 * 1000 * 1000)
LCD_BK_LIGHT_ON_LEVEL  = const (1)
LCD_BK_LIGHT_OFF_LEVEL = const (0)

BUFFER_SIZE = WIDTH * HEIGHT // 20

print(1)
# Initialize the SPI bus
spi_bus = SPI.Bus(
    host = SPI_HOST,
    mosi = SPI_MOSI,
    miso = SPI_MISO,
    sck = SPI_SCK
)
print(2)
# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus = spi_bus,
    dc = LCD_DC,
    cs = LCD_CS,
    freq=LCD_FREQ ,
    spi_mode = 0
)
print(3)
fb1 = display_bus.allocate_framebuffer(BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

gc.collect()
print(4)
display = gc9a01.GC9A01(
    data_bus=display_bus,
    display_width=WIDTH,
    display_height=HEIGHT,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    backlight_pin=LCD_BACKLIGHT,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

print(5)
display.set_power(True)

display.init()
display.set_backlight(100)
lv.init()

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)  # 0x000000 , 0x0000FF

label = lv.label(scrn)
label.set_text("HELLO WORLD!!")
label.set_style_text_color(lv.color_hex(0x0000FF), 0)
label.align(lv.ALIGN.CENTER, 0, 0)

print("Objects created")
```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make: 
```shell
 python3 make.py esp32 BOARD=ESP32_GENERIC_C3 DISPLAY=gc9a01
```
* Model: ESP32_GENERIC_C3

VsevolodVasin, 2025-04-02:
> are you using the Thonny IDE?
> 

Yes, but if you need to use another ide I can try

kdschlosser, 2025-04-02:
Thonny is what the problem is. It does a soft reset after flashing the board and after uploading a file. You need to perform a hard reset for it to work properly. It has to do with how the hardware is handled. A soft reset only resets the MicroPython stack and not the underlying hardware.

VsevolodVasin, 2025-04-02:
> Thonny is what the problem is. It does a soft reset after flashing the board and after uploading a file. You need to perform a hard reset for it to work properly. It has to do with how the hardware is handled. A soft reset only resets the MicroPython stack and not the underlying hardware.
> 

My case is described after a hard reset. The problem is that after the screen starts there is no picture on the screen, it appears after the crash (last frame)

kdschlosser, 2025-04-02:
A hard reset is pressing the reset button or unplugging the USB and plugging it back in. It crashes when you do this?