lvgl-micropython/lvgl_micropython #320: My ESP32S3+ili9488 doesn't work and there is no error message

celiu-2000, 2025-04-01:
This is my compilation instruction, it runs normally without any error：
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9488 INDEV=ft6x36
`
Here is my code:
It runs fine, but there is no display on the screen and no error message

```
import lcd_bus
from micropython import const
from machine import SPI

# display settings
_WIDTH = const(320)
_HEIGHT = const(480)
_BL = const(45)

_RST = const(3)
_DC = const(8)
_MOSI = const(11)
_MISO = const(13)
_SCK = const(12)

_HOST = const(2)  # SPI2
_LCD_CS = const(10)
_LCD_FREQ = const(4000000)

_TOUCH_CS = const(18)
_TOUCH_FREQ = const(10000000)

spi_bus = SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

import ili9488  # NOQA
import lvgl as lv  # NOQA
import fs_driver

display = ili9488.ILI9488(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB888,
    color_byte_order=ili9488.BYTE_ORDER_RGB,
)

display.set_power(True)
display.init()
display.set_backlight(0)

fs_drv = lv.fs_drv_t()
fs_driver.fs_register(fs_drv, 'S')
scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

# slider = lv.slider(scrn)
# slider.set_size(300, 50)
# slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.set_style_text_color(lv.color_make(0x00, 0x00, 0x00), 0)
label.align(lv.ALIGN.CENTER, 0, -50)

def click(evt):
    print('click')

btn = lv.button(scrn)
btn.set_pos(200, 200)
btn.set_size(100, 100)
btn.add_event_cb(click, lv.EVENT.ALL, None)
```

This is the ESP32S3 development board I use:

[image]

kdschlosser, 2025-04-01:
change the following.

```python
_HOST = const(1)  # SPI2
```

```python
display.set_backlight(100)
```

kdschlosser, 2025-04-01:
You also don't have a touch driver installed so you will need to do that as well.

celiu-2000, 2025-04-03:
This is the modified code, but the screen still shows no response. No matter what parameters I modify, it cannot be driven properly：

```
import lcd_bus
from micropython import const
from machine import SPI

# display settings
_WIDTH = const(320)
_HEIGHT = const(480)
_BL = const(45)

_RST = const(3)
_DC = const(8)
_MOSI = const(11)
_MISO = const(13)
_SCK = const(12)

_HOST = const(1)  # SPI2
_LCD_CS = const(10)
_LCD_FREQ = const(20000000)

_TOUCH_CS = const(18)
_TOUCH_FREQ = const(10000000)

spi_bus = SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

import ili9488  # NOQA
import lvgl as lv  # NOQA
import fs_driver

display = ili9488.ILI9488(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
    color_space=lv.COLOR_FORMAT.RGB888,
    color_byte_order=ili9488.BYTE_ORDER_RGB,
)

display.set_power(True)
display.init()
display.set_backlight(100)

fs_drv = lv.fs_drv_t()
fs_driver.fs_register(fs_drv, 'S')
scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

# slider = lv.slider(scrn)
# slider.set_size(300, 50)
# slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.set_style_text_color(lv.color_make(0x00, 0x00, 0x00), 0)
label.align(lv.ALIGN.CENTER, 0, -50)

def click(evt):
    print('click')

btn = lv.button(scrn)
btn.set_pos(200, 200)
btn.set_size(100, 100)
btn.add_event_cb(click, lv.EVENT.ALL, None) # 设置按钮被按下后的回调函数

```

But I used firmware compiled by others, and this screen can be driven normally, so I can ensure that it is not a wiring issue. Although the firmware compiled by others can display normally, there are some other problems that have forced me to try compiling it myself. Here is the screen driver code for this firmware:

```
import lvgl as lv
import time
from espidf import VSPI_HOST
from ili9XXX import ili9488
disp = ili9488(miso=13, mosi=11, clk=12, cs=10, dc=8, rst=3, spihost=VSPI_HOST, mhz=20, power=-1, backlight=-1, factor=16, hybrid=True, width=480, height=320, invert=False, double_buffer=True, half_duplex=False,rot=-2)
```
What is the name of this firmware：lv_micropython_esps3n16r8_color32.bin