lvgl-micropython/lvgl_micropython #323: Two Issues with ESP32-S3-BOX-3: GT911 Touch Offset and SD Card Driver Problems

kk8900, 2025-04-09:
Hello,
I'm encountering two issues while using the ESP32-S3-BOX-3 development board from Espressif. I hope someone can help me out.
The development board documentation can be found here: [ESP-BOX AIoT Development Framework](https://github.com/espressif/esp-box)
Issue 1: GT911 Touch Offset
I used the following code to test the display and touch functions. The display works fine, but there is a touch offset. I referred to [GT911 Initialization failed and it cannot be used with board ](https://github.com/lvgl-micropython/lvgl_micropython/issues/116), but the problem still persists. The touch only works correctly when I touch about 5-10 mm below the slider.
`import screen
import lvgl as lv

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)
slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO LVGL_MICROPYTHON!')
label.align(lv.ALIGN.CENTER, 0, -50)`

screen.py:
`import lcd_bus
from micropython import const
import machine
import lcd_utils

spi_bus = machine.SPI.Bus(
    host=1,
    mosi=6,
    miso=-1,
    sck=7,    
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=40000000,
    dc=4,
    cs=5
)

import ili9341
import lvgl as lv

ili9341.ILI9341._ORIENTATION_TABLE = (
    0x0,
    0x60,
    0x40 | 0x80,
    0x20 | 0x40 | 0x80
)

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=320,
    display_height=240,
    backlight_pin=47,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import i2c
import task_handler
import gt911, pointer_framework

display.init(2)

i2c_bus = i2c.I2C.Bus(host=0, scl=18, sda=8, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=0x5D, reg_bits=gt911.BITS)

indev = gt911.GT911(touch_dev,startup_rotation=pointer_framework.lv.DISPLAY_ROTATION._180)

display.set_backlight(1)

if not indev.is_calibrated:
    indev.calibrate()
indev._cal.mirrorX = False
indev._cal.mirrorY = True
indev._cal.save()

indev.enable_input_priority()

display.set_rotation(lv.DISPLAY_ROTATION._180)

th = task_handler.TaskHandler()
`

Issue 2: SD Card Driver
I tried to use both SPI mode and SDMMC mode to drive the SD card, but neither worked. I also referred to [Mounting SD CARD with ELECROW 5" Display](https://github.com/lvgl-micropython/lvgl_micropython/issues/147), but still couldn't resolve the issue.
SPI MODE
`import os
import machine
import vfs
import time

spi_bus = machine.SPI.Bus(
    host=2,        # SPI2 (HSPI)
    mosi=14,       # SD_CMD (DI) → IO14
    miso=9,        # SD_DAT0 (DO) → IO9
    sck=11,        # SD_CLK → IO11
)

try:
    sd = machine.SDCard(
        spi_bus=spi_bus,
        cs=12,      # SD_DAT3 (CS) → IO12
        freq=20000000  
    )
    
    vfs.mount(sd, "/sd")
    print("SPI MODE）")
    print(os.listdir("/"))
    print(os.listdir("/sd"))

except Exception as e:
    print(e)
`
 SDMMC MODE
`import os
import machine
import vfs
import time

try:
    sdcard = machine.SDCard(
        slot=1,
        clk=11,    
        cmd=14,  
        d0=9,    
        d1=13,  
        d2=42, 
        d3=12, 
        cd=-1, 
        wp=-1, 
        width=4,
        freq=400000
    )
    
    vfs.mount(sdcard, "/sd")
    print("SDMMC MODE")
    print("os.listdir("/"))
    print("os.listdir("/sd"))

except Exception as e:
    print(e)

`

I would really appreciate any help on these issues. Thank you!

kdschlosser, 2025-04-09:
OK so with the SD card your host is wrong for the comment you have. You need to set that to `1`. 

* SPI Host 1 = 0, 
* SPI Host 2 = 1 
* SPI Host 3 = 2

This is how it is done in the ESP32 SDK so I kept it that way.

The other thing is there is a speed limit for SPI SSD of 10mHz. You have it set to 20mHz

The touch is easy to fix. You are using the GT911 touch IC and that IC you have the ability to set the touch resolution. It doesn't come defaulted to what your display is so we need to change that...

add this code after  you create the touch driver and before you do any display rotation.  I also suggest having this code before the touch calibration and you are going to need to redo the calibration once this code is added. 

```python
indev.hw_size != (320, 240):
    fw = indev.firmware_config
    fw.width = 320
    fw.height = 240
    fw.save()
    del fw

```

kdschlosser, 2025-04-09:
Here is a link to all of the things you can adjust using that firmware config for the GT911 touch IC...

https://github.com/lvgl-micropython/lvgl_micropython/blob/main/api_drivers/common_api_drivers/indev/gt911_extension.py

Just remember to call `save` after you make changes to things.

kdschlosser, 2025-04-09:
your SPI for the SDCard might have some pins that are flip flopped... Manufacturers are a pain with labeling inconsistencies between devices. You may want to flip flop these pins as a test. 

```
    mosi=14,  # SD_CMD (DI) → IO14
    miso=9,  # SD_DAT0 (DO) → IO9
```

MOSI - Master Out Slave In
MISO = Master In Slave Out 

DI = Data In
DO = Data Out
a lot of times SPI devices have their pins labeled as to what it needs to connect to in the MCU
MI = DI
MO = DO 

I have seen this many times and it sure does make things confusing. The way SPI pins are named and how manufacturers tend to label things is what makes it confusing.

kk8900, 2025-04-20:
> OK so with the SD card your host is wrong for the comment you have. You need to set that to `1`.
> 
> * SPI Host 1 = 0,
> * SPI Host 2 = 1
> * SPI Host 3 = 2
> 
> This is how it is done in the ESP32 SDK so I kept it that way.
> 
> The other thing is there is a speed limit for SPI SSD of 10mHz. You have it set to 20mHz
> 
> The touch is easy to fix. You are using the GT911 touch IC and that IC you have the ability to set the touch resolution. It doesn't come defaulted to what your display is so we need to change that...
> 
> add this code after you create the touch driver and before you do any display rotation. I also suggest having this code before the touch calibration and you are going to need to redo the calibration once this code is added.
> 
> if indev.hw_size != (320, 240):
>     fw = indev.firmware_config
>     fw.width = 320
>     fw.height = 240
>     fw.save()
>     del fw

Thank you very much for your reply. I have been on a business trip recently and haven’t had the time to do the testing. I apologize for that.

I tried the code you provided, but the touch functionality still has significant issues. It seems that the touch is only working in a small area in the middle of the lower half of the screen, and other parts are not responsive. I’m not sure if this is a hardware issue. I’ll try flashing back to the official firmware and test it again.

The SD card has the same problem—it doesn’t get recognized by the driver. I also tested it with the official MicroPython firmware, but it still didn’t work. It might be a hardware design issue, which I can’t seem to resolve.

Thank you once again for your reply. Is there a community for lvgl-micropython where I can learn more?

kdschlosser, 2025-04-20:
You are using what is called the CYD (Cheap Yellow Display) board. I believe I should have a config file set up for that display. I will check and see if I do.

kk8900, 2025-05-01:
> You are using what is called the CYD (Cheap Yellow Display) board. I believe I should have a config file set up for that display. I will check and see if I do.

This is the ESP32-S3-BOX03 development board from Espressif. It has quite a few users, so it should not be a CYD. I tested it today with the official firmware, and both the touch screen and the SD card are working fine. However, when I use LVGL-Micropython, the touch only works at a small area at the center bottom of the screen, as shown in the image below:

[image]

This is really troubling, but I still hope to get your help. 
Thank you very much.

kk8900, 2025-05-02:
Thank you very much for your reply. After modifying according to your suggestions, I can now drive the display and touch screen normally. Here is the complete code, which might be helpful to others.
```python
import lcd_bus
from micropython import const
import machine
import lcd_utils

spi_bus = machine.SPI.Bus(
    host=1,
    mosi=6,
    miso=-1,
    sck=7,    
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=40000000,
    dc=4,
    cs=5
)

import ili9341  # NOQA
import lvgl as lv  # NOQA

ili9341.ILI9341._ORIENTATION_TABLE = (
    0x0,
    0x60,
    0x40 | 0x80,
    0x20 | 0x40 | 0x80
)

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=320,
    display_height=240,
    backlight_pin=47,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import i2c
import task_handler
import gt911, pointer_framework

display.init(2)

i2c_bus = i2c.I2C.Bus(host=0, scl=18, sda=8, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=0x5D, reg_bits=gt911.BITS)

indev = gt911.GT911(touch_dev,startup_rotation=pointer_framework.lv.DISPLAY_ROTATION._180)#
if indev.hw_size != (320, 240):
    fw = indev.firmware_config
    fw.width = 320
    fw.height = 240
    fw.save()
    del fw

display.set_backlight(1)

indev.enable_input_priority()

display.set_rotation(lv.DISPLAY_ROTATION._180)

th = task_handler.TaskHandler()

closed 2025-05-02 by kk8900

kdschlosser, 2025-05-02:
So everything is working correctly now?

There are things that have to be done in a very specific order for some displays and thee is not much I am able to do to make that process easier without adding a large amount of additional boiler plate code. It's not something that is easily explained either because it doesn't apply to all display boards...

Glad to see that you got it up and running like it should be.

kk8900, 2025-05-04:
> So everything is working correctly now?
>
> There are things that have to be done in a very specific order for some displays and thee is not much I am able to do to make that process easier without [...]

The display and touch functions are working normally now, but the SD card still doesn't work. I can't get it to function even with the standard MicroPython firmware. I'm not sure what the problem is. What's worse, my SD card seems to be damaged beyond repair, so I haven't been able to test it further.

kdschlosser, 2025-05-04:
is your SD Card reader an SPI reader?

Do you have a datasheet for the display board you are using?

kk8900, 2025-05-07:
> is your SD Card reader an SPI reader?
> 
> Do you have a datasheet for the display board you are using?

I’m really sorry, but I’m having trouble accessing GitHub at the moment and I can’t upload images either. Here’s the link to the development board’s schematic. Could you please check it out for me?
[ESP32-S3-BOX-3](https://github.com/espressif/esp-box/blob/master/hardware/SCH_ESP32-S3-BOX-3_V1.0/SCH_ESP32-S3-BOX-3-SENSOR-01_V1.1_20230922.pdf)

kdschlosser, 2025-05-07:
OK I had to add onto the SDCard driver because the way it was written didn't allow for setting the pins for an MMC interface. Your SDCard reader is set up for a 4 lane MMC interface. I added the needed things to get it working for you. You are going to have to clone the repo again and build the firmware so you get the updated driver.

Here is a code example of how to use it. Let me know if there are any issues with it either when compiling or when trying to use the driver.

Make sure to read the notes in the code example as there are some things you need to follow.

```python
import machine
from micropython import const
import os

_MMC_DATA0 = const(9)
_MMC_DATA1 = const(13)
_MMC_DATA2 = const(42)
_MMC_DATA3 = const(12)
_MMC_CLK = const(11)
_MMC_CMD = const(14)
_MMC_POWER = const(43)
_MMC_WIDTH = const(4)
_MMC_SLOT = const(1)

# turn on the power to the SDCard reader
sd_card_power = machine.Pin(_MMC_POWER, machine.Pin.OUT)
sd_card_power.value(1)

# initilize the SDCard driver
# make sure to have a card inserted before trying to start the driver. 
# If there is no card inserted it is not going to work. The CD (Card Detect) 
# pin is not connected so a card MUST be installed before the driver 
# is initialized.
sd_card = machine.SDCard(
    slot=_MMC_SLOT,
    width=_MMC_WIDTH,
    cmd=_MMC_CMD,
    clk=_MMC_CLK,
    data_pins=(
        _MMC_DATA0,
        _MMC_DATA1,
        _MMC_DATA2,
        _MMC_DATA3
    )
)

# the driver should automatically mount the SD Card to the "/sdcard" folder. 
# I think that is the folder name. You may have to double check on that one in
# the MicroPython documentation.
print(os.listdir('/sdcard'))
```

Vincent1-python, 2025-05-08:
> OK I had to add onto the SDCard driver because the way it was written didn't allow for setting the pins for an MMC interface. Your SDCard reader is set up for a 4 lane MMC interface. I added the [...]

I re-cloned and recompiled the firmware, and ran it according to your code, but it still doesn’t work. I also tried two different SD cards, but the error persists.
Traceback (most recent call last):
  File "<stdin>", line 41, in <module>
OSError: [Errno 2] ENOENT
I modified the code as follows, but it still doesn’t work:
```
import machine
from micropython import const
import os
import vfs

_MMC_DATA0 = const(9)
_MMC_DATA1 = const(13)
_MMC_DATA2 = const(42)
_MMC_DATA3 = const(12)
_MMC_CLK = const(11)
_MMC_CMD = const(14)
_MMC_POWER = const(43)
_MMC_WIDTH = const(4)
_MMC_SLOT = const(1)

sd_card_power = machine.Pin(_MMC_POWER, machine.Pin.OUT)
sd_card_power.value(1)

sd_card = machine.SDCard(
    slot=_MMC_SLOT,
    width=_MMC_WIDTH,
    cmd=_MMC_CMD,
    clk=_MMC_CLK,
    data_pins=(
        _MMC_DATA0,
        _MMC_DATA1,
        _MMC_DATA2,
        _MMC_DATA3
    )
)

vfs.mount(sd_card, '/sd')
os.listdir('/sd')

print(os.listdir('/sd'))'''
```

kdschlosser, 2025-05-08:
OK so a little bit of education on MMC VS SDCard. Both look the same but they are not. MMC cards will work in an MMC Card Reader and also in an SD Card reader but an SD Card will not work in an MMC Card Reader.  You have an MMC Card reader so you need to specifically buy an MMC Card not an SD Card.

kk8900, 2025-05-09:
> OK so a little bit of education on MMC VS SDCard. Both look the same but they are not. MMC cards will work in an MMC Card Reader and also in an SD Card reader but an SD Card will not work in an MMC Card Reader. You have an MMC Card reader so you need to specifically buy an MMC Card not an SD Card.

It seems I still have a lot to learn.

I checked the dimensions and found that an MMC card is 32mm × 24mm, while a MicroSD card is 15mm × 11mm. The card slot on this development board can only accommodate a MicroSD card.

Moreover, the official documentation only mentions the name "MicroSD Card" and does not specify that it is an MMC card:

MicroSD Card SPI mode:
1. DAT3 works as CS (chip select)
2. CMD works as DI (Data In)
3. CLK works as Clock
4. DAT0 works as DO (Data out)

kdschlosser, 2025-05-09:
But every single one of the drivers that is available for that board the ones that are written by espressif are done using MMC not SD.

kk8900, 2025-05-12:
> But every single one of the drivers that is available for that board the ones that are written by espressif are done using MMC not SD.

Thank you for your answer.