lvgl-micropython/lvgl_micropython #324: st7735 driver problem

zhou19830318, 2025-04-09:
import lvgl as lv
import st7735
import lcd_bus
from machine import SPI, Pin

# Initialize LVGL
lv.init()

Pin(36).on() # Enable LED

# Initialize the SPI bus
spi_bus = SPI.Bus(
    host=1,
    mosi=4,
    miso=-1,
    sck=5
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=3,
    freq=2000000
)

# Initialize the st7735 display driver
display = st7735.ST7735(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    reset_pin=2,
    reset_state=st7735.STATE_HIGH,
    power_on_state=st7735.STATE_HIGH,
    backlight_pin=1,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

# Initialize display
display.init()
display.set_backlight(100)

# Create a screen and load it
scr = lv.obj()
lv.screen_load(scr)

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello, LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import utime as time
time_passed = 1000

while True:
    start_time = time.ticks_ms()
    time.sleep_ms(1)  # sleep for 1 ms
    lv.tick_inc(time_passed)
    lv.task_handler()
    end_time = time.ticks_ms()
    time_passed = time.ticks_diff(end_time, start_time)

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 37, in <module>
  File "display_driver_framework.py", line 187, in __init__
MemoryError: memory allocation failed, allocating 1009068456 bytes

kdschlosser, 2025-04-09:
what are you using for a build command?

zhou19830318, 2025-04-09:
I used this command to compile the firmware "python3 make.py esp32 BOARD=ESP32_GENERIC_S3 --flash-size=4M DISPLAY=st7735 INDEV=ft6x36", Actually, I don't have touchscreen capability.

closed 2025-04-20 by zhou19830318