lvgl-micropython/lvgl_micropython #325: XPT2046, x-axis inverted after calibration

Kalrkloss, 2025-04-10:
**Describe the bug**
Calibration runs with screen orientation 0°, but afterwards the x-axis of the touch screen is inverted, meaning when touching an object on the right side of the screen, the left side is instead selected, and vice versa. The y-axis is fine. Can I configure this somehow?

**Expected behavior**
Both axis being configured right.

```python

```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make: ESP32
* Model: S3 WROOM N16R8

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: octal
* FLASH:  octal

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: Windows 10 WSL v2 (doesn't matter, the same happens when compiling with Linux (Debian))

**Build Command**

```shell
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9341 INDEV=xpt2046
```

kdschlosser, 2025-04-10:
yes as a matter of fact you can...

```python
import touch_cal_data

touch_cal = TouchCalData('My Display')

indev = xpt2046.XPT2046(touch_dev, touch_cal=touch_cal)

if not indev.is_calibrated:
    indev.calibrate()
    touch_cal.mirrorX = not touch_cal.mirrorX
    touch_cal.save()
```

only perform the calibration with the rotation set to zero. If you set it to any other rotation the numbers are going to get all mixed up resulting in the coordinates mapping to off display areas.

Kalrkloss, 2025-04-11:
Many thanks!

Can I also do this in the board config?

kdschlosser, 2025-04-11:
???

Kalrkloss, 2025-04-11:
I mean, are there some settings in the board config, that would define this mirroring automatically at build time?

Kalrkloss, 2025-04-11:
Oh, and I forgot: How do I rotate the screen properly after the calibration?
Whenever I rotate with display.set_rotation(lv.DISPLAY_ROTATION._270) or another angle, the touch is all over the place.
Do I also have to rotate the touch data?

Kalrkloss, 2025-04-11:
Ok, found it myself. I actually have to adapt the mirroring to the display angle, for lv.DISPLAY_ROTATION._90 this would be 
touch_cal.mirrorY = not touch_cal.mirrorY, while X stays untouched.

fabse-hack, 2025-06-06:
@Kalrkloss can you provide me an example code, how you handle it?

i have the same setup! display and board :D 
ili9341 with xpt2046 and esp32 s3