lvgl-micropython/lvgl_micropython #326: Guru Meditation Error after soft reset

Kalrkloss, 2025-04-11:
**Describe the bug**
After doing a soft reset with CTRL-D or machine.soft_reset(), I get the following error:
``A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MicroPython 
IDF version : c9763f62
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x421b8d17  PS      : 0x00060a30  A0      : 0x8200304a  A1      : 0x3fcbe150  
A2      : 0x02000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x820b1800  A9      : 0x3fcbe150  
A10     : 0x02000001  A11     : 0x00000002  A12     : 0x000001f3  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000002  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0200000d  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x421b8d14:0x3fcbe150 0x42003047:0x3fcbe170 0x42002d9d:0x3fcbe1b0 0x420d1427:0x3fcbe1f0

ELF file SHA256: 1bf88956d

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40383266
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1108
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbdc
load:0x403cc700,len:0x2f4c
entry 0x403c989c
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-04-10; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
``

But: boot.py and main.py are executed without a problem.
When running the current code file from Thonny with "Run current script", I get the error:

``%Run -c $EDITOR_CONTENT

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MicroPython 
IDF version : c9763f62
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060a30  A0      : 0x8200304a  A1      : 0x3fcbe150  
A2      : 0x3c21662c  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x821b8d1c  A9      : 0x3fcbe140  
A10     : 0x3c21662c  A11     : 0x00000002  A12     : 0x000001f3  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000002  SAR     : 0x00000002  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  

Backtrace: 0xfffffffd:0x3fcbe150 0x42003047:0x3fcbe170 0x42002d9d:0x3fcbe1b0 0x420d1427:0x3fcbe1f0

ELF file SHA256: 1bf88956d

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40383266
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1108
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbdc
load:0x403cc700,len:0x2f4c
entry 0x403c989c
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-04-10; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MicroPython 
IDF version : c9763f62
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

Core  1 register dump:
PC      : 0x420409aa  PS      : 0x00060a30  A0      : 0x82038099  A1      : 0x3fcbcdb0  
A2      : 0x3fcbce50  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3fcbce7c  
A6      : 0x000000ef  A7      : 0x3fcbcdb0  A8      : 0x40bcfd94  A9      : 0x000010a3  
A10     : 0x000010a3  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3c2541a8  
A14     : 0x000000ef  A15     : 0x3fcbcd80  SAR     : 0x00000008  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40bcfd94  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x420409a7:0x3fcbcdb0 0x42038096:0x3fcbce30 0x42048ea9:0x3fcbced0 0x420458d1:0x3fcbcfe0 0x42045754:0x3fcbd020 0x42045891:0x3fcbd050 0x4203399e:0x3fcbd090 0x42033794:0x3fcbd100 0x42033721:0x3fcbd130 0x4203717a:0x3fcbd170 0x420248c1:0x3fcbd240 0x4202507b:0x3fcbd350 0x42026c02:0x3fcbd410 0x42027119:0x3fcbd450 0x42026b98:0x3fcbd4a0 0x4202ffcf:0x3fcbd4f0 0x4203159d:0x3fcbd670 0x4203132a:0x3fcbd770 0x420311f2:0x3fcbd7c0 0x42030f89:0x3fcbd810 0x42030c87:0x3fcbd880 0x4203069b:0x3fcbd8b0 0x42079431:0x3fcbd8f0 0x42078fb1:0x3fcbd930 0x42007571:0x3fcbd970 0x420167c5:0x3fcbd990 0x420076ad:0x3fcbd9d0 0x420bdabd:0x3fcbda10 0x420bdb85:0x3fcbda30 0x4037a4d2:0x3fcbda50 0x420b512b:0x3fcbdaf0 0x420bdabd:0x3fcbdb20 0x420ecdc7:0x3fcbdb40 0x420ece11:0x3fcbdb80 0x420bdabd:0x3fcbdba0 0x420bdb45:0x3fcbdbc0 0x420be704:0x3fcbdbf0 0x420be94d:0x3fcbdc70 0x4037a8c7:0x3fcbdc90 0x420b512b:0x3fcbdd30 0x420bdabd:0x3fcbdda0 0x420bdb85:0x3fcbddc0 0x4037a4d2:0x3fcbdde0 0x420b512b:0x3fcbde80 0x420bdabd:0x3fcbdeb0 0x420bdb85:0x3fcbded0 0x4037a4d2:0x3fcbdef0 0x420b512b:0x3fcbdf90 0x420bdabd:0x3fcbdfc0 0x420bdb85:0x3fcbdfe0 0x4037a4d2:0x3fcbe000 0x420b512b:0x3fcbe0a0 0x420bdabd:0x3fcbe0d0 0x420bdad2:0x3fcbe0f0 0x420f3457:0x3fcbe110 0x420f359a:0x3fcbe1a0 0x420d147e:0x3fcbe1f0

ELF file SHA256: 1bf88956d

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40383266
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1108
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbdc
load:0x403cc700,len:0x2f4c
entry 0x403c989c
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-04-10; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
``

**Expected behavior**
Soft reset without error.

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make: ESP32-S3
* Model: ESP32-S3-WROOM N16R8

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM:  octal
* FLASH:   octal

 - OS: [e.g. iOS]
Windows 10, WSL v2
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9341 INDEV=xpt2046

kdschlosser, 2025-04-11:
you cannot do a soft reset. Soft reset only resets the software and not the hardware so what happens is when the drivers start back up for the display and attempt to reattach to the GPIO pins you get the core panic.

Kalrkloss, 2025-04-13:
What's the official way to restart after changing the code then?
Always a hardware reset?

kdschlosser, 2025-04-13:
yes

Kalrkloss, 2025-04-16:
Any idea how to hack Thonny to do a machine.reset() after saving a file, instead of a soft reset?

kdschlosser, 2025-04-16:
search the discussions in this repo. I modified the files needed in Thonny and posted them in a discussion with instructions on how to apply them,. I didn't test the code so it may need to be modified some to get it to work properly. but you are more then welcome to give it a test and see what happens

Carglglz, 2025-06-03:
> you cannot do a soft reset. Soft reset only resets the software and not the hardware so what happens is when the drivers start back up for the display and attempt to reattach to the GPIO pins you get the core panic.

The core panic happens because these LVGL root pointers are not handled properly on soft-reset

https://github.com/lvgl-micropython/lvgl_micropython/blob/f7fbf64b19683e988bd66e5e395393cd8b39267e/gen/lvgl_api_gen_mpy.py#L1505

See
https://github.com/lvgl/lv_binding_micropython/commit/f76c25fbc5a19071ebc693c3d03b8aa04a5296bd

https://github.com/lvgl/lvgl/pull/7973 

https://github.com/lvgl/lv_binding_micropython/pull/383

which fixes this and also makes `lv.init()`/`lv.deinit()` unnecessary 👍🏼

kdschlosser, 2025-06-04:
Then should should do the same thing without needing to much about with the code generator or using any none standard functionality in MicroPython to fix the issue...

https://github.com/lvgl-micropython/lvgl_micropython/commit/e3443318a73bc7fe59a1458f8c90766e84b8c29d