lvgl-micropython/lvgl_micropython #331: is it possible to update LVGL?

NickyDark1, 2025-04-26:
hello,

before building the .bin is it possible to update the lib?
https://github.com/lvgl-micropython/lvgl_micropython/tree/main/lib
how?

thank you so much

kdschlosser, 2025-04-26:
update it to what? It's already at the latest release of LVGL.

NickyDark1, 2025-04-28:
Yes, thank you very much. I would like to know how to update it if it is not updated in the future.

kdschlosser, 2025-04-28:
with each new release of LVGL it breaks something that I have to go and fix. so it's not exactly all that easy sometimes to change to a newer version of LVGL.

NickyDark1, 2025-04-28:
Thank you so much for the prompt response. I'm here testing and I'm delighted with this Git. I have a lot of questions to ask :)

thank you so much.

closed 2025-04-28 by NickyDark1