lvgl-micropython/lvgl_micropython #334: I don't know if I'm wrong, but the touch is not working for me with the 'cst816s.py' library.

NickyDark1, 2025-04-28:
# I am using this device
```
https://www.waveshare.com/esp32-s3-touch-lcd-2.htm
```

# device:
<p align="center"> 
ESP32-S3 2inch Capacitive Touch Display Development Board, 240×320
</p>
<p align="center">
    [image]
</p>

looking at the library at this url:
```
https://github.com/lvgl-micropython/lvgl_micropython/blob/main/api_drivers/common_api_drivers/indev/cst816s.py
```

# My code table touch, no work driver cst816s 
This code displays the table in three columns and its rows, but when you click on it to scroll, it doesn't do anything. I'm new, but I made another implementation where it does work with touch. What am I doing wrong with this code?
```py
import lvgl as lv
import time
from machine import Pin, I2C
import lcd_bus
import st7789
import task_handler
import machine

# Importar el controlador CST816S compartido
from cst816s import CST816S # internal library

# Configuración de pines para la pantalla
_WIDTH = 240
_HEIGHT = 320
_BL = 1       # Pin de retroiluminación
_RST = -1     # Pin de reset, -1 si no se usa
_DC = 42      # Pin DC
_MOSI = 38    # Pin MOSI
_MISO = 40    # Pin MISO
_SCK = 39     # Pin SCK
_HOST = 2     # SPI2
_LCD_CS = 45  # Pin CS para LCD
_LCD_FREQ = 40000000  # Frecuencia SPI para LCD

# Configuración de pines para el touch (I2C)
_TP_SDA = 48  # Pin SDA para touch
_TP_SCL = 47  # Pin SCL para touch

# Configurar bus SPI
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

# Configurar bus LCD
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

# Inicializar LVGL
lv.init()

# Crear el display
display = st7789.ST7789(
[... 662 lines cut ...]
                print("Fin de scroll sin inercia")
    
    except Exception as e:
        print(f"Error en manejo táctil: {e}")

# Inicialización
setup_touch_handling()

# Bucle principal
print("Iniciando bucle principal...")
while True:
    try:
        # Manejo de eventos táctiles para scroll
        handle_touch_event()
        
        # Pausa para no saturar el sistema
        time.sleep_ms(30)
    except Exception as e:
        print(f"Error en el bucle: {e}")
        time.sleep_ms(1000)
```

I'd like to know what I'm doing wrong with the 'cst816s.py' library that Git has internally. How can I correct my error without having to resort to external libraries for touch?

Thank you very much.

kdschlosser, 2025-04-28:
I don't think the driver that is in this library has been tested yet. Which means it might have some hiccups that need to be sorted out. I am going to use the driver you posted as a reference to see what I might have mucked up.

NickyDark1, 2025-04-28:
Thank you very much, I will recompile micropython.

NickyDark1, 2025-04-28:
I've already recompiled lvgl-micropythonwith native cst816s and it still doesn't work. Could it be a problem with my code?

# My code:
```py
import lvgl as lv
import time
from machine import Pin, I2C
import lcd_bus
import st7789
import task_handler
import machine

# Importar el controlador CST816S compartido
from cst816s import CST816S # internal library

# Configuración de pines para la pantalla
_WIDTH = 240
_HEIGHT = 320
_BL = 1       # Pin de retroiluminación
_RST = -1     # Pin de reset, -1 si no se usa
_DC = 42      # Pin DC
_MOSI = 38    # Pin MOSI
_MISO = 40    # Pin MISO
_SCK = 39     # Pin SCK
_HOST = 2     # SPI2
_LCD_CS = 45  # Pin CS para LCD
_LCD_FREQ = 40000000  # Frecuencia SPI para LCD

# Configuración de pines para el touch (I2C)
_TP_SDA = 48  # Pin SDA para touch
_TP_SCL = 47  # Pin SCL para touch

# Configurar bus SPI
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

# Configurar bus LCD
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

# Inicializar LVGL
lv.init()

# Crear el display
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

# Encender y configurar el display
display.set_power(True)
display.init()
display.set_backlight(80)  # Brillo al 80%

# Configurar el I2C para el touch
print("Inicializando I2C para el touch...")
i2c = I2C(1, scl=Pin(_TP_SCL), sda=Pin(_TP_SDA), freq=100000)
[... 149 lines cut ...]
# Bucle principal
while True:
    try:
        if touch:
            # Verificar si hay un toque
            coords = touch._get_coords()
            
            if coords:
                state, x, y = coords
                stats["touch_events"] += 1
                
                # Mostrar coordenadas cada 10 eventos para no saturar
                if stats["touch_events"] % 10 == 0:
                    print(f"Toque detectado en ({x}, {y})")
                    
                # Actualizar etiqueta de debug cada 50 eventos
                if stats["touch_events"] % 50 == 0:
                    debug_label.set_text(f"Toques: {stats['touch_events']}")
        
        # Pausa para no saturar el sistema
        time.sleep_ms(30)
    except Exception as e:
        print(f"Error en el bucle: {e}")
        time.sleep_ms(1000)
```

NickyDark1, 2025-04-28:
# url:
modify the library
```
https://github.com/lvgl-micropython/lvgl_micropython/blob/main/api_drivers/common_api_drivers/indev/cst816s.py
```

# cst816s_adapted.py
This code works on the device
```py

# Copyright (c) 2024 - 2025 Kevin G. Schlosser
# Adapted version for better I2C compatibility

from micropython import const
import pointer_framework
import time
import machine

I2C_ADDR = const(0x15)
BITS = const(8)

# 0x00: No gesture
# 0x01: Swipe up
# 0x02: Swipe down
# 0x03: Swipe left
# 0x04: Swipe right
# 0x05: Single click
# 0x0B: Double click
# 0x0C: Long press
_GestureID = const(0x01)

# 0: No finger
# 1: 1 finger
_FingerNum = const(0x02)

# & 0xF << 8
_XposH = const(0x03)
_XposL = const(0x04)

#  & 0xF << 8
_YposH = const(0x05)
_YposL = const(0x06)

_RegisterVersion = const(0x15)

_BPC0H = const(0xB0)
_BPC0L = const(0xB1)

_BPC1H = const(0xB2)
_BPC1L = const(0xB3)

_ChipID = const(0xA7)
#_ChipIDValue = const(0xB5)
_ChipIDValue = const(0xB6)

_ProjID = const(0xA8)
_FwVersion = const(0xA9)

# ===============================
_MotionMask = const(0xEC)

# Enables continuous left and right sliding
_EnConLR = const(0x04)
# Enables continuous up and down sliding
_EnConUD = const(0x02)
# Enable double-click action
_EnDClick = const(0x01)
# ===============================

# Interrupt low pulse output width.
[... 503 lines cut ...]
# Bucle principal
while True:
    try:
        if touch:
            # Verificar si hay un toque
            coords = touch._get_coords()
            
            if coords:
                state, x, y = coords
                stats["touch_events"] += 1
                
                # Mostrar coordenadas cada 10 eventos para no saturar
                if stats["touch_events"] % 10 == 0:
                    print(f"Toque detectado en ({x}, {y})")
                    
                # Actualizar etiqueta de debug cada 50 eventos
                if stats["touch_events"] % 50 == 0:
                    debug_label.set_text(f"Toques: {stats['touch_events']}")
        
        # Pausa para no saturar el sistema
        time.sleep_ms(30)
    except Exception as e:
        print(f"Error en el bucle: {e}")
        time.sleep_ms(1000)
```

kdschlosser, 2025-12-27:
You are going to run into problems with your code because you are performing a rotation prior to calibrating the display. The README.md file in the root of this repository explicitly states to not perform a rotation prior to doing the calibration. 

It is also a known thing where Thonny doesn't play nice with the code in this repository. So I recommend using something else like VSCode or PyCharm.

You also need to read up on the use of `const` in the MicroPython documentation because you are using it incorrectly.

closed 2025-12-29 by kdschlosser