lvgl-micropython/lvgl_micropython #336: file pointer_framework.pyi, important touch or?

NickyDark1, 2025-04-29:
I notice that this file is important to use touch but I only find the URL pointer_framework.pyi
```
https://github.com/lvgl-micropython/lvgl_micropython/blob/main/pointer_framework.pyi
```

 where would the file pointer_framework.c be located? or what would be the file that does the management?

```py
import pointer_framework
```

What would be the exact location? I'd love to be able to observe it for some implementations. 

Thank you very much.

kdschlosser, 2025-04-29:
all of the drivers that are written in Python are in the folder api_drivers. "indev" means INput DEVices.

kdschlosser, 2025-04-29:
the pyi files are stub files and those work with an IDE like VSCode, Visual Studio or PyCharm  and they allow features in the IDE like code completion, intellisense and type checking to work properly.

NickyDark1, 2025-04-30:
Thanks. I see that pointer_framework is important, but I don't know which file or files will be used to obtain that code and transfer it to .c (or if it's already there). I want to make some internal implementations to speed up the touch a little bit. But I think that's the driver to transfer touch data to LVGL, or am I mistaken?

kdschlosser, 2025-04-30:
changing it to C is not going to make it any faster.

If you want it to be faster then you need to use the following lines of code in Python

```
indev.enable_input_priority()
```

That will cause the program to run in a loop while there is user input. everything else except interrupts stop being processed and all processing is of the touch input and refreshing the display to populate anything that has changed as a result of that input.

It's the only way to get the display to update when input is actually happening. Otherwise it only gets updated once every 33 to 66 milliseconds.

You don't want to change the refresh timeout which is what the polling for input is tied to because if that is too fast you end up with the CPU being tied up doing a whole lot of nothing except checking to see if updates have happened. It's pointless to have the loop running that fast. The loop by default will check for input every 33 milliseconds and with the above code attached to it then all continuous touch will be an instant read and update of the display. It makes a noticeable impact on how fast the UI responds to user input.

I had also though about doing a ramping timer that would speed up the polling loop if there was input found every single time the input was read and then it would slow it back down to the default in 1 millisecond increments every time the loop didn't read any input. This would improve the response times for consecutive press and release type of input. People have said that this works really well so I left it how it was.

kdschlosser, 2025-04-30:
You also have to remember there is a typical response rate of about 20 milliseconds for the touch panel itself. it can be longer than that if the touch panel goes to sleep between touches because it takes it a bit to wake up and register that a touch has happened.

kdschlosser, 2025-04-30:
If your touch screen is working OK without doing a calibration then don't do one. doing a calibration is going to add some time because of the computations that are done. It is on my TODO list to move those calculations into C code to improve the speeds. I have not gotten to it yet.

NickyDark1, 2025-04-30:
Thank you very much for the prompt response.
I haven't even calibrated the screen yet because I saw that it works normally and I really liked that you left it alone. As for writing in .c, I still need to integrate it since I see that the library is important. I saw that LVGL is not used internally but in calls in MicroPython for the touch and the display. I ask you, isn't it better to integrate everything for the display and touch in '.c'? So that it is a little faster, even if it is a few nanoseconds, haha.

How could I convert the LCD and touch drivers that are in Micropython code to .c since I see that if I make changes I would damage everything.

NickyDark1, 2025-04-30:
I saw a problem with the touch screen. It would be when I rotate the screen 90 or 270 degrees, the touch screen inverts. If I click on one part, the touch screen would click on another. Clicking where I pressed doesn't do anything, only the edge.

For example, 
I have this widget:
https://docs.lvgl.io/master/details/widgets/tabview.html

[image]
1- I press the edge of the screen.
2- The touch sensor says I'm pressing the tab (without pressing it).

and if I click on the tab, it would click on the edge of the screen.

It only happens when I rotate 90 and 270 degrees

Why does this behavior occur? What am I doing wrong when rotating the screen? How can I fix it?

Thank you very much.