lvgl-micropython/lvgl_micropython #339: On rgb_display every new line is shifted 1 pixel

getznerh, 2025-05-02:
Your change in file rgb_bus_rotation.c with commit ID 662a3bc results in each line being shifted by 1 pixel when outputting text and images.

getznerh, 2025-05-02:
i removed this y += 1, but it does not help. I have to search again

getznerh, 2025-05-02:
I removed the changes from  662a3bc and e29be7a and now everthing works as expected.
I hav a 7" ESP32S3 with ch422g expander and g911 touch. The display uses rgb_display.
The build command is:
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4 FROZEN_MANIFEST=~/frozenmodules-gz/manifest.py DISPLAY=rgb_display EXPANDER=ch422g INDEV=gt911 INDEV=xpt2046 --enable-cdc-repl=n --enable-jtag-repl=n --enable-cdc-repl=y

but as mentioned before, i have also changed the ch422g.py for the device registration in the framework.

getznerh, 2025-05-02:
I reactivated the change from 662a3bc and it is still working OK. So the change from e29be7a is the reason for the shifting 1 pixel

NickyDark1, 2025-05-02:
Hi, 
I'd like to ask you a few questions. I've rotated the screen 90 and 270 degrees, and do you have any touch apps? I'm having some issues when I rotate the screen those degrees and the touch is inverted.