lvgl-micropython/lvgl_micropython #342: axs15231b driver issue - JC3248W535 board - object has no attribute '_color_space'

alvins82, 2025-05-07:
Trying to use JC3248W535 ESP32S3 board. When I 'import axs15231b' throws an error.

Note - I have tried straga driver (https://github.com/straga/micropython_lcd) - works fine.

>>> import lv_config
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lv_config.py", line 81, in <module>
  File "axs15231b.py", line 55, in __init__
AttributeError: 'AXS15231B' object has no attribute '_color_space'

kdschlosser, 2025-05-07:
I just pushed a fix for the problem. give it a go and see if it work correctly.

kdschlosser, 2025-05-07:
OK so it is displaying which is a good thing. I will look at your code to see if there is anything that jumps out at me.

kdschlosser, 2025-05-07:
Try running this code.

I am only adding a single control just to keep it simple. I swapped the width and height around as I do not believe that this display supports hardware rotation and a lot of display manufacturers advertise the panel resolution incorrectly. Lets see if that is what the issue is.

If that doesn't correct it then I will dig into the driver to see what is going on.

kdschlosser, 2025-05-07:
skip my last comments.

I am working on the driver now. It is going to take me a bit to finish it up...

alvins82, 2025-05-29:
Update on this - just tried this in the latest 1.25.0 build (saw you had done some updates. I'm not getting any tearing but also not seeing anything on the screen.

zyhsck, 2025-08-17:
I am also on the same model of development board, but I compiled and burned according to the burning command output to me, and the result was a black screen, could you please tell me about the process?

kdschlosser, 2025-08-17:
I believe I just found the issue. Took me a while to track down the problem and it was something that was really dumb and simple to fix. That's usually how it goes.