lvgl-micropython/lvgl_micropython #343: [1310/1974] Building C object esp-idf/main_esp32/CMakeFiles/idf_main_esp32.dir//machine_sdcard.c.obj FAILED

Digital-Apprentice, 2025-05-09:
Hi, there's an error during compiling firmware for
Cgeap Yellow Display ESP32-2432S028R - ESP32 WROOM with additional 8MB SPIRAM memory, display ILI9341 or newer version ILI9XXX and indev XPT2046)
Build command:
python3 make.py esp32 BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM clean 
or 
python3 make.py esp32 clean on a fresh repository install

Error:
[1310/1974] Building C object esp-idf/main_esp32/CMakeFiles/idf_main_esp32.dir//machine_sdcard.c.obj
FAILED: esp-idf/main_esp32/CMakeFiles/idf_main_esp32.dir//machine_sdcard.c.obj
/home/exit/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -DBERKELEY_DB_CONFIG_FILE="/home/exit/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h" -DESP_PLATFORM -DFFCONF_H="/home/exit/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h" -DIDF_VER="c9763f62" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY=“(1)” -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/exit/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/config -I/home/exit/lvgl_micropython/lib/micropython -I/home/exit/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/exit/lvgl_micropython/ext_mod/spi3wire -I/home/exit/lvgl_micropython/ext_mod/spi3wire/include -I/home/exit/lvgl_micropython/ext_mod/lcd_bus -I/home/exit/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/exit/lvgl_micropython/lib -I/home/exit/lvgl_micropython/lib/lvgl -I/home/exit/lvgl_micropython/lib/micropython/ports/esp32 -I/home/exit/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC -I/home/exit/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/exit/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/home/exit/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/home/exit/lvgl_micropython/lib/esp-idf/components/freertos/config/xtensa/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/exit/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32/. -I/home/exit/lvgl_micropython/lib/esp-idf/components/heap/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/log/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/soc/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/soc/esp32 -I/home/exit/lvgl_micropython/lib/esp-idf/components/soc/esp32/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/hal/esp32/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/hal/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_rom/include/esp32 -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32 -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_common/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_system/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_system/port/soc -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/private -I/home/exit/lvgl_micropython/lib/esp-idf/components/xtensa/esp32/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/xtensa/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/xtensa/deprecated_include -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/include/apps -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/include/apps/sntp -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/lwip/src/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/port/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/port/freertos/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/exit/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/exit/lvgl_micropython/lib/esp-idf/components/app_update/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bootloader_support/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_app_format/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_bootloader_format/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/esp_partition/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/include/esp32/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/common/osi/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/common/api/include/api -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/common/btc/profile/esp/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/common/hci_log/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hr/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/htp/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/prox/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/cts/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hid/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/sps/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/include -I/home/exit/lvgl_micropython/lib/esp-idf/components/bt/hos [... cut ...]
ninja failed with exit code 1, output of the command is in the /home/exit/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_std Ierr_output_11366 and /home/exit/lvgl_micropyth on/lib/micropython/port s/esp32/build-ESP32_GENERIC/log/idf_py_ stdout_output_11366
-e See [Build Troubleshootin g · micropython/micropython Wiki · GitHub](https://github.com/micropy 
thon/micropython/wiki/Build-Troubleshooting)
make: Leaving directory ‘/home/exit/lvgl_micropython/lib/micropython/ports/esp32’
make: *** [Makefile:71: all] Error 1 I

bh1rio, 2025-05-17:
same error with

python3 make.py esp32 BOARD=ESP32_GENERIC DISPLAY=st7796 INDEV=gt911

kwinter745321, 2025-05-17:
Same error with
python3 make.py esp32  BOARD=ESP32_GENERIC   DISPLAY=ili9341 INDEV=xpt2046
or
python3 make.py esp32  BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM  DISPLAY=ili9341 INDEV=xpt2046

The issue begins with file:     machine_sdcard.c