lvgl-micropython/lvgl_micropython #345: Possible problem with keypad driver

asherevan, 2025-05-13:
**Describe the bug**
Custom drivers (like the T-Deck) do not function correctly. They function correctly in themselves, but they do not make any change to the GUI

**Expected behavior**
(For example) whenever I press a key on the keyboard it types text in text areas.

**Images**

If it is a visual issue then take a picture of it or a video and add it here.

**If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)**

```shell
PASTE OUTPUT HERE
```

**If your problem is running the binding please past the code you are using below.***
I am using the drivers you wrote for the T-Deck.

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make:ESP32 
* Model:S3

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: quad OR octal
* FLASH:  quad OR octal

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: Ubuntu

**Build Command**

```shell
BUILD COMMAND HERE
```

ThomasFarstrike, 2025-06-08:
In created https://github.com/lvgl-micropython/lvgl_micropython/issues/378 for this.

asherevan, 2025-06-13:
@kdschlosser After you pushed that update with the keypad_framework it now returns this error:
```
File "keypad_framework.py", line 14, in __init__
NameError: name 'indev_drv' isn't defined
```

kdschlosser, 2025-06-14:
@asherevan 

OK that is fixed.

asherevan, 2025-06-14:
@kdschlosser Now with a fresh clone it returns this error when compiling:
```
/home/ubuntu/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c:191:50: error: expected ';' before 'self'
191 |         self->panel_io_config.dma_burst_size = 64
|                                                  ^
|                                                  ;
192 |         self->panel_io_config.flags.refresh_on_demand = 0;
|         ~~~~
At top level:
cc1: note: unrecognized command-line option '-Wno-unterminated-string-initialization' may have been intended to silence earlier diagnostics
```
With this build command:
```
python make.py esp32 DISPLAY=st7789 INDEV=gt911 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16  --optimize-size --ota --partition-size=3000000
```

kdschlosser, 2025-06-15:
sorry about that. all fixed

asherevan, 2025-06-15:
@kdschlosser  Thank you. I built it again but whenever I press a key, it still doesn't register it.

asherevan, 2025-07-14:
I really need to figure out how to make this work. I am working on a file editor program so the keyboard is really important. Do you have any ideas?

kdschlosser, 2025-07-14:
This should be all fixed. Have you tried to clone a fresh copy of the repo and build form that...

I need an exact error in order to fix it.

kdschlosser, 2025-07-15:
May 26th is when I updated the MicroPython version and June 10th is when I pushed a commit that fixes the keypad driver. So if you have not updated the repo since I did the MicroPython update I suggest that you do so because you more then likely don't have the fix.

kdschlosser, 2025-07-15:
Do a clone again just to make sure that you have the fix. Don't update your existing clone, do a new clone.

asherevan, 2025-07-16:
Fresh clone. Still doesn’t work. I know that the keyboard reading itself is
working though, it is just not inputting anything in textareas still.

kdschlosser, 2025-07-16:
I do not believe so. I will compile it tonight and see what is going on with it. It's going to be something really dumb that I have done that I need to correct.

col135, 2025-07-17:
The issue could be related to having to manage 'groups' for your input driver when NOT using a pointer type indev driver. After some mucking around I got my simple debounced button navigation (UP, DOWN and ENTER buttons) going Ok with the following code. I'm yet to return to this code to explore it further, it might be related to your lack of input issue as well (ie having to direct non pointer input to 'groups')?

I used 'lv_demo_keypad_encoder.c' as a reference to get this working correctly

from encoder_framework_keys import *
g = lv.group_create()
g.set_default()
indev = EncoderDriver()
indev.set_group(g)
indev.enable(1)               # has to be done here for some reason even though it seems to be done in the driver?

asherevan, 2025-07-29:
That was the problem! Thanks!

closed 2025-07-29 by asherevan

kdschlosser, 2025-07-29:
I can do that. the user is able to collect that default group from the driver using the get_group method if they wanted to adjust or alter it in some way.