lvgl-micropython/lvgl_micropython #349: What is preferred approach for freezing fonts?

dcmcshan, 2025-05-14:
I was previously dynamically loading fonts using binfont_create and a path to the .font.  Now I am trying to freeze the whole system, and I need to find another way since I cannot seems to freeze the filesystem path.  Am currently going to try and convert them to python modules, but perhaps you have a better approach?

Thanks!

dcmcshan, 2025-05-14:
converting them to python modules seems to work well enough...

kdschlosser, 2025-05-14:
you would need to use CPython to open the font file in binary mode and read the file. Then use binscii.hexlify to convert the byte data into a hexadecimal string representation of the byte data. I recommend using binascii.hexlify to do this because each byte of data is represented by 2 hexadecimal characters ie: `0xAB 0xCD 0x11 0x1 0x9` becomes `'ABCD110109'`. Baking the font into the firmware is going to be expensive in both storage and also in load time. You are going to end up using at least 2 times the amount of flash space baking it into the firmware. This is because you now need to use 2 bytes to represent 1 actual byte of the font data. If you are willing to go that route here is an example of what you would do..

create a python module called `my_font.py`. in that file add the code below. name the file of your font "my_font.bin" and put that file into the same folder as this script. Run the python script from CPython.  If you have more than one font then create a script for each font and run it. It will build a manifest file that contain all of the fonts. You only need to add the additional build parameter a single time. This will bake all of the fonts into the firmware for you. Using this method is nice because it will resize the partition to fit with the fonts added. 

```
# I recommend removing all of the comments from this file.
def load():
    import lvgl as lv
    import sys

    font_data = ''
    font_data = bytearray([int(font_data[i: i + 2], 16) for i in range(0, len(font_data), 2)])
    font_data_mv = memoryview(font_data)
    font = lv.binfont_create_from_buffer(font_data_mv, len(font_data))
    
    # This should return the created font and then unload the module partially.
    # It is going to be your responsibility to delete the imported module from
    # the module you imported it from.

    # Example:
    #
    #     import my_font
    #
    #     my_font_buf, my_font_mv, my_loaded_font = my_font.load()
    #
    #     # This next line is what unloads the module which is going to
    #     # free up resources.
    #     del my_font

    try:
        # we return the bytearray and the memoryview because the font was
        # created passing a memoryview object which is a pointer to the
        # location in RAM where the dat is located. If we pass the bytearray
        # directly to the function to create the font then a copy of the data
        # gets made and it will consume twice as much RAM until this function
        # exits and the original bytearray gets garbage collected. This is not
        # ideal if the data is large so we return both the memory view and the
        # bytearray so a reference is held which will keep the boject from
        # getting garbage collected.
        return font_data, font_data_mv, font
    finally:
        del sys.modules[__name__]

if __name__ == '__main__':
    # this code is only going to run when you run the script using CPython.
    # then when you bake it into the firmware and import it it will run what
    # is needed to run
    import binascii
    import os

    # I am making some assumptions. The first one is you are going to
    # have the same filename for this script that you have for the font file.
    # IDK what the font file is going to have for an extension. I am using "bin"
    # so you may need to change that.
    # This script when run directly using CPython is going to read the font
    # file (so long as it has the same name as this script) and it will then
    # edit this script and insert the data formatted in a manner that will
    # hopefully keep the amount of flash used to store this script to as little
    # as possible.
    path = filename = os.path.split(__file__)[-1]
    name = os.path.splitext(filename)[0]
    with open(f'{name}.bin', 'rb') as f:
        data = f.read()

    data = binascii.hexlify(data).decode('utf-8')

    with open(__file__, 'r') as f:
        src_file = f.read().splitlines()

    # place the font data into this file
    for i, line in enumerate(src_file):
        if line.strip().startswith('font_data'):
            src_file[i] = f"    font_data = '{data}'"
            break

    src_file = '\n'.join(src_file)

    with open(__file__, 'w') as f:
        f.write(src_file)
    
    # create the manifest file
    manifest = os.path.join(path, 'manifest.py')
    
    entry = f"freeze('{path}', '{filename}')"
    
    if os.path.exists(manifest):
        with open(manifest, 'r') as f:
            manifest_data = f.read()
    else:
        manifest_data = ''
    
    if entry not in manifest_data:
        manifest_data += f'\n{entry}\n'
        
        with open(manifest, 'w') as f:
            f.write(manifest_data)
            
    print('Add the line below to your build command.')
    print(f"FROZEN_MANIFEST={manifest}")

```

** DISCLAIMER: The code above is pseudo code. It has not been tested for functionality. There could be errors or things that will need to be corrected. I do recommend removing all of the comments if you are going to use the script. You can also remove all of the build code by creating a function in a separate module to handle the creating of the script to minimize the amount of repeat code between multiple font modules.

dcmcshan, 2025-05-15:
Hm, that kinda sucks.  do you think we should have a non-user filesystem or something for this sort of thing?  might also be useful for image files, etc.?

dcmcshan, 2025-05-15:
Do you think lvgl_micropython could support some sort of font manifest for additional fonts? Which takes the .ttf, makes .font and edits lv_conf.h appropriately?

/*Optionally declare custom fonts here.
 *You can use these fonts as default font too and they will be available globally.
 *E.g. #define LV_FONT_CUSTOM_DECLARE   LV_FONT_DECLARE(my_font_1) LV_FONT_DECLARE(my_font_2)*/
#define LV_FONT_CUSTOM_DECLARE

kdschlosser, 2025-05-15:
There is no benefit to baking the files into the firmware. It's not faster at all. The only reason why there is a performance gain for baking python modules in is because they get compiled into bytecode so that step doesn't have to be done at runtime. Any other file there really is nothing to be gained from it.

dcmcshan, 2025-05-15:
So I want to be able to OTA the firmware, and it needs fonts.  What do you recommend?