lvgl-micropython/lvgl_micropython #353: driver display spd2010 screen does not turn on

NickyDark1, 2025-05-20:
# Build
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.46B
```
make clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT

# Compilar con make
make esp32 \
    BOARD=ESP32_GENERIC_S3 \
    BOARD_VARIANT=SPIRAM_OCT \
    MICROPY_PY_THREAD=1 \
    FLASH_SIZE=16MB \
    ESP32_SPIRAM=y \
    ESP32_SPIRAM_MODE=1 \
    OPTIMIZATION=size \
    MICROPY_USB_CDC=1 \
    MICROPY_PY_THREAD_GIL=2 \
    DISPLAY=spd2010 \
```

# My code micropython 
```py
# waveshare_esp32s3_146.py
# Configuration for Waveshare ESP32-S3-Touch-LCD-1.46 with SPD2010 controller
'''
https://github.com/lvgl-micropython/lvgl_micropython/blob/main/api_drivers/common_api_drivers/display/spd2010/spd2010.py
https://github.com/lvgl-micropython/lvgl_micropython/blob/main/lcd_bus.pyi

'''
import lvgl as lv
import time
import json
from machine import Pin, SPI
import lcd_bus
from spd2010 import SPD2010  # Import the SPD2010 display driver
import display_driver_framework  # Import to access constants like BYTE_ORDER_RGB
import task_handler
import machine
import gc

# Display dimensions for 1.46" display (240x240 is common for this size)
_WIDTH = 240   
_HEIGHT = 240

# LCD pins from schematic
_BL = 1        # Backlight pin (IO1)
_RST = 0       # Reset pin (IO0)
_DC = 42       # Data/Command pin (using IO42/SDA2)
_LCD_CS = 21   # Chip select (IO21)

# SPI pins from schematic
_HOST = 2      # SPI host
_SCK = 40      # Clock pin (LCD_SCK -> IO40)
_MOSI = 46     # Data line 0 (LCD_SDA0 -> IO46)
_MISO = 45     # Data line 1 (LCD_SDA1 -> IO45)

# SPI frequency for the display
_LCD_FREQ = 40000000  # 40MHz

# Añadir después de importar las librerías, antes de configurar el bus SPI
# Secuencia de reset manual
if _RST != -1:
    reset_pin = Pin(_RST, Pin.OUT)
    reset_pin.value(0)  # Reset activo (LOW)
    time.sleep_ms(20)   # Mantener reset por 20ms
    reset_pin.value(1)  # Liberar reset
    time.sleep_ms(200)  # Esperar a que el dispositivo se inicialice

# Encender backlight manualmente después de inicializar el display
if _BL != -1:
    bl_pin = Pin(_BL, Pin.OUT)
    time.sleep_ms(100)  # Esperar antes de encender el backlight
    bl_pin.value(1)     # Encender backlight (asumiendo lógica HIGH activa)

# Configure SPI bus for the ESP32-S3
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

# Create LCD bus with the SPI bus
# Note: Setting quad=True for quad SPI mode according to the lcd_bus.pyi definition
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
    quad=True  # Enable quad SPI mode
)

# Initialize LVGL
lv.init()

# Create the display with SPD2010 driver
print("Initializing SPD2010 display...")
display = SPD2010(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RST,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,  # Using RGB565 color format
    color_byte_order=display_driver_framework.BYTE_ORDER_RGB,  # Use the constant from display_driver_framework
    #color_byte_order=display_driver_framework.BYTE_ORDER_RGB,  # Use the constant from display_driver_framework
    #color_byte_order=display_driver_framework.BYTE_ORDER_RGB,  # Use the constant from display_driver_framework
)

# Turn on and configure the display
display.set_power(True)
display.init()
display.set_backlight(100)  # 80% brightness

'''
# Añadir después de importar las librerías, antes de configurar el bus SPI
# Secuencia de reset manual
if _RST != -1:
    reset_pin = Pin(_RST, Pin.OUT)
    reset_pin.value(0)  # Reset activo (LOW)
    time.sleep_ms(20)   # Mantener reset por 20ms
    reset_pin.value(1)  # Liberar reset
    time.sleep_ms(200)  # Esperar a que el dispositivo se inicialice

# Encender backlight manualmente después de inicializar el display
if _BL != -1:
    bl_pin = Pin(_BL, Pin.OUT)
    time.sleep_ms(100)  # Esperar antes de encender el backlight
    bl_pin.value(1)     # Encender backlight (asumiendo lógica HIGH activa)
'''

# Initialize task handler for LVGL
th = task_handler.TaskHandler()
if hasattr(th, 'enable'):
    th.enable()

print("Display initialization complete")

```

# No error.
[image]

```
>>> help("modules")
__main__          button_framework  io_expander_framework               socket
_asyncio          cmath             json              spd2010
_boot             collections       keypad_framework  spi3wire
_espnow           cryptolib         lcd_bus           ssl
_indev_base       deflate           lcd_utils         struct
_onewire          dht               lvgl              sys
_spd2010_init     display_driver_framework            machine           task_handler
_thread           ds18x20           math              time
_webrepl          encoder_framework micropython       tls
aioespnow         errno             mip/__init__      touch_cal_data
apa106            esp               neopixel          touch_calibrate
array             esp32             network           uasyncio
asyncio/__init__  espnow            ntptime           uctypes
asyncio/core      flashbdev         onewire           umqtt/robust
asyncio/event     framebuf          os                umqtt/simple
asyncio/funcs     fs_driver         platform          upysh
asyncio/lock      gc                pointer_framework urequests
asyncio/stream    hashlib           random            vfs
binascii          heapq             re                webrepl
bluetooth         i2c               requests/__init__ webrepl_setup
btree             inisetup          rgb_display_framework               websocket
builtins          io                select
Plus any modules on the filesystem
```

Apparently the code worked without errors but the screen is not actually turning on. What is going wrong?

kdschlosser, 2025-05-20:
I would need to see the data sheet for the display from the display manufacturer. If they have any code examples those would also be helpful (programming language doesn't matter)

NickyDark1, 2025-05-22:
Thank you very much, 

here I have a basic repository with just one button using the LCD and touch with LVGL.

https://github.com/NickyDark1/lvgl_arduino_only_btn_msg

I also attach a wiki:
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.46B

diagrams:
https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.46/ESP32-S3-Touch-LCD-1.46.pdf

kdschlosser, 2025-05-23:
I wanted to let you know that I am working on this issue. I have added the touch driver for your board and I am in the process of writing the fusion math to handle mems sensors. I am writing that code in C because there is a very large amount of floating point math involved and when that is done in Python code it really can slow things down. What the fusion math does is it takes the magnetometer, gyroscope and accelerometer readings for the x, y and z axis and crunches that raw data into something a little more friendly to work with which is roll, yaw and pitch. That's for a 9DOF mems IMU. I will also handle a 6DOF IMU which doesn't have a magnetometer. with the 6DOF you will get roll and pitch but not yaw.

When I am done adding this you are going to be able to physically hold the display and rotate it and the UI will rotate with it. I am going to make it so the user is going to be able to have an unlocked rotation where the UI will rotate to any angle and locked where it will rotate to 0, 90, 180 and 270 only.

NickyDark1, 2025-05-23:
Thank you very much.
If you need anything for me to check on the device I'm here.

kdschlosser, 2025-05-24:
I am getting close to finishing. I am going to need to make sure it compiles and fix any errors then you will be able to test. Possibly later tonight or tomorrow.

NickyDark1, 2025-05-26:
Thanks, let me know and I'll confirm. If you need anything, let me know.