lvgl-micropython/lvgl_micropython #354: ESP32C3,ST7735 can`t be compiled.

zundamon114514noda, 2025-05-24:
**Describe the bug**
When I building for ESP32C3,ST7735,It doesn`t work.Script raised an error at [1271/1935] Building C object esp-idf/main_esp32c3/CMakeFiles/__idf_main_esp32c3.dir/__/machine_hw_spi.c.obj.
**Expected behavior**
/home/other/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c: In function 'machine_hw_spi_quad_bus_make_new':
 error: 'mp_machine_hw_spi_quad_bus_type' undeclared (first use in this function); did you mean 'mp_machine_hw_spi_bus_type'?
513 |         self->base.type = &mp_machine_hw_spi_quad_bus_type;
|                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|                            mp_machine_hw_spi_bus_type

/home/other/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:439:28: error: 'mp_machine_hw_spi_octal_bus_type' undeclared (first use in this function); did you mean 'mp_machine_hw_spi_bus_type'?
439 |         self->base.type = &mp_machine_hw_spi_octal_bus_type;
|                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|                            mp_machine_hw_spi_bus_type

kdschlosser, 2025-05-24:
I just pushed a commit yesterday that seems to have broke it. I will fix it this evening. I forgot to declare a class before it gets used is what happened. Not a big deal to fix, just give me a little bit to do it.

GWRon, 2025-05-24:
I needed to use an older revision (`git checkout fe5276c681b3d19efc6cfc4712175c16343101b7`) to get rid of a different error in the same C file.

Then it failed with some python errors when I tried to execute `$ python3 make.py esp32 BOARD=ESP32_GENERIC_C3 DISPLAY=ILI9341 INDEV=XPT2046 deploy` - somehow in `esp32.py` the following code received some unexpected input data:
```python
            if deploy:
                if PORT == '(PORT)':
                    ports = find_esp32_ports(chip)

                    if len(ports) > 1:
                        query = []
                        for i, port in enumerate(ports):
                            query.append(str(i + 1) + ': ' + port)

                        query.append('')
                        query.append('Which ESP32? :')

                        res = input('\n'.join(query))
                        res = int(res) - 1
                    else:
                        res = 0

                    PORT = ports[res]

                    for i, arg in args:
                        if arg == '(PORT)':
                            args[i] = f'{PORT}'
                            break
```
(args was not successfully unpacking 2 elements)
changed that to
```python
                    for i, arg in enumerate(args):
```

next the following-up esp-execution passed `"-m"` and `"esptool"` as args to the esptool - which is only valid when executing it from the commandline, not from within python (maybe something esptool changed meanwhile - newer idf or so? cannot explain more: am only experimenting with that ESP stuff since yesterday).

```python
                # Remove "-m" and "esptool" if present
                if '-m' in args and 'esptool' in args:
                    try:
                        m_index = args.index('-m')
                        if args[m_index + 1] == 'esptool':
                            del args[m_index:m_index + 2]  # remove both '-m' and 'esptool'
                    except IndexError:
                        pass  # In case 'esptool' is missing or in wrong position
                #print("args:") #debug output
                #print(args) #debug output
                esptool.main(args)
```
I added above to remove that args (yes, one could do `args = args[2:]` or so to get rid of it, if it is always in the first position.

Afterwards I was able to finally compile and upload it to a "ESP32 C3 SuperMini" but was not able to fully try out things but `help("modules")` at least shows a lot of more modules than what I was able to see with lv_micropython (was not able to add the ili9314 and xpt-touch drivers).

kdschlosser, 2025-05-24:
don't add `deploy` to your build command for the time being. just do a compile without flashing. The builder will output the command you seed to run to flash at the end.

The deploy has not been tested as of yet. You might still however run into a compile issue because of the MCU you are using. I don't have a C3 to test with.

There are differences between the C series MCU's and the S series and I have not really dove into messing with the C series MCU's, mostly because they are single core and don't perform as well as the S series MCU's. That makes them a less than ideal choice to run a display. Most times I see the C series line of MCU's is when they are used as a second ESP32 on a display board and they get used as an IO Expander. Kind of shows you how board/display manufacturers feel about them as as well. They are not well suited to run a display, while they can run a display is it's a small display with an SPI connection I personally don't recommend using it for doing that. Just because it can be done doesn't mean it should be done, it's not going to give you the best results.

Now I am not saying that I am not going to help to get it to work. I will if you really want to stick with using the C series MCU. You may not be all that happy with the results in the end is what I am getting at. Graphics processing has a very large amount of overhead and you put that on top of having to run a Python interpreter and the MCU is really going to have to put some effort in especially with the single core.

Here is some information that will be helpful to you when picking an MCU to use.

* esp32:     
  * CPU: Xtensa dual-core 32-bit LX6 microprocessor, 240 MHz, Ultra-low-power (ULP) FSM co-processor 20Mhz
  * RAM: 520 KiB + PSRAM
  * Notes: Only having available DMA memory in internal RAM makes it best suited for small displays that a small frame buffer is needed, 480 x 320 is the maximum resolution to use double buffering. 

* esp32-s2:
  * CPU: Single-core Xtensa LX7 CPU, 240 MHz, Ultra-low-power (ULP) FSM co-processor 20Mhz
  * RAM: 320 KiB + PSRAM
  * Notes: The single core is not ideal when running a display

* esp32-s3:
  * CPU: Dual-core Xtensa LX7 CPU, 240 MHz, single-precision FPU, Ultra-low-power (ULP) FSM co-processor 20Mhz, Ultra-low-power RISC-V (RV32IMC) coprocessor 17.5 MHz
  * RAM: 512 KiB + PSRAM
  * Notes: This is the most popular MCU when running a display

* esp32-c61:
  * CPU: 32-bit RISC-V single-core
  * RAM: 320 KiB + PSRAM
  * Notes: The single core is not ideal when running a display

* esp32-c2:
  * CPU: 32-bit RISC-V single-core 120 MHz, RV32IMC ISA
  * RAM: 272 KiB
  * Notes: Not supported by MicroPython due to the low amount of available RAM

* esp32-c3:
  * CPU: Single-core 32-bit RISC-V CPU 160 MHz
  * RAM: 400 KiB + PSRAM
  * Notes: The single core is not ideal when running a display

* esp32-c5:
  * CPU: Single-core 32-bit RISC-V CPU 240 MHz, Low-power 32-bit RISC-V CPU 40 MHz
  * RAM: 384 KiB + PSRAM
  * Notes: The single core is not ideal when running a display

* esp32-c6: 
  * CPU: Single-core 32-bit RISC-V CPU 160 MHz, Low-power 32-bit RISC-V CPU 20 MHz. both processors have RV32IMAC  
  * RAM: 512 KiB
  * Notes: The single core is not ideal when running a display

* esp32-h2:
  * CPU: Single-core 32-bit RISC-V CPU 96 MHz
  * RAM: 320 KiB 
  * Notes: don't even bother using this, I don't think it's use is supported in MicroPython because of the low amount of RAM

* esp32-p4:
  * CPU: dual-core 32-bit RISC-V CPU 400 MHz, RV32IMAFC_Zicsr_Zifencei, single-precision Floating Point Unit, Low performance single-core 32-bit RISC-V CPU 40 MHz, 2D graphics accelerator
  * RAM: 768 KiB
  * Notes: Not supported in MicroPython yet. I cannot wait until it is added... This is going to be the go to MCU for running a display because it supports high speed DSI connections to a display. This MCU doesn't have WiFi or BlueTeeth so you will see a C series ESP32 on the same display board when the availability of these MCU's increases.  

I am hopefully going to be working with NXP in the near future to add their line of MCU's into the binding ecosystem. Their MCU's have these kinds of specs..

6 core Arm Cortex-A55 (2.31gHz per core), Arm Cortex-M7 (300 mHz) co processor, Arm Cortex-M33 (150 mHz) co processor, NXP eIQ Neutron Neural Processing Unit
1376kB Internal RAM
LPDDR5 External RAM
3D GPU (OpenGL ES 3.2, Vulkan 1.2, OpenCL 3.0)
2D GPU
DSI 4-lane, 2.5 Gbps/lane
LVDS 8-lane

Yee HAW!! LOL...  The above is a pre-production MCU (I use the term MCU loosely, it's more like a computer). but it has some serious power that's for sure. You can run a full tilt OS on it as well as bare metal. I am sure the price tag is going to be pretty high for that processor. That is their top of the line. There are many other lower variations available that would not cost anywhere near what this thing is going to cost. What really puts NXP ahead of the game is they have different lines of MCU's that are purpose made to handle different tasks. They have a whole line that is specifically designed to have displays attached. That means they are going to perform better when being used for that purpose. I just started the dialog with them about adding support for their MCU's literally just a few days ago so we will see what happens.

GWRon, 2025-05-25:
Thanks for the sum-up. Pretty interesting one.

I must confess: I have a lot (read around 15) of the single core MCUs (esp8266/wemos d1 mini, ESP32 S2, ESP32 C3) laying around here (bought them all for 1,20-1,50 each, to enhance the model railway of my father with some more dynamic... features -- but am bought too many "spare" ones :D). So I thought using them for some other things (non-phone RGB wall light "configurer" etc.). Nothing heavy in the sense of required processor load. Thought a single core one could do. 

This is why I bought some cheap ili9341 displays (was on sale for < 3 euro) and am now trying to get things work.

All I got for now is a white screen ... small examples do nothing, and the bigger one (using a tabview) lead to a reboot - maybe the esp32C3 is not properly configured - or I am doing things wrong.

Am not planning to derail this issue here ... so I try to keep it short.

```
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it.
[...]

LVGL MicroPython 
IDF version : c9763f62
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.
[...]

ELF file SHA256: 304c3aad2

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380f28
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xfd8
load:0x403cc710,len:0x948
load:0x403ce710,len:0x2c20
entry 0x403cc710
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-05-24; ESP32C3 module with ESP32C3
```
So either something in the lvgl binding lead to a corrupted stack or something else (idf stuff...dunno, am new to this) made the device give up.

closed 2025-06-10 by kdschlosser