lvgl-micropython/lvgl_micropython #359: Compile issue - micropython 1.25

alvins82, 2025-05-27:
Fresh checkout. Note - the build keeps running after the error and seems to complete but not sure about it because of the red error message (see below). Btw. I'm not sure I'm supposed to be attempting to compile as I know you just checked in the updates.

Cmd:  python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16

Error:

GEN /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/frozen_content.c
[1211/2124] Generating ../../genhdr/moduledefs.collected
Module registrations updated
[2123/2124] Generating binary image from built executable                                                                                                                           
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.bin
[2124/2124] cd /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/esptool_py && /Users/xxx/.espressif/python_env/idf5.4_py3.12_env/bin/python /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/partition_table/partition-table.bin /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.bin
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/esptool_py/CMakeFiles/app_check_size
cd /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/esp-idf/esptool_py && /Users/xxx/.espressif/python_env/idf5.4_py3.12_env/bin/python /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/partition_table/partition-table.bin /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.bin
Error: app partition is too small for binary micropython.bin size 0x25fbe0:
- Part 'factory' 0/0 @ 0x10000 size 0x25a000 (overflow 0x5be0)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_79644 and /Users/xxx/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_79644
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1

***** Resizing Partition *****

***** Running build again *****

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.

alvins82, 2025-05-27:
Ignore this for the moment. I saw the running build again - sorry I was blind. Still an issue with the OTA update - will open seperate issue if I can't resolve.

closed 2025-05-27 by alvins82

kdschlosser, 2025-05-27:
LOL...

The OTA stuff is not fully tested out yet. To be honest I don't even remember how I set it up. You can specify an OTA build by adding `--ota` to the build command and you can also set the application partition size using --partition-size and this will set the application partition size to whatever it is you want and it will override the resizing of the partition. The reason why this is beneficial is because of updates making the firmware larger. Trying to mess around with resizing a partition OTA is not fun to do. So what you would do if you have the extra space is you make the application partition size 1mb larger than the current firmware this way you specify the same partition size there after and even when updates to the firmware take place and if the application grown in size it is going to be a really long time before the firmware would grow to fill an additional 1mb. The ESP32 would be very outdated by then...

That's the trick to not having to monkey around with it.

You can also use this feature if you do not want to keep on having to upload your program to the ESP32 and/or you have data saved in the NVS that you want to retain. By using the --partition size and making the application partition larger any time you want to update the firmware you specify that same partition size and compile the firmware. What you don't do on those subsequent firmware builds is you omit the ` --erase-all` parameter when calling esptool to flash the firmware . This will leave all of your NVS and user files on the fat filesystem untouched.