lvgl-micropython/lvgl_micropython #360: Unable to boot - ESP32s3

alvins82, 2025-05-27:
**Describe the bug**
After building and flashing firmware to board - I am stuck in boot loop. This is the JTAG serial output.

```shell
ELF file SHA256: 

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376344
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce2820,len:0xeb0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc1c
load:0x403cb700,len:0x2fa4
entry 0x403c88ac

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MicroPython 
IDF version : 67c1de1e
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x420aa13c: 00004136 00a0b2cc 2a8102ad
Core  0 register dump:
PC      : 0x420aa142  PS      : 0x00060d30  A0      : 0x8037581b  A1      : 0x3fceb2d0  
A2      : 0x3fceb308  A3      : 0x00000020  A4      : 0x0000000c  A5      : 0xfc000000  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x82144982  A9      : 0x3fceb290  
A10     : 0x07c00000  A11     : 0xfffe0000  A12     : 0x00000000  A13     : 0x600c4064  
A14     : 0x3fceb2a4  A15     : 0x3fceb2a4  SAR     : 0x00000006  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x420aa13f:0x3fceb2d0 0x40375818:0x3fceb2f0 0x403ccb3d:0x3fceb340 0x403ccef9:0x3fceb380 0x403c8919:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
```

**Build Command**

```shell
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16
```

kdschlosser, 2025-05-27:
That's not JTAG output.

JTAG is also disabled as a default and you would need to use the following build commands to enable it...

`--enable-jtag-repl=y --enable-cdc-repl=n --enable-uart-repl=n`

You might be using CDC and to use that you should specify these commands..

`--enable-cdc-repl=y --enable-uart-repl=n`

Do you also mind decoding that backtrace for me. Do a google search for `esp32 decode backtrace` and you will find instruction on how to do it. It's pretty easy to do it just require a really long command to do it and I don't remember all of it off the top of my head.

You will need to locate the xtensa binary. it is located in one of the folders under ~/.espressif

alvins82, 2025-05-27:
Sorry I wasn't sure which ELF file to run it again so I have included both.

```shell
(base) esp32 % xtensa-esp32-elf-addr2line -pfiaC -e ./build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.elf -a 0x420aa13f:0x3fceb2d0 0x40375818:0x3fceb2f0 0x403ccb3d:0x3fceb340 0x403ccef9:0x3fceb380 0x403c8919:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710 
0x420aa13f: esp_chip_info at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32s3/chip_info.c:13
0x40375818: start_other_core at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/esp_system/port/cpu_start.c:273
 (inlined by) call_start_cpu0 at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/esp_system/port/cpu_start.c:620
0x403ccb3d: ?? ??:0
0x403ccef9: ?? ??:0
0x403c8919: ?? ??:0
0x40045c01: ?? ??:0
0x40043ab6: ?? ??:0
0x40034c45: ?? ??:0
(base) esp32 % xtensa-esp32-elf-addr2line -pfiaC -e ./build-ESP32_GENERIC_S3-SPIRAM_OCT/bootloader/bootloader.elf -a 0x420aa13f:0x3fceb2d0 0x40375818:0x3fceb2f0 0x403ccb3d:0x3fceb340 0x403ccef9:0x3fceb380 0x403c8919:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710 
0x420aa13f: ?? ??:0
0x40375818: ?? ??:0
0x403ccb3d: set_cache_and_start_app at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/bootloader_support/src/bootloader_utility.c:958
 (inlined by) unpack_load_app at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/bootloader_support/src/bootloader_utility.c:832
 (inlined by) load_image at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/bootloader_support/src/bootloader_utility.c:742
0x403ccef9: bootloader_utility_load_boot_image at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/bootloader_support/src/bootloader_utility.c:544
0x403c8919: call_start_cpu0 at /Users/xxx/git-personal/lvgl_micropython/lib/esp-idf/components/bootloader/subproject/main/bootloader_start.c:57
0x40045c01: ?? ??:0
0x40043ab6: ?? ??:0
0x40034c45: ?? ??:0
```

kdschlosser, 2025-05-27:
your backtrace is right here...

```
0x420aa13f:0x3fceb2d0 0x40375818:0x3fceb2f0 0x403ccb3d:0x3fceb340 0x403ccef9:0x3fceb380 0x403c8919:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
```

This should be the command you need to run.. if you run it from the root directory of the cloned repo
```
~/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32-elf-addr2line -aspCfire "lib/micropython/ports/esp32/build/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.elf" 0x420aa13f:0x3fceb2d0 0x40375818:0x3fceb2f0 0x403ccb3d:0x3fceb340 0x403ccef9:0x3fceb380 0x403c8919:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
```

alvins82, 2025-05-27:
Note - I have also have a generic ESP32 board and compiled and flashed on that. No problem.

kdschlosser, 2025-05-28:
and you said that vanilla micropython works without any issues?  The same version??

alvins82, 2025-05-28:
> and you said that vanilla micropython works without any issues? The same version??

Build from here yes works - https://micropython.org/download/ESP32_GENERIC_S3/

kdschlosser, 2025-05-28:
Give it a try. I just pushed a commit that will hopefully fix it. I have not tried to compile it at all so there might be errors.

Actually, I think I know where one might be. Hold on a sec.

kdschlosser, 2025-05-28:
ok it should be good now.

I recommend a fresh clone to make sure you get all of the updates.

alvins82, 2025-05-28:
Ok let me give it a shot - give me 10mins

alvins82, 2025-05-28:
```shell
In file included from /Users/alvin/git-personal/lvgl_micropython/lib/micropython/py/mpconfig.h:91,
from /Users/alvin/git-personal/lvgl_micropython/lib/micropython/py/mpstate.h:31,
from /Users/alvin/git-personal/lvgl_micropython/lib/micropython/py/runtime.h:29,
from /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c:29:
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:148:1: warning: missing terminating " character
148 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:164:1: warning: missing terminating " character
164 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:220:1: warning: missing terminating " character
220 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:233:1: warning: missing terminating " character
233 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:245:1: warning: missing terminating " character
245 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:258:1: warning: missing terminating " character
258 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:270:1: warning: missing terminating " character
270 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:290:1: warning: missing terminating " character
290 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:302:1: warning: missing terminating " character
302 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:319:1: warning: missing terminating " character
319 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:331:1: warning: missing terminating " character
331 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:348:1: warning: missing terminating " character
348 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:3203:1: warning: missing terminating " character
3203 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:3225:1: warning: missing terminating " character
3225 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:4631:1: warning: missing terminating " character
4631 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:4651:1: warning: missing terminating " character
4651 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:4992:1: warning: missing terminating " character
4992 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:5017:1: warning: missing terminating " character
5017 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:5266:1: warning: missing terminating " character
5266 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:5292:1: warning: missing terminating " character
5292 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:5589:1: warning: missing terminating " character
5589 | "
| ^
/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/mpconfigport.h:5621:1: warning: missing terminating " character
5621 | "
[... 2909 lines cut ...]
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/alvin/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.bt_funcs.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script /Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
[... cut ...]
-- Build files have been written to: /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/bootloader
ninja: build stopped: subcommand failed.
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
ninja failed with exit code 1, output of the command is in the /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_88189 and /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_88189
make: *** [all] Error 1
```

kdschlosser, 2025-05-28:
OK you can delete that post because I am not going to keep on scrolling that...

alvins82, 2025-05-28:
> OK you can delete that post because I am not going to keep on scrolling that...

Amended (sorry).

kdschlosser, 2025-05-28:
OK that problem is fixed.

Hopefully...

alvins82, 2025-05-28:
```shell
[1448/2124] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_sdcard.c.obj                                                                                
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_sdcard.c.obj
/Users/alvin/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/Users/alvin/git-personal/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DCFG_TUSB_MCU=OPT_MCU_ESP32S3 -DESP_PLATFORM -DFFCONF_H=\"/Users/alvin/git-personal/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"67c1de1e\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_BOARD_BUILD_NAME=\"ESP32_GENERIC_S3-SPIRAM_OCT\" -DMICROPY_HW_BOARD_NAME="\"Generic ESP32S3 module with Octal-SPIRAM\"" -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/config -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/Users/alvin/git-personal/lvgl_micropython/ext_mod/spi3wire -I/Users/alvin/git-personal/lvgl_micropython/ext_mod/spi3wire/include -I/Users/alvin/git-personal/lvgl_micropython/ext_mod/lcd_bus -I/Users/alvin/git-personal/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_lcd/rgb/include -I/Users/alvin/git-personal/lvgl_micropython/lib -I/Users/alvin/git-personal/lvgl_micropython/lib/lvgl -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython/lib/tinyusb/src -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython/shared/tinyusb -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32 -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3 -I/Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/freertos/config/xtensa/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/dma/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/ldo/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/debug_probe/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32s3/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/heap/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/heap/tlsf -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/log/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/soc/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/soc/esp32s3 -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/soc/esp32s3/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/soc/esp32s3/register -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/hal/esp32s3/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/hal/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/include -I/Users/alvin/git-personal/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3/include/esp32s3 -I/Users/alvin/git-pers [... cut ...]
[1457/2124] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/Users/alvin/git-personal/lvgl_micropython/lib/micropython/lib/littlefs/lfs2.c.obj               
ninja: build stopped: subcommand failed.
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
ninja failed with exit code 1, output of the command is in the /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_24388 and /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_24388
make: *** [all] Error 1
```

rena2019, 2025-05-28:
```
> /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c: In function 'machine_sdcard_make_new':
> /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c:179:16: error: 'arg_vals' undeclared (first use in this function)
> 179 |     int freq = arg_vals[ARG_freq].u_int;
> |                ^~~~~~~~
> /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c:179:16: note: each undeclared identifier is reported only once for each function it appears in
> /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c:226:5: error: expected '}' before 'else'
> 226 |     else {
> |     ^~~~
> /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c:205:27: warning: unused variable 'spi_host_id' [-Wunused-variable]
> 205 |         spi_host_device_t spi_host_id = self->host.slot;
> |                           ^~~~~~~~~~~
> [1457/2124] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/Users/alvin/git-personal/lvgl_micropython/lib/micropython/lib/littlefs/lfs2.c.obj               
> ninja: build stopped: subcommand failed.
> -e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
> ninja failed with exit code 1, output of the command is in the /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_24388 and /Users/alvin/git-personal/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_24388
> make: *** [all] Error 1
```

I have same issue if I build with:
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4 DISPLAY=nv3041a INDEV=gt911`

alvins82, 2025-05-29:
No more crash. Boots up fine.

kdschlosser, 2025-05-29:
YAY!!!

I am useful for something. LOL...