lvgl-micropython/lvgl_micropython #361: Another compiled issue v1.25.0

tetopik, 2025-05-27:
Hi, Kevin. According to this newest update for `v1.25.0` with `IDFv5.4.0`, I'm compiling for all of my `esp32` line boards like the generic one, S2, S3, and even C3.

ESP32 generic and ESP32-S3 compiled flawlessly, except for ESP32-S2 and ESP32-C3.

Here's for the S2:
```
[1306/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_sdcard.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_sdcard.c.obj
/home/tetopik/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s2-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DCFG_TUSB_MCU=OPT_MCU_ESP32S2 -DESP_PLATFORM -DFFCONF_H=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"67c1de1e\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_BOARD_BUILD_NAME=\"LOLIN_S2_MINI\" -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-LOLIN_S2_MINI/config -I/home/tetopik/micropython/lvgl_micropython/lib/micropython -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire -I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire/include -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/tetopik/micropython/lvgl_micropython/lib -I/home/tetopik/micropython/lvgl_micropython/lib/lvgl -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/tinyusb/src -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/shared/tinyusb -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/boards/LOLIN_S2_MINI -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-LOLIN_S2_MINI -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32s2 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/dma/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/ldo/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/debug_probe/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32s2/. -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32s2/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/heap/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/heap/tlsf -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/log/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32s2 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32s2/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32s2/register -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/esp32s2/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s2/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s2/include/esp32s2 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s2 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_common/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/private -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/esp32s2/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/deprecated_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include/apps -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include/apps/sntp -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/lwip/src/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/port/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/port/freertos/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/app_update/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bootloader_support/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_app_format/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_bootloader_format/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_partition/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_timer/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_wifi/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_wifi/include/local -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_wifi/wifi_apps/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_wifi/wifi_apps/nan_app/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_event/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_phy/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_phy/esp32s2/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_netif/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/driver/deprecated -I/home/tetopik/micropython/lvgl_micropyt [... cut ...]
26 | #define MICROPY_HW_ENABLE_UART_REPL  (0)
|         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[1349/2003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopi[1350/2003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopi[1351/2003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopi[1352/2003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/littlefs/lfs2.c.obj
ninja: build stopped: subcommand failed.
```

Many thanks, m8!

tetopik, 2025-05-27:
I'm adding a macro `#if SOC_SDMMC_HOST_SUPPORTED` in `machine_sdcard.c` file before both the `else` line for the `if(is_spi)` statement.

```c
static mp_obj_t machine_sdcard_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *all_args) {
    // check arguments
    enum {
        ARG_slot,
        ARG_width,
        ARG_cd,
        ARG_wp,
        ARG_cmd,
        ARG_clk,
        ARG_data_pins,
        ARG_spi_bus,
        ARG_cs,
        ARG_freq,
    };

    static const mp_arg_t make_new_args[] = {
        { MP_QSTR_slot,      MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int =  1} },
        { MP_QSTR_width,     MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int =  1} },
        { MP_QSTR_cd,        MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        { MP_QSTR_wp,        MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        // Added so a user can set their own custom data pins
        { MP_QSTR_cmd,       MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        { MP_QSTR_clk,       MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        { MP_QSTR_data_pins, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = mp_const_none} },
        // These are only needed if using SPI mode
        { MP_QSTR_spi_bus,   MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = mp_const_none} },
        { MP_QSTR_cs,        MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        // freq is valid for both SPI and SDMMC interfaces
        { MP_QSTR_freq,      MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 20000000} },
    };

    mp_arg_val_t args[MP_ARRAY_SIZE(make_new_args)];
    mp_arg_parse_all_kw_array(
        n_args,
        n_kw,
        all_args,
        MP_ARRAY_SIZE(make_new_args),
        make_new_args,
        args
    );

    bool is_spi;
    int slot_num;
    sdcard_card_obj_t *self = mp_obj_malloc_with_finaliser(sdcard_card_obj_t, &machine_sdcard_type);

    if (args[ARG_spi_bus].u_obj != mp_const_none) {
        mp_machine_hw_spi_bus_obj_t *spi_bus = MP_OBJ_TO_PTR(args[ARG_spi_bus].u_obj);
        slot_num = (int)spi_bus->host;
        is_spi = true;
        self->spi_device.spi_bus = spi_bus;
    } else {
        slot_num = (int)args[ARG_slot].u_int;
        is_spi = false;

        if (slot_num < 0 || slot_num > 3) {
            mp_raise_ValueError(MP_ERROR_TEXT("slot number must be between 0 and 3 inclusive"));
        }
    }

    self->flags = 0;
    self->sdspi_handle = -1;
    // Note that these defaults are macros that expand to structure
    // constants so we can't directly assign them to fields.
    uint32_t freq = (uint32_t)args[ARG_freq].u_int;
    if (is_spi) {
        sdmmc_host_t _temp_host = SDSPI_HOST_DEFAULT();
        _temp_host.max_freq_khz = (int)(freq / 1000);
        self->host = _temp_host;
    } 
    #if SOC_SDMMC_HOST_SUPPORTED    
    else {
        sdmmc_host_t _temp_host = SDMMC_HOST_DEFAULT();
        _temp_host.max_freq_khz = (int)(freq / 1000);
        self->host = _temp_host;
    }
    #endif

    if (is_spi) {
        // Needs to match spi_dev_defaults above.
        self->host.slot = slot_num;
    }

    check_esp_err(self->host.init());
    self->flags |= SDCARD_CARD_FLAGS_HOST_INIT_DONE;

    if (is_spi) {
        // SPI interface
        self->dev_config = (sdspi_device_config_t){
            .host_id = (spi_host_device_t)self->host.slot,
            .gpio_cs = (int)args[ARG_cs].u_int,
            .gpio_cd = (int)args[ARG_cd].u_int,
            .gpio_wp = (int)args[ARG_wp].u_int,
            .gpio_int = SDSPI_SLOT_NO_INT
        };

        if (self->spi_device.spi_bus->state == MP_SPI_STATE_STOPPED) {
            mp_machine_hw_spi_bus_initilize(self->spi_device.spi_bus);
        }

        check_esp_err(sdspi_host_init_device(&self->dev_config, &self->sdspi_handle));

        mp_machine_hw_spi_bus_add_device(&self->spi_device);
    } 
    #if SOC_SDMMC_HOST_SUPPORTED
    else {
        // SD/MMC interface
        sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

        // Stronger external pull-ups are still needed but apparently
        // it is a good idea to set the internal pull-ups anyway.
        // slot_config.flags = SDMMC_SLOT_FLAG_INTERNAL_PULLUP;
        slot_config.gpio_cd = (int)args[ARG_cd].u_int;
        slot_config.gpio_wp = (int)args[ARG_wp].u_int;

        int clk = (int)args[ARG_clk].u_int;
        int cmd = (int)args[ARG_cmd].u_int;
        int width = (int)args[ARG_width].u_int;

        if (clk != -1) slot_config.clk = clk;
        if (cmd != -1) slot_config.cmd = cmd;

        if (args[ARG_data_pins].u_obj != mp_const_none) {
            mp_obj_tuple_t *t = MP_OBJ_TO_PTR(args[ARG_data_pins].u_obj);
            if ((int)t->len != width) {
                mp_raise_ValueError(MP_ERROR_TEXT("width does not match the number of data pins provided"));
            }

            slot_config.d0 = (int)mp_obj_get_int(t->items[0]);

            if (width >= 4) {
                slot_config.d1 = (int)mp_obj_get_int(t->items[1]);
                slot_config.d2 = (int)mp_obj_get_int(t->items[2]);
                slot_config.d3 = (int)mp_obj_get_int(t->items[3]);
            }
            if (width == 8) {
                slot_config.d4 = (int)mp_obj_get_int(t->items[4]);
                slot_config.d5 = (int)mp_obj_get_int(t->items[5]);
                slot_config.d6 = (int)mp_obj_get_int(t->items[6]);
                slot_config.d7 = (int)mp_obj_get_int(t->items[7]);
            }
        }

        if (width == 1 || width == 4 || (width == 8 && slot_num == 0)) {
            slot_config.width = width;
        } else {
            mp_raise_ValueError(MP_ERROR_TEXT("width must be 1 or 4 (or 8 on slot 0)"));
        }

        check_esp_err(sdmmc_host_init_slot(self->host.slot, &slot_config));
    }
    #endif // SOC_SDMMC_HOST_SUPPORTED

    return MP_OBJ_FROM_PTR(self);
}
```

I don't know what I'm doing, it just compiled succesfully for all esp32 boards mentioned before.
I didn't test the sdcard module yet.

kdschlosser, 2025-05-27:
It's not working properly at the present time. Or so I am told. The firmware crashes once it is flashed and the ESP32 first boots.

If you can confirm that is happening that would be a big help.

I do know that I need to make some changes to the code. Due to the very large number of changes made to the ESP-IDF it would take me weeks to go through them to see what I need to change to get it to compile for all of the different ESP32's I figured getting it to compile with the S3 and then address each issue when users tell me there is an issue. This is an easier way to correct things because I do not have to spend weeks reading over the ESP-IDF code to see what needs to be changed in my code.

You are on the right track with your fix. The better solution would be this..

```c
static mp_obj_t machine_sdcard_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *all_args) {
    // check arguments
    enum {
        ARG_slot,
        ARG_width,
        ARG_cd,
        ARG_wp,
        ARG_cmd,
        ARG_clk,
        ARG_data_pins,
        ARG_spi_bus,
        ARG_cs,
        ARG_freq,
    };

    static const mp_arg_t make_new_args[] = {
        { MP_QSTR_slot,      MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int =  1} },
        { MP_QSTR_width,     MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int =  1} },
        { MP_QSTR_cd,        MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        { MP_QSTR_wp,        MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        // Added so a user can set their own custom data pins
        { MP_QSTR_cmd,       MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        { MP_QSTR_clk,       MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        { MP_QSTR_data_pins, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = mp_const_none} },
        // These are only needed if using SPI mode
        { MP_QSTR_spi_bus,   MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = mp_const_none} },
        { MP_QSTR_cs,        MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = -1} },
        // freq is valid for both SPI and SDMMC interfaces
        { MP_QSTR_freq,      MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 20000000} },
    };

    mp_arg_val_t args[MP_ARRAY_SIZE(make_new_args)];
    mp_arg_parse_all_kw_array(
        n_args,
        n_kw,
        all_args,
        MP_ARRAY_SIZE(make_new_args),
        make_new_args,
        args
    );

    bool is_spi;
    int slot_num;
    sdcard_card_obj_t *self = mp_obj_malloc_with_finaliser(sdcard_card_obj_t, &machine_sdcard_type);

    if (args[ARG_spi_bus].u_obj != mp_const_none) {
        mp_machine_hw_spi_bus_obj_t *spi_bus = MP_OBJ_TO_PTR(args[ARG_spi_bus].u_obj);
        slot_num = (int)spi_bus->host;
        is_spi = true;
        self->spi_device.spi_bus = spi_bus;
    } else {
        slot_num = (int)args[ARG_slot].u_int;
        is_spi = false;

        if (slot_num < 0 || slot_num > 3) {
            mp_raise_ValueError(MP_ERROR_TEXT("slot number must be between 0 and 3 inclusive"));
        }
    }

    self->flags = 0;
    self->sdspi_handle = -1;
[... 74 lines cut ...]
                slot_config.d4 = (int)mp_obj_get_int(t->items[4]);
                slot_config.d5 = (int)mp_obj_get_int(t->items[5]);
                slot_config.d6 = (int)mp_obj_get_int(t->items[6]);
                slot_config.d7 = (int)mp_obj_get_int(t->items[7]);
            }
        }

        if (width == 1 || width == 4 || (width == 8 && slot_num == 0)) {
            slot_config.width = width;
        } else {
            mp_raise_ValueError(MP_ERROR_TEXT("width must be 1 or 4 (or 8 on slot 0)"));
        }

        check_esp_err(sdmmc_host_init_slot(self->host.slot, &slot_config));

    #else
        mp_raise_OSError(MP_ERROR_TEXT("MMC card reader support is not available on this MCU"));
    #endif // SOC_SDMMC_HOST_SUPPORTED
    }
    return MP_OBJ_FROM_PTR(self);
}

```

The way you have it if the MCU doesn't support MMC no error will be produced and a non functioning `machine.SDCard` instance will be returned and if operations get performed on that instance it will more than likely cause a core panic and crash. Raising an exception is the way to go if a user attempts to use the MMC portion of the SDCard reader on an MCU that doesn't support it.

tetopik, 2025-05-28:
Replace the `machine_sdcard_make_new` function in `machine_sdcard.c` with the code above, ESP32-S2 and ESP32-C3 fails:
```
/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c: In function 'machine_sdcard_make_new':
/home/tetopik/micropython/lvgl_micropython/lib/micropython/py/misc.h:335:26: error: passing argument 1 of 'mp_raise_OSError' makes integer from pointer without a cast [-Wint-conversion]
335 | #define MP_ERROR_TEXT(x) (mp_rom_error_text_t)MP_COMPRESSED_ROM_TEXT(x)
/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/machine_sdcard.c:266:26: note: in expansion of macro 'MP_ERROR_TEXT'
266 |         mp_raise_OSError(MP_ERROR_TEXT("MMC card reader support is not available on this MCU"));
|                          ^~~~~~~~~~~~~
/home/tetopik/micropython/lvgl_micropython/lib/micropython/py/runtime.h:269:36: note: expected 'int' but argument is of type 'const char *'
269 | NORETURN void mp_raise_OSError(int errno_);
|                                ~~~~^~~~~~
[1305/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/modesp32[1306/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/modsocke[1307/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/__/pins.[1308/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/tinyusb/src/device/usbd_contr[1309/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_hw_spi.c.obj                                                                    [1310/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopi[1311/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/tinyusb/src/class/cdc/cdc_dev[1312/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/tinyusb/src/common/tusb_fifo.[1313/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/oofatfs/ff.c.obj             [1314/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/modespno[1315/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/tinyusb/src/portable/synopsys[1316/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/tinyusb/src/device/usbd.c.obj[1317/1980] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/littlefs/lfs2.c.obj          
ninja: build stopped: subcommand failed.
```

tetopik, 2025-05-29:
So I guess it's solved now, thanks man

kdschlosser, 2025-05-29:
If you say so.. LOL... There are a lot of changes I am making and I know I did some work to the SDCard driver. There is going to be a really large update coming pretty soon.

tetopik, 2025-05-30:
The main focus of this specific thread is to get it compiled, and now it does.

I got nothing to do with sdcard module in the meantime.

Can't wait to see the upcoming update, thanks for the fantastic effort you put on this Kevin