lvgl-micropython/lvgl_micropython #372: No ESP32 working firmware possible

getznerh, 2025-06-02:
Hi everybody! I am working with ESP32 for many months and trying to get a working ESP32 with LVGL and Micropython since i found the tremendous work from Kevin. But now i am a bit frustrated. Since Beginning of May i can not get a working firmware. The last reallly firmware which is without problems (for me!) ist from End of April with commit 24c10b.
The version fe5276c from 7.May has problems with displaying bitmaps, they are destroyed. I can correct this with removing some changes in rgb_bus_rotation.c.
The different changes until using MP 1.25 and esp-idf 5.4 can not be compiled. And since 1.25 the firmware crashes, when task_handler is starting. The display flashes shortly an probably after copiing some frame buffer, the program stops and reboots with the reason Wathdog.

I have: Windows 11 24H2 with WSL and Ubuntu 24.04
Python 3.12 and 32 GB RAM
I use a Elecrow ESP32S3 5" Display
I clone the repository in a empty directory
and then i checkout the desired verion
then i compile with the build command
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT 
    --flash-size=4 DISPLAY=rgb_display INDEV=gt911 INDEV=xpt2046 
    --enable-cdc-repl=n --enable-jtag-repl=n
I flush the firmware under wsl with the command from the build-command
then i copy the small test program easy5.py

With the SDL Version, the actual verson from 1.6. ist OK
but not with ESP32S3.

The small testprogramm easy5.py i send as attachment.
Regards Hans

[easy5.txt](https://github.com/user-attachments/files/20557413/easy5.txt)

kdschlosser, 2025-06-03:
you are calling the display `init` method  too late. You need to call that before creating the indev driver instance.

```python

from micropython import const  # NOQA
import time
import lvgl as lv  # NOQA
import lcd_bus  # NOQA

_WIDTH = const(800)
_HEIGHT = const(480)

lv.init()

from i2c import I2C
import gt911
#from machine import Pin
_CTP_SCL = const(20)
_CTP_SDA = const(19)

_LCD_FREQ = const(13000000)
_PCLK_ACTIVE_NEG = const(1)

_HSYNC_PULSE_WIDTH = const(4)
_HSYNC_BACK_PORCH = const(43)
_HSYNC_FRONT_PORCH = const(8)

_VSYNC_PULSE_WIDTH = const(4)
_VSYNC_BACK_PORCH = const(12)
_VSYNC_FRONT_PORCH = const(8)

# Die PIN Nummern stimmen mit dem Beispiel in https://pastebin.com/raw/aNBxmQhg überein
_PCLK = const(0)
_HSYNC = const(39)
_VSYNC = const(41)
_DE = const(40)
_DISP = const(-1)
_BCKL = const(2)
_DRST = None
_DPWR = None

# touch drvier
I2C_BUS = I2C.Bus(
    host=1,
    scl=_CTP_SCL,
    sda=_CTP_SDA,
    freq=400000,
    use_locks=False
)

TOUCH_DEVICE = I2C.Device(
    I2C_BUS,
    dev_id=gt911.I2C_ADDR,
    reg_bits=gt911.BITS
)

# Display Anschlüsse
_DATA15 = const(14)  # B7
_DATA14 = const(21)  # B6
_DATA13 = const(47)  # B5
_DATA12 = const(48)  # B4
_DATA11 = const(45)  # B3
_DATA10 = const(4)  # G7
_DATA9 = const(16)  # G6
_DATA8 = const(15)  # G5
_DATA7 = const(7)  # G4
_DATA6 = const(6)  # G3
_DATA5 = const(5)  # G2
_DATA4 = const(1)  # R7
_DATA3 = const(9)  # R6
_DATA2 = const(46)  # R5
_DATA1 = const(3)  # R4
_DATA0 = const(8)  # R3

display_bus = lcd_bus.RGBBus(
    hsync=_HSYNC,
    vsync=_VSYNC,
    de=_DE,
#    disp=_DISP,
    pclk=_PCLK,
    data0=_DATA0,
    data1=_DATA1,
    data2=_DATA2,
    data3=_DATA3,
    data4=_DATA4,
    data5=_DATA5,
    data6=_DATA6,
    data7=_DATA7,
    data8=_DATA8,
    data9=_DATA9,
    data10=_DATA10,
    data11=_DATA11,
    data12=_DATA12,
    data13=_DATA13,
    data14=_DATA14,
    data15=_DATA15,
    freq=_LCD_FREQ,
    hsync_front_porch=_HSYNC_FRONT_PORCH,
    hsync_back_porch=_HSYNC_BACK_PORCH,
    hsync_pulse_width=_HSYNC_PULSE_WIDTH,
    hsync_idle_low=True,
    vsync_front_porch=_VSYNC_FRONT_PORCH,
    vsync_back_porch=_VSYNC_BACK_PORCH,
    vsync_pulse_width=_VSYNC_PULSE_WIDTH,
    vsync_idle_low=True,
    de_idle_high=False,
    pclk_idle_high=False,
    pclk_active_low=_PCLK_ACTIVE_NEG,
#     disp_active_low=False,
#     refresh_on_demand=False
)

# lt kdschlosser wird der Buffer automatisch erzeugt
# allenfalls hier kleiner Buffer wegen Geschwindigkeit
# buf1 = bus.allocate_framebuffer(_WIDTH * _HEIGHT * 2, lcd_bus.MEMORY_SPIRAM)
# buf2 = bus.allocate_framebuffer(_WIDTH * _HEIGHT * 2, lcd_bus.MEMORY_SPIRAM)

import rgb_display  # NOQA

display = rgb_display.RGBDisplay(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
#     frame_buffer1=buf1,
#     frame_buffer2=buf2,
    reset_pin=_DRST,
    reset_state=rgb_display.STATE_HIGH,
    power_pin=_DPWR,
    power_on_state=rgb_display.STATE_HIGH,
    backlight_pin=_BCKL,
    backlight_on_state=rgb_display.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=False # sonst stimmen die Farben nicht
)
# this is where the power and init functions shuld be located.
display.set_power(True)
display.init()

indev = gt911.GT911(TOUCH_DEVICE, startup_rotation=lv.DISPLAY_ROTATION._0)
if indev.hw_size != (_WIDTH, _HEIGHT):
    fw_config = indev.firmware_config
    fw_config.width = _WIDTH
    fw_config.height = _HEIGHT
    fw_config.save()
    del fw_config

print('Scan i2c bus...')
devices = I2C_BUS.scan()

if len(devices) == 0:
    print("No i2c device !")
else:
    print('i2c devices found:',len(devices))

for device in devices:  
    print("Decimal address: ",device," | Hexa address: ",hex(device))
#---------------------------------------------------------------------------

import task_handler
import time

#---------------------------------------------------------------------------
# hier wird eine Menge von Buttons mit unterschiedlichen Farben generiert

display.set_backlight(100)

scrn = lv.screen_active()
scr = scrn # die Beispiele haben überall scr

#---------------------------------------------------------------------------
# zudem noch ein Slider zu Test des Toucinterfaces

scrn.set_style_bg_color(lv.color_hex(0xff7733), 0)
slider = lv.slider(scrn)
slider.set_size(200, 45)
slider.center()

# Callback bei Slider Touch
def on_value_changed(e):
    # the slider is at the module level and there for is a global. Just call the slider directly.
    print('VALUE_CHANGED:', slider.get_value())

    # if you want to use the get_target_obj method the returned object from that is simply a location to memory. 
    # you need to tell it what is at that location. That is done using the `__cast__` method. 
    # print('VALUE_CHANGED:', e.get_target_obj().__cast__().get_value())

slider.add_event_cb(on_value_changed, lv.EVENT.VALUE_CHANGED, None)

th = task_handler.TaskHandler()

# da kommt aber REPL wieder
print("hallo")

# don't use the watchdog timer as it tends to reboot the ESP when it really shouldn't be. 
# There are operations that do take a bit of time to complete when using a display.
```

I apologize for the issues with the RGB driver. The RGB driver is challenging to say the least. This is because the buffer has to be sent in an endless loop to the display because the display doesn't have an graphics memory. This causes a very large amount of processor time to be used. All of my displays and MCU's are packed away in boxes because I am renovating my office area/workshop so I don't have access to them to test with at the moment. I will get the issues sorted out it is going to be a while longer before I will be able to get to my gear to test with. Sorry for the inconvenience.

fabse-hack, 2025-06-06:
your building command is wrong too .....

```
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT
--flash-size=4 DISPLAY=rgb_display INDEV=gt911 INDEV=xpt2046
--enable-cdc-repl=n --enable-jtag-repl=n
```

the (display is wrong ?? and) the INDEV is double ....

getznerh, 2025-06-08:
my building script ist NOT wrong. rgb_display ist a valid display and double INDEV enables me, to use the firmware for more than one type of MCU. This ist checked! The hint with position of display.init() before gt911 ist OK but does not change anything.
Regards Hans

kdschlosser, 2025-06-09:
Heyyy, That's a heck of  lot more informative...

OK so now I can kind of get an idea of where the problem is stemming from I can give you some direction.

Add `--debug` to your build command, recompile and flash the firmware and then upload your test script and give it a run. Paste all of the output into a text file and attach the file to a post here.

kdschlosser, 2025-06-10:
OK so scratch that last post I made. I did find an issue and I corrected it. Give it a try and see if it works for ya.

getznerh, 2025-06-10:
I had to correct one error:
in rgb_bus.h 
    #if CONFIG_IDF_TARGET_ESP32S3
        #define RGB_LCD_BUS_NEEDS_SEPARATE_RESTART_LINK 1
is missing:        #define RGB_LCD_PANEL_MAX_FB_NUM         3 // maximum supported frame buffer number
    #endif
i took the right value from  .\lvgl_micropython\lib\esp-idf\components\esp_lcd\rgb\esp_lcd_panel_rgb.c

But i have the same problems and send you the new debugot1.txt. If you like i can also decode the backtrace

[debugout1.txt](https://github.com/user-attachments/files/20676437/debugout1.txt)

kdschlosser, 2025-06-10:
OK so we are onto the next issue.

getznerh, 2025-06-11:
Hi Kevin! off course, i will try as long as you want! Now you have removed machine_timer.c and *.h and now it is running nearly as expected. I append another debugout3.txt. My easy5.py testprogram shows at first the slider. When you change the slider, your rotate function destroys the slider as you can see in the picture. I found out this is caused by your changes because of the black lines when rotated. I will try to correct this, but a little bit later the day.

[debugout3.txt](https://github.com/user-attachments/files/20686698/debugout3.txt)

[image]

kdschlosser, 2025-06-12:
OK I pushed the changes to the repo. I was damned close to having it right. Just a couple of small changes needed. I was correct about adding that +1 to get rid of the black line but I needed to adjust the other rotations to account for it. It's some pretty complex math to do a rotation and copying from one buffer to another and keeping the position correct inside the buffer the data is being copied into in order to stitch a complete frame from a series of partial frames. 

I wanted to have as little math being done as possible in the iteration loops. This would help to keep the speed up which is a really important thing to do. Any display IC that utilizes full frame buffers is going to benefit from using this driver. It offloads the work onto the second core of the ESP so the first core is able to keep on rendering new data. The only thing that limits how fast it is able to work is the SPIRAM, The bottleneck caused by the SPIRAM is because of it's speed, some of that is able to be mitigated by overclocking the SPI speed for the SPIRAM and the SPI Flash. If you have an ESP32-S3 that has octal SPIRAM and octal SPI flash you can overclock the SPI for them and get an effective clock speed of 240mHz instead of the default 80mHz. That is a considerable difference. So instead of seeing 40mHz operations on the RAM, `80 / 2` because if 2 reads occurring at the same time or possibly a read and a write you end up at 120mHz for each operation. Theoretically it should be 3 times faster, realistically it's probably 2 times faster. which is still a significant increase.