lvgl-micropython/lvgl_micropython #377: How to create 3 different SPI bus instance?

tetopik, 2025-06-07:
Hi, Kevin.

Using a CYD board like ESP32-2432S028R  initialize 3 different SPI bus for it's own display, touchscreen, and sdcard reader. Create those 3 SPI instance throw an error `no SPI(3)`. The SoftSPI module also seems incompatible, is there any other approach to solve this?

kdschlosser, 2025-06-07:
You should not need 3 different SPI instances. I believe all of those things are all on the same SPI bus some only a single SPI bus instance is needed and that instance you pass to the display bus driver, the SD card driver and for the touch screen you create an `SPI.Device` instance passing the bus to that and then you pass that `SPI.Device` instance to the touch driver...

How you know what needs to be done is by looking at the pins that are used. If the MOSI and MISO pins are the same then they share the same bus.

Here is an example.

This is pseudo code and is for example purposes only.

```python
from micropython import const
import machine
import lcd_bus
import xpt2046

_WIDTH = const(240)
_HEIGHT = const(240)

_HOST = const(1)  # SPI Host 2
_SCK = const(11)
_MOSI = const(12)
_MISO = const(13)

_SD_CS = const(14)
_LCD_CS = const(15)
_TP_CS = const(16)

_LCD_DC = const(17)

_LCD_FREQ = const(80000000)
_SD_FREQ = const(10000000)
_TP_FREQ = const(10000000)

spi_bus = machine.SPI.Bus(host=_HOST, mosi=_MOSI, miso=_MISO, sck=_SCK)

tp_device = machine.SPI.Device(spi_bus=spi_bus, freq=_TP_FREQ, cs=_TP_CS)
display_bus = lcd_bus.SPIBus(spi_bus=spi_bus, freq=_LCD_FREQ, dc=_LCD_DC, cs=_LCD_CS)
sd_card = machine.SDCard(spi_bus=spi_bus, cs=_SD_CS, freq=_SD_FREQ)

import st7796

display = st7796.ST7796(display_bus, _WIDTH, _HEIGHT, ...)

display.set_power(1)
display.init()
display.set_backlight(100)

touch_panel = xpt2046.XPT2046(tp_device)
if not touch_panel.is_calibrated:
    touch_panel.calibrate()

import task_handler

th = task_handler.TaskHandler()

# your LVGL code here....

```

tetopik, 2025-06-07:
Hi, thanks for the quick answer.

Please do believe me, Kevin. 

Following is the code I use:
```py
def display_init():
    from machine import freq
    freq(240_000_000)

    DSP_HOST = const(1)
    DSP_MOSI = const(13)
    DSP_MISO = const(12)
    DSP_SCK  = const(14)
    DSP_DC   = const(2)
    DSP_BLK  = const(21)
    DSP_CS   = const(15)
    DSP_FREQ = const(24_000_000)
    
    TCH_HOST = const(2)
    TCH_MOSI = const(32)
    TCH_MISO = const(39)
    TCH_SCK  = const(25)
    TCH_CS   = const(33)
    TCH_IRQ  = const(36)
    TCH_FREQ = const(2_000_000)
    
    SDC_HOST = const(3)
    SDC_MOSI = const(23)
    SDC_MISO = const(19)
    SDC_SCK  = const(18)
    SDC_CS   = const(5)
    SDC_FREQ = const(1_000_000)

    from machine import SoftSPI, SPI, SDCard
    dsp_spi = SPI.Bus(host=DSP_HOST, mosi=DSP_MOSI, miso=DSP_MISO, sck=DSP_SCK)
    tch_spi = SPI.Bus(host=TCH_HOST, mosi=TCH_MOSI, miso=TCH_MISO, sck=TCH_SCK)
    sdc_spi = SPI.Bus(host=SDC_HOST, mosi=SDC_MOSI, miso=SDC_MISO, sck=SDC_SCK)
    
    from lcd_bus import SPIBus as LCDBus
    dsp_bus = LCDBus(spi_bus=dsp_spi, dc=DSP_DC, cs=DSP_CS, freq=DSP_FREQ)
    tch_bus = SPI.Device(spi_bus=tch_spi, freq=TCH_FREQ, cs=TCH_CS)
    
    try:
        import os, vfs
        sd = SDCard(spi_bus=sdc_spi, cs=SDC_CS, freq=SDC_FREQ)
        vfs.mount(sd, "/sd")
        print('SDCard Mounted')
        print(os.listdir('/'))
    except Exception as e:
        print(f'SDCard Error: {e}')
    
    import ili9341
    display = ili9341.ILI9341(
        data_bus = dsp_bus,
        display_width  = 240,
        display_height = 320,
        backlight_pin  = DSP_BLK,
        backlight_on_state = ili9341.STATE_PWM,
        color_space = lv.COLOR_FORMAT.RGB565,
        color_byte_order = ili9341.BYTE_ORDER_BGR,
        rgb565_byte_swap = True)
    display.init(1)
    
    from xpt2046 import XPT2046
    indev = XPT2046(tch_bus)
    if not indev.is_calibrated:
        display.set_backlight(100)
        indev.calibrate()
    display.set_rotation(lv.DISPLAY_ROTATION._270)

    from task_handler import TaskHandler
    th = TaskHandler()
    
    return display, indev
```

Here's the schematics:

`DISPLAY`
[image]

`TOUCH`
[image]

`SDCARD`
[image]

tetopik, 2025-06-08:
Where is the `_TCH_CS` being passed on that code?

kdschlosser, 2025-06-08:
Add this code.. after you initialize the SoftSPI for the touch panel.

```
machine.Pin(_TCH_CS, machine.Pin.OUT, value=0)
```

kdschlosser, 2025-06-08:
just to let you know the "_" for the constants at the top needs to be there to make them actual constant. That means they mimic how `#define` works in C code. That is if you are using `const` from the micropython module. Using this uses no additional memory which is why it is nice to use. It allows you to place all of those integer numbers that never change at the top f the script so they are easy to locate should a change need to be made.

kdschlosser, 2025-06-08:
I have not updated the SoftSPI in MicroPython so it functions like the hardware SPI driver. In Vanilla MicroPython the hardware SPI driver doesn't allow you to have more then one device attached to a bus and it doesn't internally handle the CS line. I have modified the hardware SPI so it handles those things and I also modified the SDCard driver as well so you are able to share the SPI bus that an SDCard reader is attached to. In vanilla MicroPython you cannot do that.

tetopik, 2025-06-08:
It works, thanks man.
And also many thanks for the advice.

closed 2025-06-08 by tetopik

tetopik, 2025-06-09:
Hi, Kevin. I'm just curious, noob or even a silly question, why don't some mods into the micropython firmware you made that looks pretty promising for the future micropython development in general, not being pushed to the vanilla micropython itself, why just "here"?