lvgl-micropython/lvgl_micropython #378: Fix included: misplaced super.__init__() in keypad_framework.py seems to break (SDL) keyboard

ThomasFarstrike, 2025-06-08:
**Describe the bug**
SDL Keyboard doesn't work, in that the UI doesn't respond to it, even after setting the group etc.

**Expected behavior**
Typing in a textarea with an physical SDLKeyboard in Unix/Linux should input the typed characters.

After searching for a while, this seems to fix it:

```
diff --git a/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py b/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py
index 3b9b0d5..f1c2b1e 100644
--- a/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py
+++ b/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py
@@ -7,6 +7,7 @@ import display_driver_framework
 
 class KeypadDriver(_indev_base.IndevBase):
     def __init__(self):  # NOQA
+        super().__init__()
         if not lv.is_initialized():
             lv.init()
 
@@ -40,7 +41,6 @@ class KeypadDriver(_indev_base.IndevBase):
         indev_drv.enable(True)  # NOQA
         self._indev_drv = indev_drv
 
-        super().__init__()
         self._set_type(lv.INDEV_TYPE.KEYPAD)  # NOQA
 
     def _get_key(self):

```

It's a bit strange to me to see that super().__init__() being done so late in the __init__() function, and I guess it was overriding the other things that had already been done or something...

ThomasFarstrike, 2025-06-08:
Also see my sdl_keyboard.py where I ensured support for a regular QUERTY keyboard:

https://github.com/MicroPythonOS/lvgl_micropython/blob/main/api_drivers/common_api_drivers/indev/sdl_keyboard.py

kdschlosser, 2025-06-10:
This should be fixed now. So I am going to close this issue.