lvgl-micropython/lvgl_micropython #380: Argument list too long when compiling

phemmer, 2025-06-08:
**Describe the bug**
I'm not sure who is responsible for this issue. There's multiple projects involved here, and I'm not sure how all the interdependencies work out, or how to reproduce the issue outside of lvgl_micropython. So apologies if this is not the right place.

```
-- Build files have been written to: /home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3
Running make in directory /home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3
Executing "make -j 18 all"...
make[1]: Entering directory '/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3'
[  1%] Built target memory.ld                                                                                                                                           
make[1]: warning: -j18 forced in submake: resetting jobserver mode.
Partition table binary generated. Contents:
*******************************************************************************
[  2%] Building C object esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp_fsm/ulp_macro.c.obj    
GEN /home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/genhdr/mpversion.h
[  2%] Built target BUILD_VERSION_HEADER
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,2408K,
vfs,data,fat,0x26a000,5720K,
[  2%] Built target micropy_extmod_btree
*******************************************************************************
[  2%] Generating ../../genhdr/qstr.i.last         
make[3]: /bin/sh: Argument list too long
make[3]: *** [esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT.dir/build.make:684: genhdr/qstr.i.last] Error 127
make[2]: *** [CMakeFiles/Makefile2:7566: esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
...
make[1]: Leaving directory '/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3'
make[1]: *** [Makefile:136: all] Error 2
make failed with exit code 2, output of the command is in the /home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/log/idf_py_stderr_output_1936461 and /home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/log/idf_py_stdout_output_1936461
```

And if we look at `build.make:684`, we see:
```
	cd /home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/esp-idf/main && /home/phemmer/.espressif/python_env/idf5.4_py3.13_env/bin/python /home/phemmer/git/lvgl_micropython/lib/micropython/py/makeqstrdefs.py pp /home/phemmer/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -E output /home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/genhdr/qstr.i.last cflags -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython -I/home/phemmer/git/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/phemmer/git/lvgl_micropython/ext_mod/spi3wire -I/home/phemmer/git/lvgl_micropython/ext_mod/spi3wire/include -I/home/phemmer/git/lvgl_micropython/ext_mod/lcd_bus -I/home/phemmer/git/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/esp_lcd/rgb/include -I/home/phemmer/git/lvgl_micropython/lib/micropython -I/home/phemmer/git/lvgl_micropython/lib -I/home/phemmer/git/lvgl_micropython/lib/lvgl -I/home/phemmer/git/lvgl_micropython/lib/micropython/shared/tinyusb -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32 -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3 -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3 -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include_private -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__mdns/include -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__mdns/private_include -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__tinyusb/src -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__tinyusb/src -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__tinyusb/src/device -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__tinyusb/lib/networking -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/app_trace/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/app_trace/private_include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/app_trace/port/include -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/app_update/include -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bootloader_support/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bootloader_support/private_include -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/include/esp32c3/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/common/osi/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/common/api/include/api -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/common/btc/profile/esp/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/common/hci_log/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/home/phemmer/git/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/ho [... cut ...]
* FLASH:  quad OR octal

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: Fedora 42

**Build Command**

```shell
python make.py esp32 BOARD=ESP32_GENERIC_S3 DISPLAY=ili9341 INDEV=gt911
```

One thing I'm not sure if relevant, is that I updated the micropython submodule to current master (5f058e). This was to pull in the fixes addressed in https://github.com/micropython/micropython/pull/17269.

kdschlosser, 2025-06-08:
The problem is being caused by something in MicroPython. I am not sure what it is. The other thing is you have no need to use the master branch as the fixes you are referencing are only for GCC 15.1 and not the xtensa compiler which is what the ISP-IDF uses.

You can see the xtensa compiler being used in the command you posted that is too long..

`/home/phemmer/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc`

The issue could be relevant. I would do a fresh clone of this repo and try and compile again without changing anything and see if it compiles properly.

phemmer, 2025-06-08:
I definitely need to update the submodule. Here's what happens when it's on f498a16 (the default)

```
Compiling mpy-cross....
make: Entering directory '/home/phemmer/git/lvgl_micropython/lib/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
GEN build/genhdr/moduledefs.split
GEN build/genhdr/moduledefs.collected
Module registrations not updated
GEN build/genhdr/root_pointers.split
GEN build/genhdr/root_pointers.collected
Root pointer registrations not updated
CC ../py/emitinlinethumb.c
../py/emitinlinethumb.c:153:9: error: initializer-string for array of ‘unsigned char’ truncates NUL terminator but destination lacks ‘nonstring’ attribute (4 chars into 3 available) [-Werror=unterminated-string-initialization]
153 |     {0, "r0\0"},
|         ^~~~~~
../py/emitinlinethumb.c:154:9: error: initializer-string for array of ‘unsigned char’ truncates NUL terminator but destination lacks ‘nonstring’ attribute (4 chars into 3 available) [-Werror=unterminated-string-initialization]
154 |     {1, "r1\0"},
|         ^~~~~~
```
^ repeated a few dozen times for various files/lines

kdschlosser, 2025-06-09:
GCC is being used for the creation of the qstr file.

kdschlosser, 2025-06-09:
hey slow down there speedy.

you can also do the following

```
export CFLAGS=-Wno-unterminated-string-initialization
```

prior to you running the build command.

froginfog, 2025-06-09:
I also encountered the same problem.
What I don't understand is that I used the same command without any problem yesterday, but it suddenly stopped working today.

froginfog, 2025-06-10:
> it has to do with the version of GCC you are running that causes the issue. I am going to add the flag needed to the compiler arguments to fix the issue with he compiler tripping up. I am also going to inspect and see what some of the includes are being added so many times.

We appreciate your response and your work on this project. We'd be grateful if this matter could be addressed as soon as possible.

kdschlosser, 2025-06-10:
@froginfog 

clone the repo again and build it again I added the compiler flags needed to stop this from happening. Hopefully it will work like it should.