lvgl-micropython/lvgl_micropython #381: JC3248W535 - axs15231b - rotation not possible + spi pin numbers wrong

de-dh, 2025-06-09:
**Describe the bug**
Rotation with axs15231b  on JC3248W535 board does not work.

I used straga's build instructions and his ```lv_config.py + test.py``` file [from this folder](https://github.com/straga/micropython_lcd/tree/master/device/JC3248W535) to build the firmware for a JC3248W535 board.

```
# build
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16

# flash
python3 -m esptool --chip esp32s3 -p /dev/tty.usbmodem11101 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m --erase-all 0x0 lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-16.bin
```

It worked as expected in standard portrait mode after I modified spi initialization.
On startup, this error was produced:
```shell
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lv_config.py", line 73, in <module>
ValueError: 4 pins needed for quad SPI not 2
```

So i changed 
```python
spi_bus = SPI.Bus(
    host=1,   # SPI2_HOST
    mosi=21,  # PIN_NUM_QSPI_DATA0
    miso=48,  # PIN_NUM_QSPI_DATA1
    sck=47,   # PIN_NUM_QSPI_PCLK
    quad_pins=(40, 39),  # PIN_NUM_QSPI_DATA2, PIN_NUM_QSPI_DATA3
)
```
to 
```python
spi_bus = SPI.Bus(
    host=1,   # SPI2_HOST
    mosi=21,  # PIN_NUM_QSPI_DATA0
    miso=48,  # PIN_NUM_QSPI_DATA1
    sck=47,   # PIN_NUM_QSPI_PCLK
    quad_pins=(21,  48, 40, 39),  # PIN_NUM_QSPI_DATA2, PIN_NUM_QSPI_DATA3
)
```
It seems that ```mosi ```and ```miso``` pins also have to be provided as an argument for ```quad_pins```.

But this is not the main issue. I would like to use the display in landscape mode.
Using ``` display.set_rotation(lv.DISPLAY_ROTATION._90)``` produces a not implemented error:

```shell
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lv_config.py", line 116, in <module>
  File "axs15231b.py", line 46, in set_rotation
NotImplementedError: 
```

<details>

<summary>lv_config.py</summary>

```python
   
from micropython import const # NOQA

import lcd_bus # NOQA
from machine import SPI, Pin # NOQA

import lvgl as lv # NOQA

WIDTH = const(480)
HEIGHT = const(320)

_WIDTH = const(480)
_HEIGHT = const(320)

_BUFFER_SIZE = _WIDTH * _HEIGHT * 2

## Panel

# LCD_QSPI_HOST           (SPI2_HOST)     SPI2_HOST=1, ok

# PIN_NUM_QSPI_CS         (GPIO_NUM_45) ok
# PIN_NUM_QSPI_PCLK       (GPIO_NUM_47) ok

# PIN_NUM_QSPI_DATA0      (GPIO_NUM_21) ok
# PIN_NUM_QSPI_DATA1      (GPIO_NUM_48) ok
# PIN_NUM_QSPI_DATA2      (GPIO_NUM_40) ok
# PIN_NUM_QSPI_DATA3      (GPIO_NUM_39) ok

# PIN_NUM_QSPI_RST        (GPIO_NUM_NC) NC
# PIN_NUM_QSPI_DC         (GPIO_NUM_8) ok
# PIN_NUM_QSPI_TE         (GPIO_NUM_38)
# PIN_NUM_QSPI_BL         (GPIO_NUM_1) ok

## Touch

# I2C_NUM                     (I2C_NUM_0) ok
# I2C_CLK_SPEED_HZ            400000 ok

# PIN_NUM_QSPI_TOUCH_SCL  (GPIO_NUM_8) ok
# PIN_NUM_QSPI_TOUCH_SDA  (GPIO_NUM_4) ok
# PIN_NUM_QSPI_TOUCH_RST  (-1)
# PIN_NUM_QSPI_TOUCH_INT  (-1)

'''
# SPI bus config
spi_bus = SPI.Bus(
    host=1,   # SPI2_HOST
    mosi=21,  # PIN_NUM_QSPI_DATA0
    miso=48,  # PIN_NUM_QSPI_DATA1
    sck=47,   # PIN_NUM_QSPI_PCLK
    quad_pins=(40, 39),  # PIN_NUM_QSPI_DATA2, PIN_NUM_QSPI_DATA3
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=8,  # PIN_NUM_QSPI_DC 8
    cs=45,  # PIN_NUM_QSPI_CS
    freq=40000000,
    spi_mode=0,
    quad=True
)
'''

spi_bus = SPI.Bus(
    host=1,   # SPI2_HOST
    mosi=21,  # PIN_NUM_QSPI_DATA0
    miso=48,  # PIN_NUM_QSPI_DATA1
    sck=47,   # PIN_NUM_QSPI_PCLK
    quad_pins=(21, 48, 40, 39),  # PIN_NUM_QSPI_DATA2, PIN_NUM_QSPI_DATA3
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=8,  # PIN_NUM_QSPI_DC 8
    cs=45,  # PIN_NUM_QSPI_CS
    freq=40000000,
    spi_mode=0,
    quad=True
)

import axs15231b

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

display = axs15231b.AXS15231B(
    display_bus,
    WIDTH,
    HEIGHT,
    frame_buffer1=fb1,
    frame_buffer2=fb1,
    backlight_pin=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
    backlight_on_state=axs15231b.STATE_PWM
)

#display.set_rotation(lv.DISPLAY_ROTATION._0)

print('Hello LCD')
print(f"Display size: {WIDTH}x{HEIGHT}")

display.set_power(True)
display.set_backlight(100)
# PWM frequency for backlight
#display._backlight_pin.freq(2000)

display.init()  # use 1 type config

display.set_rotation(lv.DISPLAY_ROTATION._90)  # NOQA

# TOUCH

# I2C_NUM_0
# I2C_CLK_SPEED_HZ            400000

# PIN_NUM_QSPI_TOUCH_SCL  (GPIO_NUM_8)
# PIN_NUM_QSPI_TOUCH_SDA  (GPIO_NUM_4)

import axs15231
from i2c import I2C

i2c_bus = I2C.Bus(host=1, sda=4, scl=8)
touch_i2c = I2C.Device(i2c_bus, axs15231.I2C_ADDR, axs15231.BITS)
indev = axs15231.AXS15231(touch_i2c, debug=False)
indev.enable_input_priority()

print('is_calibrate is', indev.is_calibrated)
```

</details>

Shouldn't this work by now since @kdschlosser wrote a new bus / display driver? Do I maybe have to change something during diplay initialization?

kdschlosser, 2025-06-09:
That display is software rotation and it uses quad SPI not RGB so there is no software rotation that is available yet. It is in the works just not finished with it.

alvins82, 2025-06-25:
@kdschlosser wanted to ask if there was any update on the rotation (sorry don't want to bug you)?

kdschlosser, 2025-06-25:
I have put some work in with it. Not done yet. It is complex because I don't want to have repeat code so handling the different LCD busses and how they write the data is what is making it complicated. The other challenge is handling the different sized frame buffers and correctly passing the information needed to perform a rotation. As an example. RGB and DSI interfaces have full frame buffers that gets sent to the display I already have code in place that will take a partial buffer that is 1/10th the size of the full frame buffer and it will rotate that buffer and place is properly within the full frame buffer. When dealing with say in I8080 or SPI display we have GRAM available so only partial buffers need to be used. So I cannot use the coordinates that are given by LVGL when performing the rotation and placement because we are not placing a partial buffer into a full buffer, instead we are placing a partial buffer into another partial buffer. The kicker is that there are some I8080 and SPI displays that have funky ways they handle the buffers. 

The goal is to make sure it is easy to use so all of the mechanics of this needs to be kept behind the scenes. We also want to make sure there is as minimal performance impact as possible so that means offloading the copying of the buffer to the second core of the ESP32 if one is available. If one is not available that make it once again more of a challenge. I don't want to make the code base really large by having 2 separate ways of dealing with the rotation based on the number of cores. If I do that it will be a maintenance headache.

straga, 2025-09-08:
@de-dh  look -> https://github.com/lvgl-micropython/lvgl_micropython/discussions/161

straga, 2025-09-09:
@de-dh - https://github.com/lvgl-micropython/lvgl_micropython/pull/451 Not work software rotation