lvgl-micropython/lvgl_micropython #383: Development board restarts when using fs_driver

kk8900, 2025-06-12:
**Description:**

Hello!

Thank you very much for developing LVGL-Micropython. It has been extremely helpful for us! I am a beginner and have encountered some issues while using it, and I hope you can provide some assistance.

I recently recompiled the LVGL-Micropython firmware for the ESP32-S3-BOX3 and obtained the LVGL 9.2.2 + Micropython 1.25 version. However, I noticed some differences compared to the previous firmware (LVGL 9.2.2 + Micropython 1.24.1), and I suspect that this might be due to the upgrade to Micropython 1.25. When I attempt to use `fs_driver`, the development board automatically restarts. Simply importing `fs_driver` causes the restart.

**Compilation Command:**

```bash
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=ili9341 INDEV=gt911
```

**Related Code:**

Here is the code I used for testing:

```python
import lcd_bus
from micropython import const
import machine

# Display settings
_WIDTH = const(320)  # Width of the display screen
_HEIGHT = const(240)  # Height of the display screen
_BL = const(47)  # Backlight pin
_DC = const(4)  # Data/Command control pin

_MOSI = const(6)  # SPI MOSI pin
_MISO = const(-1)  # SPI MISO pin (not used)
_SCK = const(7)  # SPI SCK pin
_HOST = const(1)  # SPI2

_LCD_CS = const(5)  # LCD chip select pin
_LCD_FREQ = const(40000000)  # SPI communication frequency for the LCD

# Touch settings
_TP_HOST = const(0)
_TP_SCL = const(18)
_TP_SDA = const(8)
_TP_FREQ = const(400000)

_BUFFER_SIZE = const(30720)

# Create an SPI bus object
spi_bus = machine.SPI.Bus(
    host=int(_HOST),     # Explicitly convert to integer
    mosi=int(_MOSI),
    miso=int(_MISO),
    sck=int(_SCK)   
)

# Create an SPI communication object for the display
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import ili9341  # NOQA
import lvgl as lv  # NOQA

ili9341.ILI9341._ORIENTATION_TABLE = (
    0x0,
    0x60,
    0x40 | 0x80,
    0x20 | 0x40 | 0x80
)

# Create a display object
display = ili9341.ILI9341(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import i2c
import task_handler
import gt911, pointer_framework

# Initialize the display
display.init(2)

# Define I2C communication parameters for the touch screen
i2c_bus = i2c.I2C.Bus(host=_TP_HOST, scl=_TP_SCL, sda=_TP_SDA, freq=_TP_FREQ, use_locks=False)
touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=0x5D, reg_bits=gt911.BITS)

# Create a touch screen device object
indev = gt911.GT911(touch_dev, startup_rotation=pointer_framework.lv.DISPLAY_ROTATION._180)

# GT911 needs to set the touch screen size
if indev.hw_size != (320, 240):
    fw = indev.firmware_config
    fw.width = 320
    fw.height = 240
    fw.save()
    del fw

# Turn on the screen backlight
display.set_backlight(1)

# Rotate the display
display.set_rotation(lv.DISPLAY_ROTATION._180)

# Create a task handler instance
th = task_handler.TaskHandler()

# The following code causes the development board to restart
#import fs_driver

#fs_drv = lv.fs_drv_t()
#fs_driver.fs_register(fs_drv, 'S')

scrn = lv.screen_active()  # Get the currently active screen object
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)  # Set the screen background color to black

# Create a slider
slider = lv.slider(scrn)
slider.set_size(200, 50)  # Set the slider size
slider.center()  # Center the slider on the screen

# Create a label
label = lv.label(scrn)
label.set_text('HELLO LVGL_MICROPYTHON!')  # Set the label content
label.align(lv.ALIGN.CENTER, 0, -50)  # Align the label to the center of the screen and offset upwards by 50
```

**Environment Information:**

- Development board model: ESP32-S3-BOX3
- Firmware version: LVGL 9.2.2 + Micropython 1.25

**Expected Result:**

I need to use custom fonts via `fs_driver`, but if `fs_driver` is not available, are there any alternative methods?

Thank you very much for your help!

kdschlosser, 2025-06-12:
when the board resets you should see the core dump data in the serial output. I need you to decode the backtrace data to know what is going on. You can read about how to do that here...

 https://jsimoesblog.wordpress.com/2022/11/04/decoding-esp32-back-trace/

kk8900, 2025-06-20:
> when the board resets you should see the core dump data in the serial output. I need you to decode the backtrace data to know what is going on. You can read about how to do that here...
> 
> https://jsimoesblog.wordpress.com/2022/11/04/decoding-esp32-back-trace/

I have learned how to parse the backtrace data and now I have successfully parsed it out:
~/lvgl_micropython/lib/esp-idf$ xtensa-esp32-elf-addr2line -pfiaC -e ~/micropython.elf 0x400858d6:0x3ffc0760 0x401a15f4:0x3ffc0800 0x401a7365:0x3ffc0870 0x401d0470:0x3ffc0890 0x401d04ad:0x3ffc08d0 0x401a7365:0x3ffc08f0 0x401b925d:0x3ffc0910 0x40089475:0x3ffc09c0 0x400894c5:0x3ffc0a00 0x400834fd:0x3ffc0a30 0x4008763d:0x3ffc0a50 0x0004001e:0x00000000
0x400858d6: mp_execute_bytecode at /home/le/lvgl_micropython/lib/micropython/py/vm.c:267
0x401a15f4: fun_bc_call at /home/le/lvgl_micropython/lib/micropython/py/objfun.c:295
0x401a7365: mp_call_function_n_kw at /home/le/lvgl_micropython/lib/micropython/py/runtime.c:726
0x401d0470: mp_call_method_self_n_kw at /home/le/lvgl_micropython/lib/micropython/py/objboundmeth.c:70
0x401d04ad: bound_meth_call at /home/le/lvgl_micropython/lib/micropython/py/objboundmeth.c:83
0x401a7365: mp_call_function_n_kw at /home/le/lvgl_micropython/lib/micropython/py/runtime.c:726
0x401b925d: machine_timer_isr at /home/le/lvgl_micropython/lib/micropython/ports/esp32/machine_timer.c:153
0x40089475: timer_process_alarm at /home/le/lvgl_micropython/lib/esp-idf/components/esp_timer/src/esp_timer.c:435
0x400894c5: timer_alarm_handler at /home/le/lvgl_micropython/lib/esp-idf/components/esp_timer/src/esp_timer.c:481
0x400834fd: timer_alarm_isr at /home/le/lvgl_micropython/lib/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:184
0x4008763d: _xt_lowint1 at /home/le/lvgl_micropython/lib/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x0004001e: ?? ??:0

I asked AI and it seems to be caused by a timer interrupt. Could you please help me figure out how to solve it?

kdschlosser, 2025-06-20:
@straga

are you able to import the `fs_driver` module???

kdschlosser, 2025-06-20:
And that is with the latest code from the repo? I will make the assumption it is. So there is something going on with the build. a setting that is not being done right for the board. More than likely it is for the flash, On the S3 since it is 16mb of flash it could be octal and not quad. On the WROVER it could be the amount of flash.

I am going to need to know the3 exact model number of both MCU's being used to be able to determine what needs to be done.

kk8900, 2025-06-22:
> Thank you [@straga](https://github.com/straga) for testing that out for me. My development stuff is still in a box somewhere because of remodeling part of my home so I am not able to test these kinds of things at the moment. I appreciate you for taking the time to do that for me.

I’m sorry that my feedback was wrong. After I retried to compile the firmware, I found that everything worked well and the previous error had been completely solved. Thanks to your efforts, I can easily get started with lvgl-micropython as a beginner. It has greatly lowered the learning threshold and brought me a lot of convenience.

kk8900, 2025-06-22:
In China, due to the particularity of the network environment, accessing GitHub is not always smooth, which may trigger some unpredictable errors during the compilation process.

kdschlosser, 2025-06-22:
I am glad the issue is sorted out. I did make a large change with the upgrade of the MicroPython version being used and also the ESP-IDF version being used. I am sure there are some issues that have yet to be discovered. I am happy that the problem has been corrected..

There is still an issue. The build script is what is does the collecting of the requirements that are needed to compile the binding. If a requirement did not get collected properly the build script should have exited with an error. If it did not do this then there is a bug that needs to be located and addressed. If you can get the problem to occur again except capture the build output then I can use that to find out what is happening and be able to have the build script react properly. 

The way the build system currently works is it looks at the exit code when running subprocesses. It is possible that a subprocess is having an issue but for some reason it doesn't exit with an error code. It might not even output the error to the stderr console buffer. I may have to look for specific phrases in the stdout console buffer to be able to detect that a problem has occurred to have the build script error and terminate. 

If you would be willing to capture the console output when building and when the error occurs I can apply the code needed to have the build reset it's state and terminate to allow the user to try it again and have the script try to collect what is needed again.

kk8900, 2025-06-22:
> I am glad the issue is sorted out. I did make a large change with the upgrade of the MicroPython version being used and also the ESP-IDF version being used. I am sure there are some issues that have [...]

There seemed to be no errors during the build process. The firmware was output normally, and the flashing process also had no errors.

kk8900, 2025-06-22:
> I am glad the issue is sorted out. I did make a large change with the upgrade of the MicroPython version being used and also the ESP-IDF version being used. I am sure there are some issues that have [...]

I've encountered another issue. When using the ESP32-S3-BOX-3 development board and flashing the latest compiled firmware (LVGL 9.2 + MPY 1.25), pressing the RST button causes the device to remain in bootloader mode instead of entering Flash boot mode. It only occasionally enters Flash boot mode after many attempts. This problem did not exist with the previous firmware version (LVGL 9.2 + MPY 1.24.1).

Another development board (ESP32-WROVER-B + ST7735) does not experience this issue when flashed with the latest compiled firmware (LVGL 9.2 + MPY 1.25).

The address of the ESP32-S3-BOX-3 development board is:https://github.com/espressif/esp-box

kdschlosser, 2025-06-22:
from looking at the diagram for the S3 box thing the reset button is not connected to the typical "reset" pin which is labeled as EN. The reset button is connected to GPIO6. The boot button is connected to GPIO0 which is where it should be. The EN pin is connected to the main power on the board so the only way to reset the thing is to disconnect the power and reconnect it while holding the boot button to change the boot mode.

The other thing is there is no UART to USB IC, what is being used in the built in USB CDC connection. To use this connection requires a couple of extra compile commands.

`--enable-uart-repl=n --enable-jtag-repl=n --enable-cdc-repl=y`

those build commands need to be added when compiling.

In order to make use of the CDC in MicroPython and to allow user access to the CDC USB a software CDC Stack is being used. When using the software CDC stack the hardware stack built into the ESP MCU gets disabled. Because of it being disabled being able to reboot or set the boot mode over the USB no longer functions. This is caused by those features not being set up in the MicroPython code. I have not yet explored using the CDC functionality nor have I messed with the C code for the USB CDC stack in MicroPython to see if I can correct the issues with it.

Try compiling and flashing vanilla MicroPython and see if the behavior is the same. If it is the same then the issue lies in MicroPython and they would be the ones that would need to locate and correct the problem.