lvgl-micropython/lvgl_micropython #385: about machine.sdcard

froginfog, 2025-06-16:
I have a question. When I use machine.SDCard() to initialize the SD card, because of my development board, I have to do machine.SDCard(slot=0, width=1, sck=11, cmd=10, data=(9,)). This is fine under the official firmware of micropython, but it will report an error in lvgl_micropython: extra keyword arguments given.
I tried to add the parameter：python3 make.py esp32 ...... CONFIG_SOC_SDMMC_USE_GPIO_MATRIX=y when compiling, but it still reports an error. Why is this happening?

kdschlosser, 2025-06-16:
adding `CONFIG_SOC_SDMMC_USE_GPIO_MATRIX` to the build command is not going to alter anything. That config option is not something that is able to be user set. It is dictated by the series of ESP32 MCU you are using.

Vanilla MicroPython was written so that an SDCard reader was not allowed to share an SPI Bus with any other device. This causes an issue because a lot of display boards are made with the SDCard reader either on the same SPI Bus as the display or on the same bus as the touch or all 3 share the same bus. That would render the SDCard unable to be used if that was the case. So I had to modify the SDCard driver in MicroPython so it would allow sharing of the SPIBus...

The new SDCard driver now takes these arguments..

```
SDCard(
        width=1,
        cd=-1,
        wp=-1,
        slot=1,
        cmd=-1,
        clk=-1,
        data_pins=None,
        spi_bus=None,
        cs=-1,
        freq=20000000
)
```

The SDCard driver is actually 2 drivers in one. It is an MMC driver and also and SD driver. These are 2 different types of card readers.

`width`, `slot`, `cmd`, `clk` and `data_pins` only apply if you have an MMC card reader. If you have an SD reader then you will use the `spi_bus` parameter. `cd`, `wp`, `cs` and `freq` are used for both types of card readers.

In order for us to share an SPI bus I had to write the code so an SPI bus object could be made and that object could then be passed to other "device" drivers. In this case the SDCard driver is a device.

```
machine.SPI.Bus(
    host,
    sck,
    mosi=-1,
    miso=-1,
    dual_pins=None,
    quad_pins=None,
    octal_pins=None
)
```

if the bus that needs to be created uses dual SPI, quad SPI or octal SPI you pass a tuple of pin numbers to `dual_pins`, `quad_pins` or `octal_pins` and do not pass anything to `miso` or `mosi`.

The instance of `machine.SPI.Bus` is what you pass to the `spi_bus` parameter of the SDCard driver and optionally supply the `cs`, `cd`, `wp` and `freq` parameters. 

cs = cable select
cd = card detect
wp = write protect
freq = spi frequency

The driver internally handles the cs pin where as in Vanilla MicroPython it did not. So that is all taken care so no need to mess about with changing the cs pin state.

kdschlosser, 2025-06-17:
it's `machine.SPI.Bus`

froginfog, 2025-06-20:
Thanks, this problem is solved

closed 2025-06-20 by froginfog