lvgl-micropython/lvgl_micropython #387: ESP32-S3-LCD-1.47 Horizontal

dcmcshan, 2025-06-27:
I have a ESP32-S3-LCD-1.47 working, but I would like to use it horizontally.  Is there a rotation parameter I can give init?

`#!/usr/bin/env python3
"""
ESP32-S3 Waveshare Display Module (No Hardware Reset)
Handles display initialization without disconnecting USB
"""

import lvgl as lv
import lcd_bus
import st7789
import machine
import time

# Display configuration - try Arduino's original PORTRAIT mode
WIDTH = 172   # Arduino's original width
HEIGHT = 320  # Arduino's original height
OFFSET_X = 34  # Arduino's original X offset
OFFSET_Y = 0   # Arduino's original Y offset

# SPI Pin Configuration (matching Arduino demo)
SPI_MOSI = 45
SPI_SCK = 40
SPI_MISO = 13  # Not used for display but needed for SPI bus

# Display Pin Configuration (matching Arduino demo)
PIN_DC = 41
PIN_CS = 42
PIN_RESET = 39
PIN_BACKLIGHT = 48

def init_display():
    # Initialize LVGL first (like MCT)
    lv.init()
    
    # Initialize SPI bus using parameters with reset fallback
    try:
        spi_bus = machine.SPI.Bus(
            host=1,
            mosi=SPI_MOSI,
            miso=SPI_MISO,
            sck=SPI_SCK
        )
    except Exception as e:
        print(f"SPI Bus initialization failed: {e}")
        print("Resetting machine...")
        machine.reset()
    
    # Initialize LCD bus using the SPI bus
    display_bus = lcd_bus.SPIBus(
        spi_bus=spi_bus,
        dc=PIN_DC,
        cs=PIN_CS,
        freq=80_000_000  # 80MHz like Arduino
    )
    
    # Allocate framebuffers
    buf_size = WIDTH * HEIGHT * 2 // 10  # Arduino formula: /20 but we use /10 for stability
    fb1 = display_bus.allocate_framebuffer(buf_size, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
    fb2 = display_bus.allocate_framebuffer(buf_size, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
    
    # Initialize ST7789 display with Arduino-compatible settings
    display = st7789.ST7789(
        data_bus=display_bus,
        display_width=WIDTH,
        display_height=HEIGHT,
        frame_buffer1=fb1,
        frame_buffer2=fb2,
        reset_pin=PIN_RESET,
        reset_state=st7789.STATE_LOW,
        power_on_state=st7789.STATE_HIGH,
        backlight_pin=PIN_BACKLIGHT,
        backlight_on_state=st7789.STATE_HIGH,
        offset_x=OFFSET_X,  # Critical 34-pixel offset from Arduino demo
        offset_y=OFFSET_Y,
        color_space=lv.COLOR_FORMAT.RGB565,
        color_byte_order=st7789.BYTE_ORDER_BGR,  # Keep BGR as it was working
        rgb565_byte_swap=True
    )
    
    # Initialize display (like MCT) - let the driver handle initialization
    display.init()
    display.set_power(True)
    
    # Set up PWM backlight (matching Arduino: 1000Hz)
    backlight_pwm = machine.PWM(machine.Pin(PIN_BACKLIGHT), freq=1000)
    backlight_pwm.duty_u16(int(65535 * 0.9))  # 90% brightness like Arduino
    
    return display

def test_display():
    """Test the display with a simple red screen to fix colors first"""
    print("Initializing display with Arduino-compatible settings...")
    display = init_display()
    
    # Get active screen (like MCT)
    screen = lv.screen_active()
    
    # Make the entire screen PURPLE
    screen.set_style_bg_color(lv.color_make(255, 0, 255), 0)  # Purple (Red + Blue)
    screen.set_style_bg_opa(lv.OPA.COVER, 0)
    
    # Add marker at (0,0) - top-left corner
    marker_0_0 = lv.obj(screen)
    marker_0_0.set_size(20, 20)  # 20x20 pixel marker (bigger)
    marker_0_0.set_pos(0, 0)
    marker_0_0.set_style_bg_color(lv.color_make(255, 255, 0), 0)  # Yellow
    marker_0_0.set_style_bg_opa(lv.OPA.COVER, 0)
    marker_0_0.set_style_border_width(0, 0)
    marker_0_0.set_style_radius(0, 0)
    
    # Add marker at (172-20, 320-20) - bottom-right corner (adjusted for marker size)
    marker_max = lv.obj(screen)
    marker_max.set_size(20, 20)  # 20x20 pixel marker (bigger)
    marker_max.set_pos(WIDTH - 20, HEIGHT - 20)  # Bottom-right corner
    marker_max.set_style_bg_color(lv.color_make(0, 255, 255), 0)  # Cyan
    marker_max.set_style_bg_opa(lv.OPA.COVER, 0)
    marker_max.set_style_border_width(0, 0)
    marker_max.set_style_radius(0, 0)
    
    # Force refresh (like MCT)
    screen.invalidate()
    lv.refr_now(None)
    
    print("Display test loaded - purple screen with corner markers (portrait 172x320)")
    print("Entire screen should be PURPLE with:")
    print("- YELLOW marker (20x20) at (0,0) - top-left corner")
    print("- CYAN marker (20x20) at (152,300) - bottom-right corner")
    print("Settings applied:")
    print("- 80MHz SPI frequency")
    print("- BGR color order with byte swap")
    print("- 1000Hz PWM backlight")
    print(f"- Offset X={OFFSET_X}, Y={OFFSET_Y}")
    print("Press Ctrl+C to exit")
    
    # Keep the display running with LVGL updates
    try:
        while True:
            lv.tick_inc(50)  # 50ms tick increment
            lv.task_handler()  # Process LVGL tasks
            time.sleep_ms(50)
    except KeyboardInterrupt:
        print("\nExiting...")

if __name__ == "__main__":
    test_display() `

dcmcshan, 2025-06-28:
(I am nit working on the touch version of above, which is apparently different still...)

kdschlosser, 2025-06-29:
you will need to adjust the orientation table

jobbyworld, 2025-11-15:
HI, i have same card (WAVESHARE ESP32-S3-Touch-LCD-1.47)

Controller chip | ESP32-S3
LCD type | ISP
Screen resolution | 172 × 320
Display chip | JD9853
Touch chip | AXS5106L

I try to build with this instructions :

```
git clone https://github.com/lvgl-micropython/lvgl_micropython.git
cd lvgl_micropython
docker run -it -v .:/micropython --name micropython ubuntu
apt-get update
apt-get install -y gcc g++ make automake python3 git gcc-arm-none-eabi libusb-1.0-0 python3-venv cmake
cd /micropython
python3 make.py esp32 clean \
  --flash-size=4 \
  BOARD=ESP32_GENERIC_S3 \
  DISPLAY=jd9853 \
  INDEV=axs5106
```

but build failed at 

```
[ 61%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/micropython/lib/lvgl/src/core/lv_obj_event.c.obj
/micropython/ext_mod/lcd_bus/esp32_src/i2c_bus.c: In function 'mp_lcd_i2c_bus_make_new':
/micropython/ext_mod/lcd_bus/esp32_src/i2c_bus.c:79:9: error: 'mp_lcd_i2c_bus_obj_t' {aka 'struct _mp_lcd_i2c_bus_obj_t'} has no member named 'port'
79 |     self->port = bus->port;
```

Can you share your build commands ?

Thanks !

kdschlosser, 2025-11-15:
I just pushed a commit that should now fix the issue with the I2C. Well not really fixed it but I disabled the I2C display bus so it's no longer conflicting with the I2C that is built into Micropython.

So perform a fresh clone and then build. 

you also don't need the `clean` in the build command. it handles it automatically in the build command. The `clean` command does a deep scrubbing by deleting the build folders and running clean when compiling mpy_cross. Unless something gets fantastically boogered up there usually is no reason to add the `clean` command