lvgl-micropython/lvgl_micropython #388: How do I build lvgl

Tinyu-Zhao, 2025-06-30:
I manually added [this folder](https://github.com/lvgl/lv_micropython/tree/master/ports/unix/variants/lvgl) to the lib/micropython/ports/unix/variants/ path, and tried to build `python3 make.py macOS VARIANT=lvgl` using this command

It builds successfully, but it never import asyncio.

[image]

kdschlosser, 2025-07-01:
This is your build command..

```
python3 make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer
```

asyncio is not available for the macOS port. See [here](https://github.com/lvgl-micropython/lvgl_micropython/discussions/384#discussioncomment-13600857)

closed 2025-07-01 by Tinyu-Zhao

kdschlosser, 2025-07-01:
Yes I have injected code into the unix port that creates a thread to execute code at a given interval for the macOS port. I tried to use macOS's Timer API without any success which is why I went the route of using a thread to do it.

The trick with using machine.Timer on any of the ports for MicroPython is to not block the main thread using looping code. You MUST let the main thread check the REPL for input. This is where tasks that gets scheduled get run. They also get run when calling `time.sleep` and `time.sleep_ms` but that doesn't happen across all ports which is why it is best to have it handled at the REPL.

IF for some reason you have a need to have looping code in the main thread I added a function that will execute scheduled tasks. Calling `lcd_bus._pump_main_thread` will run any tasks that are scheduled. This function should not be called from any thread other than the main thread. If there is looping code that run in the main thread it should be called from inside of that loop. If it is not called then the task handler that updates the display will not run properly.

Tinyu-Zhao, 2025-07-02:
> Yes I have injected code into the unix port that creates a thread to execute code at a given interval for the macOS port. I tried to use macOS's Timer API without any success which is why I went the [...]

Thank you very much for your help in bringing a great experience to macOS users.