lvgl-micropython/lvgl_micropython #391: lvgl_micropython is different than micropython

quantrpeter, 2025-07-02:
Hi
   I got CYD (cheap yellow display). I did build the micropython (the official one) and below code work. When I use the same step to build lvgl_micropython (this repo). It failed to run, why?

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't create 'SPI' instances
```

```
from machine import Pin, SPI
from ili9341 import Display, color565
import time
 
# Initialize SPI for ILI9341 display
display_spi = SPI(1, baudrate=60000000, sck=Pin(14), mosi=Pin(13))
display = Display(display_spi, dc=Pin(2), cs=Pin(15), rst=Pin(15), width=320, height=240, rotation=90)
 
# Turn on backlight
backlight = Pin(21, Pin.OUT)
backlight.on()
 
# Clear display (white background)
display.clear(color565(255, 255, 255))
 
# Animation parameters
rect_width = 30
rect_height = 30
x = 0
y = 0
dx = 6  # Increased speed
dy = 6  # Increased speed
rect_color = color565(255, 0, 0)  # Red rectangle
bg_color = color565(255, 255, 255)  # White background
 
# Animation loop
while True:
    # Clear previous rectangle (minimize cleared area)
    display.fill_rectangle(x, y, rect_width, rect_height, bg_color)
     
    # Update position
    x += dx
    y += dy
     
    # Bounce off edges
    if x <= 0 or x >= 320 - rect_width:
        dx = -dx
    if y <= 0 or y >= 240 - rect_height:
        dy = -dy
     
    # Draw new rectangle
    display.fill_rectangle(x, y, rect_width, rect_height, rect_color)
     
    # Reduced delay for faster animation
    time.sleep(0.02)  # ~50 FPS
```

thanks
Peter

asherevan, 2025-07-02:
It’s better! Maybe check out [#193](https://github.com/lvgl-micropython/lvgl_micropython/issues/193) if you need to know how to use the new system?

kdschlosser, 2025-07-02:
The readme file contains all of this kind of information..

There had to be changes made to MicroPython in order to get things to work properly. MicroPython did not have SPI set up properly and because of that you could not share the same SPI bus between multiple devices without having to tear the bus down and recreate it if you wanted to switch to communicating to a new device. This is very inefficient and it would not work with an SPI LCD display or an SPI SDCard reader. It states right in the MicroPython documentation that you cannot have an SDCard reader in the same bus as any other device. This is not a limitation of the hardware but it a limitation caused by how MicroPython was written. The changes that I made were necessary to allow a user to be able to use a display, touch screen and SDCard that were all on the same bus. Each device on the bus is able to operate at different speeds and it also handles selecting the device using the cs wire. The user doesn't have to write any code to manage any of these things. It is all handled for them.

Another reason why it needed to change is because of the use of DMA memory. the cs pin MUST be handled by the driver because performing an SPI transfer is a non blocking call when needing to write data to an SPI display. The user would not know when to change the state of the cs wire and could end up with corrupted data if it doesn't change when the transfer is done sending.

It is also recommended to use an added i2c driver instead of machine.I2C as well... There are Python stub files on the root of the repository directory that show what the API is for the various pieces of MicroPython and also for things that have been added to MicroPython to allow this project to work properly. Read over the filed that end in `.pyi` to become familiar with what is added and changed.