lvgl-micropython/lvgl_micropython #394: unable to display image

quantrpeter, 2025-07-03:
hi
   screen just blank, black. i tried to convert png into binary format, just no luck. pleae help

```
import lcd_bus
from micropython import const
import machine
import ili9341
import lvgl as lv
from time import sleep

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(21)
_RST = const(17)
_DC = const(2)

_MOSI = const(13)
#_MISO = const(12)
_SCK = const(14)
_HOST = const(1)  # SPI2

_LCD_CS = const(15)
_LCD_FREQ = const(40000000)

#_TOUCH_CS = const(9)
#_TOUCH_FREQ = const(1000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    #miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RST,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=_BL,
    backlight_on_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)

display.set_power(True)
display.init(1)
display.set_color_inversion(False)
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

scrn = lv.screen_active()

img = lv.image(scrn)
img.set_src("colorful.png")
img.set_size(200, 200)
img.center()

import task_handler
task_handler.TaskHandler()

sleep(2)
```

quantrpeter, 2025-07-05:
If i change to this

```
from fs_driver import fs_register

fs_drv = lv.fs_drv_t()
fs_register(fs_drv, "S")

img = lv.image(scrn)
img.set_src("S:colorful.png")
img.set_size(128, 128)
img.center()

print('end')

# import task_handler
# task_handler.TaskHandler()
# sleep(2)

import utime as time
# time_passed = 1000
while True:
    # start_time = time.ticks_ms()
    time.sleep_ms(20)
    # lv.tick_inc(time_passed)
    lv.task_handler()
    # end_time = time.ticks_ms()
    # time_passed = time.ticks_diff(end_time, start_time)
```

then I see part of the image on screen, then crashed

```
/Users/peter/Desktop>mpremote reset && mpremote run image.py     
s1
s2
s3
s4
end
Traceback (most recent call last):
  File "<stdin>", line 94, in <module>
MemoryError: memory allocation failed, allocating 65539 bytes
```

kdschlosser, 2025-07-05:
also what esp32 are you using?
How much psram?

quantrpeter, 2025-07-05:
If i make the image smaller, change from 128x128 to 80x80, it can be displayed. But I tried to display 128x128 in wave share pico, it has 2MB ram, but still fail, 2MB ram should be enough for 128x128, don't understand why

kdschlosser, 2025-07-05:
OK so a C6 with no PSRAM (SPIRAM) is really not going to be a strong enough MCU to run this library. You are not going to have enough ram to load any images...

there is only 500K of ram available and after MicroPython loads you might only have 250k to 300k left available. between frame buffers and drivers and LVGL objects that much memory doesn't go very far.

use `gc.gc.mem_free` and `gc.mem_alloc` to check the amount of memory instead of `micropython.mem_info`. Let me know what those numbers are.

also Thonny doesn't play nice with this library I recommend using mpremote which is in `lib/micropython/tools`