lvgl-micropython/lvgl_micropython #395: ST7735 problem

quantrpeter, 2025-07-03:
hi
   must not wire issue, because i use micropython + ST7735.py, it works. But this repo can't. Don't know why

[image]

```
import lcd_bus
from micropython import const
import machine
from time import sleep
  
import st7735
import lvgl as lv

# display settings
_WIDTH = 128
_HEIGHT = 128
_BL = 19
_RST = 14
_DC = 15
  
_MOSI = 21
_MISO = 20
_SCK = 22
_HOST = 1  # SPI2
  
_LCD_CS = 18
_LCD_FREQ = 40000000
  
_OFFSET_X = 2
_OFFSET_Y = 2
  
print('s1');
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)
 
print('s2');
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)
  
print('s3');
display = st7735.ST7735(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
    reset_state=st7735.STATE_HIGH,
    backlight_on_state=st7735.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7735.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y
)

print('s4');
import task_handler
th = task_handler.TaskHandler()

# display.set_power(True)
display.init(1)
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xff0000), 0)

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, 50)

print('end')
while True:
    lv.task_handler()
    sleep(0.01)
```

kdschlosser, 2025-07-03:
the task handler needs to be after your initialize the display.

quantrpeter, 2025-07-04:
moved to the end, still the same

tetopik, 2025-07-04:
According to your `non-lvgl` code:
```py
def _reset( self ) :
    '''Reset the device.'''
    self.dc(0)
    self.reset(1)
    time.sleep_us(500)
    self.reset(0)
    time.sleep_us(500)
    self.reset(1)
    time.sleep_us(500)
```

These line of your code `reset_state=st7735.STATE_HIGH` should be `reset_state=st7735.STATE_LOW`

kdschlosser, 2025-07-04:
OK so here are some thing that I am seeing and I have questions about...

for the LVGL example you gave you have the spi bus speed set to 40mHz and when using that other code you have the SPI bus speed set to 20mHz. any reason as to why?

In the LVGL example you are initing the display as type B but in the other code you are initing the display as type r-red. is there a reason for that as well??

You gotta compare apples to apples with this kind of a thing.

I suggest that you decrease the SPI speed for the LVGL test down to 20mHz and that you init the display as type r-red..

```
display.init(st7735.TYPE_R_RED)
```

quantrpeter, 2025-07-05:
yeah, display.init(st7735.TYPE_R_RED) works. Thanks so much

closed 2025-07-05 by quantrpeter

kdschlosser, 2025-07-05:
No worries m8!

Just had to do some digging to find out what was causing the issue. I am glad it is sorted out and that it was something simple to solve.