lvgl-micropython/lvgl_micropython #400: How to swap image

quantrpeter, 2025-07-05:
Hi
   How to swap image, tried different ways but screen still not update

```
img = lv.image(scrn)
img.set_src("S:colorful.png")
img.set_size(128, 128)
img.center()

p4=Pin(20, Pin.IN, Pin.PULL_UP)  # Button pin

print("end")

while True:
    time.sleep_ms(20)
    lv.task_handler()
    sleep(0.2)
    if not p4.value():  # Button pressed
        print("Button pressed")
        sleep(0.1)  # Debounce delay
        img.set_src("S:blue.png")
        img.invalidate() 
        scrn.invalidate()
        lv.image.cache_drop("S:colorful.png")
```

thanks

kdschlosser, 2025-07-05:
you need to refresh the display by calling `lv.refr_now(lv.screen_active().get_display())` after you change out the images.

quantrpeter, 2025-07-06:
thanks so much, it works

closed 2025-07-06 by quantrpeter