lvgl-micropython/lvgl_micropython #404: add_event_cb seems not working

quantrpeter, 2025-07-10:
Hi
    When i click on the touch-LCD, I can see debug message, but the event registered by add_event_cb never fires

[image]

```
import lcd_bus
from micropython import const
import machine
import time  # Add time module for clock functionality
import xpt2046
import fs_driver

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(21)
_RST = const(17)
_DC = const(2)
 
_MOSI = const(13)
#_MISO = const(12)
_SCK = const(14)
_HOST = const(1)  # SPI2
 
_LCD_CS = const(15)
_LCD_FREQ = const(40000000)
 
_TOUCH_CS = const(33)
_TOUCH_FREQ = const(10_000_000)
 
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    #miso=_MISO,
    sck=_SCK
)
 
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)
 
import ili9341
import lvgl as lv
 
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RST,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=_BL,
    backlight_on_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)
 
display.set_power(True)
display.init(1)
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

spi_bus_touch = machine.SPI.Bus(
    host=2,
    mosi=32,
    miso=39,
    sck=25
)

touch_dev = machine.SPI.Device(
   spi_bus=spi_bus_touch,
   freq=_TOUCH_FREQ,
   cs=_TOUCH_CS,
)
 
indev = xpt2046.XPT2046(touch_dev,debug=True,startup_rotation=lv.DISPLAY_ROTATION._90)
# indev.calibrate()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFFFFFF), 0)  # White background

# SemiBlock label at the top
semiblock_label = lv.label(scrn)
semiblock_label.set_text("SemiBlock")
semiblock_label.align(lv.ALIGN.TOP_MID, -45, 20)
semiblock_label.set_style_text_color(lv.color_hex(0xFF80C0), 0)  # Pinkly blue color
semiblock_label.set_style_transform_scale(600, 0)  # Scale text to 200% (2x bigger)

import task_handler
th = task_handler.TaskHandler()

def event_cb(self,e):
	print("Clicked", e)
	btn = e.get_target_obj()
	label = btn.get_child(0)
	label.set_text(str(self.cnt))
      
btn = lv.button(scrn)
btn.set_size(100, 50)
btn.center()
btn.add_event_cb(event_cb, lv.EVENT.CLICKED, None)
btn.add_event_cb(event_cb, lv.EVENT.PRESSED, None)
label = lv.label(btn)
label.set_text("Click me!")
label.center()

while True:
    time.sleep_ms(250)
    lv.task_handler()
```

thanks
Peter

kdschlosser, 2025-07-10:
because you are not incrementing the tick time. `lv.tick_inc`

kdschlosser, 2025-07-10:
and a 250ms delay should not be done. You should ahve that set to 10 milliseconds at most and you should be passing 10 to `lv.tick_inc` when you call it. Calling that function needs to be done before you call `lv.task_handler`

kdschlosser, 2025-07-10:
Hold up.. I noticed that you are using  the `task_handler` module... You don't need to have the while loop at the end if you are using that...

kdschlosser, 2025-07-10:
You are also overwriting the events when you are registering them...

```

btn.add_event_cb(event_cb, lv.EVENT.CLICKED, None)
btn.add_event_cb(event_cb, lv.EVENT.PRESSED, None)
```

That's wrong.

This is how you should be doing it.

```
btn.add_event_cb(event_cb, lv.EVENT.CLICKED | v.EVENT.PRESSED, None)
```

kdschlosser, 2025-07-11:
you are not showing me any kind of error. just saying "it's not working" doesn't give me any information as to what is going on.

I see that you have debugging enabled for the touch driver. is it outputting anything when you touch the screen? If so what is it outputting?

Have you preformed a screen calibration at all. did that run successfully?

quantrpeter, 2025-07-11:
yes , in the original message, on the right hand side of the screen

[image]

kdschlosser, 2025-07-12:
IDK what you did to get the coordinates to be converted like they are because they are WAYYYYYY off.

Are you using the command that is output at the end of the build when you flash the firmware? If you are not I do recommend that you do because that flash command erases the ESP32 before writing new data to it.

If the NVS is not cleared it could be picking up bytes that have not been zeroed out and that is the reason why you are having issues.

I would flash the firmware again using the command that gets output at the end of the build and DO NOT perform a calibration and keep the debugging active for the touch driver. Give it a try and post post the logging output. Not a screen shot, copy and paste 10 lines of the output for me.

I noticed that you are setting the rotation before the touch driver is started.. That is a NO NO. You need to set the rotation after the touch driver has been started and also after you have done a calibration.

Use this code below to test it out.
copy the 10 lines of debugging output for me. 

Then uncomment the 2 lines for the calibration and run the script again. The calibration will run and it should function properly..
Once you have finished the calibration then you can touch the display some more and get 10 line of debugging output for me. 

```python
import lcd_bus
from micropython import const
import machine
import time  # Add time module for clock functionality
import xpt2046
import fs_driver

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(21)
_RST = const(17)
_DC = const(2)
 
_MOSI = const(13)
#_MISO = const(12)
_SCK = const(14)
_HOST = const(1)  # SPI2
 
_LCD_CS = const(15)
_LCD_FREQ = const(40000000)
 
_TOUCH_CS = const(33)
_TOUCH_FREQ = const(10_000_000)
 
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    #miso=_MISO,
    sck=_SCK
)
 
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)
 
import ili9341
import lvgl as lv
 
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RST,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=_BL,
    backlight_on_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)
 
display.set_power(True)
display.init(1)
display.set_backlight(100)

spi_bus_touch = machine.SPI.Bus(
    host=2,
    mosi=32,
    miso=39,
    sck=25
)

touch_dev = machine.SPI.Device(
   spi_bus=spi_bus_touch,
   freq=_TOUCH_FREQ,
   cs=_TOUCH_CS,
)
 
indev = xpt2046.XPT2046(touch_dev,debug=True,startup_rotation=lv.DISPLAY_ROTATION._90)

#if not indev.is_calibrated:
#    indev.calibrate()

display.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFFFFFF), 0)  # White background

# SemiBlock label at the top
semiblock_label = lv.label(scrn)
semiblock_label.set_text("SemiBlock")
semiblock_label.align(lv.ALIGN.TOP_MID, -45, 20)
semiblock_label.set_style_text_color(lv.color_hex(0xFF80C0), 0)  # Pinkly blue color
semiblock_label.set_style_transform_scale(600, 0)  # Scale text to 200% (2x bigger)

# import task_handler
# th = task_handler.TaskHandler()

def event_cb(self,e):
	print("Clicked", e)
	btn = e.get_target_obj()
	label = btn.get_child(0)
	label.set_text(str(self.cnt))
      
btn = lv.button(scrn)
btn.set_size(100, 50)
btn.center()
btn.add_event_cb(event_cb, lv.EVENT.CLICKED | lv.EVENT.PRESSED | lv.EVENT.RELEASED, None)
label = lv.label(btn)
label.set_text("Click me!")
label.center()

while True:
    time.sleep_ms(10)
    lv.tick_inc(10)
    lv.task_handler()
```

quantrpeter, 2025-07-12:
Thanks, when i tap on the button on LCD, it print. So the click event still hasn't been fired.

```
/Users/peter/workspace/cheap-yellow-display-micropython-lvgl-example>mpremote run touch.py
XPT2046_TP_DATA(x=2099, y=2150, z=1375)
XPT2046(raw_x=122, raw_y=167, x=-954, y=9271, state=PRESSED)
XPT2046_TP_DATA(x=2099, y=2208, z=1920)
XPT2046(raw_x=122, raw_y=172, x=-988, y=9577, state=PRESSED)
XPT2046_TP_DATA(x=2112, y=2208, z=1920)
XPT2046(raw_x=123, raw_y=172, x=-977, y=9491, state=PRESSED)
XPT2046_TP_DATA(x=2163, y=2176, z=1856)
XPT2046(raw_x=126, raw_y=169, x=-922, y=9036, state=PRESSED)
XPT2046_TP_DATA(x=2112, y=2176, z=1407)
XPT2046(raw_x=123, raw_y=169, x=-956, y=9302, state=PRESSED)
XPT2046(raw_x=123, raw_y=169, x=-956, y=9302, state=RELEASED)
XPT2046_TP_DATA(x=2048, y=2048, z=864)
XPT2046(raw_x=119, raw_y=159, x=-932, y=9035, state=PRESSED)
XPT2046(raw_x=119, raw_y=159, x=-932, y=9035, state=RELEASED)
XPT2046_TP_DATA(x=2048, y=2176, z=1375)
XPT2046(raw_x=119, raw_y=169, x=-1001, y=9655, state=PRESSED)
XPT2046(raw_x=119, raw_y=169, x=-1001, y=9655, state=RELEASED)
XPT2046_TP_DATA(x=2163, y=2048, z=1375)
XPT2046(raw_x=126, raw_y=159, x=-853, y=8417, state=PRESSED)
XPT2046_TP_DATA(x=2176, y=2048, z=1407)
XPT2046(raw_x=127, raw_y=159, x=-842, y=8331, state=PRESSED)
```

kdschlosser, 2025-07-12:
OK so you got an event and the crash happened in your callback. The calibration has worked. You can see it in the data...

this is what it was before calibration...

```
XPT2046(raw_x=122, raw_y=167, x=-954, y=9271, state=PRESSED)
```

and this is what it is after calibration....

```
XPT2046(raw_x=128, raw_y=160, x=123, y=156, state=PRESSED)
```

You are not performing an erase of the esp32 when you are flashing the firmware. So I am going to ask once again what are you doing to flash the firmware to the ESP32?

Because you are not erasing the ESP32 prior to flashing the firmware is the reason why the `indev.is_calibrated` if not working correctly. It's because there is data that is stored from a prior calibration you did that was a bad calibration caused by you having the display rotated before you initialized the touch device.

quantrpeter, 2025-07-13:
hi
   i use estool.py erase the whole flash, but the result is the same

```
esptool.py erase_flash
esptool.py --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m --erase-all 0x0 build/lvgl_micropy_ESP32_GENERIC-4.bin
```

thanks

quantrpeter, 2025-07-13:
github copilot helped me to fix, I changed "def event_cb(self, e)" to "def event_cb(e)" then it works

```
import lcd_bus
from micropython import const
import machine
import time  # Add time module for clock functionality
import xpt2046
import fs_driver

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(21)
_RST = const(17)
_DC = const(2)
 
_MOSI = const(13)
#_MISO = const(12)
_SCK = const(14)
_HOST = const(1)  # SPI2
 
_LCD_CS = const(15)
_LCD_FREQ = const(40000000)
 
_TOUCH_CS = const(33)
_TOUCH_FREQ = const(10_000_000)
 
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    #miso=_MISO,
    sck=_SCK
)
 
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)
 
import ili9341
import lvgl as lv
 
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RST,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=_BL,
    backlight_on_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)
 
display.set_power(True)
display.init(1)
display.set_backlight(100)

spi_bus_touch = machine.SPI.Bus(
    host=2,
    mosi=32,
    miso=39,
    sck=25
)

touch_dev = machine.SPI.Device(
   spi_bus=spi_bus_touch,
   freq=_TOUCH_FREQ,
   cs=_TOUCH_CS,
)
 
indev = xpt2046.XPT2046(touch_dev,debug=True,startup_rotation=lv.DISPLAY_ROTATION._90)

if not indev.is_calibrated:
	indev.calibrate()

display.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFFFFFF), 0)  # White background

# SemiBlock label at the top
semiblock_label = lv.label(scrn)
semiblock_label.set_text("SemiBlock")
semiblock_label.align(lv.ALIGN.TOP_MID, -45, 20)
semiblock_label.set_style_text_color(lv.color_hex(0xFF80C0), 0)  # Pinkly blue color
semiblock_label.set_style_transform_scale(600, 0)  # Scale text to 200% (2x bigger)

def event_cb(e):
	print("Clicked", e)
	btn = e.get_target_obj()
	label = btn.get_child(0)
	label.set_text("Peter")
      
btn = lv.button(scrn)
btn.set_size(100, 50)
btn.center()
btn.add_event_cb(event_cb, lv.EVENT.CLICKED | lv.EVENT.PRESSED | lv.EVENT.RELEASED, None)
label = lv.label(btn)
label.set_text("Click me!")
label.center()

# import task_handler
# th = task_handler.TaskHandler()

while True:
    time.sleep_ms(10)
    lv.tick_inc(10)
    lv.task_handler()
```

closed 2025-07-13 by quantrpeter