lvgl-micropython/lvgl_micropython #405: Support axs5106l

quantrpeter, 2025-07-11:
Hello
    Will we support axs5106l touch driver? https://www.waveshare.com/1.47inch-touch-lcd.htm
thanks

kdschlosser, 2025-07-11:
I did some looking and I am not able to find that IC anywhere except being used on that display.. The drivers they have listed for use do not exist either..

for arduino they say to use the `axs51061_device` library... it doesn't exist
for esp32 they say to use the esp_lcd_touch_axs5106 component... it doesn't exist

I cannot write a driver for it if IO am not able to locate a datasheet that has a list of the registers that need to be used to collect the touch input data from the IC.

kdschlosser, 2025-07-11:
OK I added it.

col135, 2025-07-14:
I just tried the axs5106 touch component but get the following error(s) using it on the WaveShare ESP32-C6-Touch-LCD-1.47. Hopefully the error output it is helpful?

Traceback (most recent call last):
  File "<stdin>", line 154, in <module>
  File "pointer_framework.py", line 107, in _read
  File "axs5106.py", line 61, in _get_coords
  File "axs5106.py", line 21, in _read_reg
  File "i2c.py", line 129, in read
  File "i2c.py", line 63, in readfrom
TypeError: can't convert memoryview to int

kdschlosser, 2025-07-15:
all done.

kdschlosser, 2025-07-15:
it's not able to find the device on the bus.

paste the output from running this code...

```python
i2c_bus = I2C.Bus(host=1, sda=18, scl=19)

print(i2c_bus.scan())
```

col135, 2025-07-15:
GPIO 20 (touch reset) is most likely in the wrong state, make sure you have the latest commit 10ebfbd
for axs5106.py that puts the reset line in the correct state (high to run). Manually setting GPIO 20 high and removing 'reset_pin=20' will confirm.

kdschlosser, 2025-07-15:
I was told that it needed to be flipped so I updated the code. Do I have to change it back now?

quantrpeter, 2025-07-18:
> The latest commit [10ebfbd](https://github.com/lvgl-micropython/lvgl_micropython/commit/10ebfbddf1a201291bd4c568b366023e3a68ee59) is correct but I suspect quantrpeter has not updated as it was only 20HRS ago you committed that change.

Hi @col135 , would you mind to post an example here please?

quantrpeter, 2025-07-18:
@kdschlosser I found out it is I2C host = 0

i2c_bus = I2C.Bus(host=0, sda=18, scl=19)

quantrpeter, 2025-07-18:
new problem, indev.calibrate() does show correct screen but touching is working find

kdschlosser, 2025-07-18:
Is there a reason why you are performing the calibration? Touchscreen calibrations are usually only done on resistive touchscreens.

kdschlosser, 2025-07-19:
flash the firmware using the command seen at the end of the build and change your code so the indev bits read as you see below.
see if that sorts out your calibration issue.

```python
display.set_power(True)
display.init()
display.set_color_inversion(True)
# display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

i2c_bus = I2C.Bus(host=0, sda=18, scl=19)
touch_i2c = I2C.Device(i2c_bus, axs5106.I2C_ADDR, axs5106.BITS)
indev = axs5106.AXS5106(touch_i2c, reset_pin=20)

if not indev.is_calibrated:
    indev.calibrate()

# rotation is done after the calibration has been run
display.set_rotation(lv.DISPLAY_ROTATION._90)
```

You can use this code if you don't want to flash the firmware again but once the calibration is done you will want to change it to the code seen in the first code block.

```python
display.set_power(True)
display.init()
display.set_color_inversion(True)
# display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

i2c_bus = I2C.Bus(host=0, sda=18, scl=19)
touch_i2c = I2C.Device(i2c_bus, axs5106.I2C_ADDR, axs5106.BITS)
indev = axs5106.AXS5106(touch_i2c, reset_pin=20)

# if not indev.is_calibrated:
indev.calibrate()

# rotation is done after the calibration has been run
display.set_rotation(lv.DISPLAY_ROTATION._90)
```

kdschlosser, 2025-07-19:
also if a special tweaking need to be done to the driver you can always "monkey patch" the driver so no actual files in the binding get altered. This is something you do in your code...

```python
import axs5106

class AXS5106(axs5106.AXS5106):
    
    def _calc_coords(self, x, y):
        if self.is_calibrated:
            cal = self._cal
            x = int(round(x * cal.alphaX + y * cal.betaX + cal.deltaX))
            y = int(round(x * cal.alphaY + y * cal.betaY + cal.deltaY))

            if cal.mirrorX:
                x = self._orig_width - x - 1
            if cal.mirrorY:
                y = self._orig_height - y - 1
        else:
            if self._startup_rotation == lv.DISPLAY_ROTATION._0:
                x = self._orig_width - x - 1
            else:
                if (
                    self._startup_rotation == lv.DISPLAY_ROTATION._180 or  # NOQA
                    self._startup_rotation == lv.DISPLAY_ROTATION._270  # NOQA
                ):
                    x = self._orig_width - x - 1
                    y = self._orig_height - y - 1
  
                if (
                    self._startup_rotation == lv.DISPLAY_ROTATION._90 or  # NOQA
                    self._startup_rotation == lv.DISPLAY_ROTATION._270  # NOQA
                ):
                    x, y = self._orig_height - y - 1, x

        return x, y

indev = AXS5106(touch_i2c, reset_pin=20)
...
```

That will reverse the X orientation so long as there is no startup rotation that has been set in the touch driver and there is no calibration that has been done.

kdschlosser, 2025-07-19:
another way of handling it is doing the following...

```python

display.set_power(True)
display.init()
display.set_color_inversion(True)
display.set_backlight(100)

from touch_cal_data import TouchCalData
touch_cal = TouchCalData('touch_cal')

i2c_bus = I2C.Bus(host=0, sda=18, scl=19)
touch_i2c = I2C.Device(i2c_bus, axs5106.I2C_ADDR, axs5106.BITS)
indev = axs5106.AXS5106(touch_i2c, reset_pin=20, touch_cal=touch_cal)

if not indev.is_calibrated:
    touch_cal.mirrorX = False
    touch_cal.mirrorY = True

    touch_cal.alphaX = 1.0
    touch_cal.betaX = 0.0
    touch_cal.deltaX = 0.0

    touch_cal.alphaY = 0.0
    touch_cal.betaY = 1.0
    touch_cal.deltaY = 0.0

    touch_cal.save()
```

That will also work, However it does have additional overhead compared to the monkey patching way of doing it.

quantrpeter, 2025-08-11:
These code was working in last month. But it failed for latest source now, please help

[image]

```
import lcd_bus
from micropython import const
import machine
from time import sleep
import jd9853
import axs5106
import lvgl as lv
from i2c import I2C

lv.init()

# display settings
_WIDTH = 172
_HEIGHT = 320
_BL = 23
_RST = 22
_DC = 15

_MOSI = 2 #SDA
_MISO = 5
_SCK = 1  # SCL
_HOST = 1  # SPI2

_LCD_CS = 14
_LCD_FREQ = 2000000
_TOUCH_FREQ = 2000000
_TOUCH_CS = 21

_OFFSET_X = 34
_OFFSET_Y = 0

print('s1');
spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    #miso=_MISO,
    sck=_SCK
)

print('s2');
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

print('s3');
display = jd9853.JD9853(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
    reset_state=jd9853.STATE_LOW,
    backlight_on_state=jd9853.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=jd9853.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y
)

print('s4');

display.set_power(True)
display.init()
display.set_color_inversion(True)
# display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

i2c_bus = I2C.Bus(host=0, sda=18, scl=19)
touch_i2c = I2C.Device(i2c_bus, axs5106.I2C_ADDR, axs5106.BITS)
indev = axs5106.AXS5106(touch_i2c,
                        debug=True,
                        startup_rotation=lv.DISPLAY_ROTATION._90,
                        reset_pin=20
                        )
# if not indev.is_calibrated:
#indev.calibrate()

# scrn = lv.screen_active()

import utime as time
time_passed = 1000

while True:
    start_time = time.ticks_ms()
    time.sleep_ms(1)  # sleep for 1 ms
    lv.tick_inc(time_passed)
    lv.task_handler()
    end_time = time.ticks_ms()
    time_passed = time.ticks_diff(end_time, start_time)
```