lvgl-micropython/lvgl_micropython #407: Build Failing

JonHylands, 2025-07-11:
I have a Waveshare 1.46" round ESP32-S3 board (https://www.waveshare.com/esp32-s3-touch-lcd-1.46b.htm?sku=29565). I'm trying to do the build (I've built normal ESP32-based MicroPython on this machine before), and I'm getting a strange error:

`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --enable-cdc-repl=y --enable-uart-repl=n --flash-size=16 DISPLAY=spd2010 INDEV=spd2010t IMU=qmi8658c EXPANDER=tca9554`

The error is:

```
make[1]: *** [Makefile:136: all] Error 2
make failed with exit code 2, output of the command is in the /home/jon/dev/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_102327 and /home/jon/dev/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_102327
```

The first file has the following contents:

```
jon@jon-Oryx-Pro:~$ more /home/jon/dev/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_102327
make[1]: warning: -j22 forced in submake: resetting jobserver mode.
make[3]: /bin/sh: Argument list too long
make[3]: *** [esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT.dir/build.make:799: genhdr/qstr.i.last] Error 127
make[2]: *** [CMakeFiles/Makefile2:6072: esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:136: all] Error 2
```

Gemini seems to think that the -j22 is probably the issue, but I haven't been able to figure out how to change that.

kdschlosser, 2025-07-11:
the actual error is 

`make[3]: /bin/sh: Argument list too long`

I need you to attach those build output files so I can see what is being called and to figure out how I can fix the problem.

kdschlosser, 2025-07-11:
We need to disable parallel building to see what is being run when the failure happens. With parallel building things don't necessarily get output to the console in the order in which they happen so it makes it hard to locate the area that is having the problem. 

what OS are you compiling on?

JonHylands, 2025-07-11:
I'm running Xubuntu 22.04.

I see the following in that file:

`def get_espidf():

    cmd = [
        [
            'git', 'submodule', 'update', '--init',
            f'--jobs {os.cpu_count()}', '--depth=1',
            '--', 'lib/esp-idf'
        ],
        ['cd lib/esp-idf'],
        [
            'git', 'submodule', 'update', '--init', '--depth=1',
            f'--jobs {os.cpu_count()}', '--',
            'components/bt/host/nimble/nimble',
            'components/esp_wifi',
            'components/esptool_py/esptool',
            'components/lwip/lwip',
            'components/mbedtls/mbedtls',
            'components/bt/controller/lib_esp32',
            'components/bt/controller/lib_esp32c3_family'
        ]
    ]
    print()
    print(f'collecting ESP-IDF v5.4.0')
    print('this might take a while...')
    result, _ = spawn(cmd, spinner=True)
    if result != 0:
        sys.exit(result)
`

should I comment out both of them?

JonHylands, 2025-07-11:
[output.txt](https://github.com/user-attachments/files/21190967/output.txt)

JonHylands, 2025-07-12:
Looks like it worked, at least in terms of finishing the build - yay!

I flashed the board successfully - does the lvgl build normally work with rshell?

JonHylands, 2025-07-14:
Actually, never mind - its working fine now. No idea what was happening before...

closed 2025-07-14 by JonHylands

JonHylands, 2025-07-14:
I think you misunderstood something, or perhaps posted in the wrong place - for me, this is running in MicroPython on an ESP32-S3 display board.

JonHylands, 2025-07-14:
Right - rshell is a Linux tool that allows you to get a "remote" shell on a MicroPython board, so you can copy files back and forth and so on. It also gives you direct access to the MicroPython REPL on the board. I'm having issues connecting to the ESP32-S3 board I flashed with the above build on my laptop - my desktop has no issues talking to it. When I plug the board in to my laptop, its coming up with a product id of 1001, which is what it has when it is in boot mode. When I plug it into my desktop, it comes up with 4001, which is the proper mode so you can get access to the REPL and such. I think it has something to do with lvgl-micropython, because if I flash the same board with stock ESP32-S3 MicroPython, it works fine.

JonHylands, 2025-07-14:
> But isn’t rshell for the pyboard?

You can use rshell with almost any MicroPython board - I use it with
stm32's, esp 32's, Pico's, etc.