lvgl-micropython/lvgl_micropython #409: lvgl designer

quantrpeter, 2025-07-15:
Hi
   lvgl designer such as square line studio can only export c code, how can it work with this project? I mean work with micropython?
   Is there any lvgl designer can export micropython code?
thx
Peter

kdschlosser, 2025-07-15:
square line studio does output MicroPython code I had thought. Maybe they removed that ability? IDK.. To my knowledge that is/was the only one. 

as a side note, the code it generated wasn't really all that good anywho.

quantrpeter, 2025-07-18:
not really working for square line studio to export the code to micropython. It has a fatal error but doesn't tell which line cause it. I think it is better to design my own LVGL designer

[image]

```
# This file was generated by SquareLine Studio
# SquareLine Studio version: SquareLine Studio 1.5.3
# LVGL version: 9.1.0
# Project name: SquareLine_Project

import lvgl as lv
# import ui_images

dispp = lv.display_get_default()
theme = lv.theme_default_init(dispp, lv.palette_main(lv.PALETTE.BLUE), lv.palette_main(lv.PALETTE.RED), False, lv.font_default())
dispp.set_theme(theme)

def ui_theme_set(idx):
   return

def SetFlag( obj, flag, value):
    if (value):
        obj.add_flag(flag)
    else:
        obj.remove_flag(flag)
    return

_ui_comp_table = {}
_ui_comp_prev = None
_ui_name_prev = None
_ui_child_prev = None
_ui_comp_table.clear()

def _ui_comp_del_event(e):
    target = e.get_target()
    _ui_comp_table[id(target)].remove()

def ui_comp_get_child(comp, child_name):
    return _ui_comp_table[id(comp)][child_name]

def ui_comp_get_root_from_child(child, compname):
    for component in _ui_comp_table:
        if _ui_comp_table[component]["_CompName"]==compname:
            for part in _ui_comp_table[component]:
                if id(_ui_comp_table[component][part]) == id(child):
                    return _ui_comp_table[component]
    return None
def SetBarProperty(target, id, val):
   if id == 'Value_with_anim': target.set_value(val, lv.ANIM.ON)
   if id == 'Value': target.set_value(val, lv.ANIM.OFF)
   return

def SetPanelProperty(target, id, val):
   if id == 'Position_X': target.set_x(val)
   if id == 'Position_Y': target.set_y(val)
   if id == 'Width': target.set_width(val)
   if id == 'Height': target.set_height(val)
   return

def SetDropdownProperty(target, id, val):
   if id == 'Selected':
      target.set_selected(val)
   return

def SetImageProperty(target, id, val, val2):
   if id == 'Image': target.set_src(val)
   if id == 'Angle': target.set_rotation(val2)
   if id == 'Zoom': target.set_scale(val2)
   return

[... 565 lines cut ...]
# COMPONENTS

# COMPONENTS
ui____initial_actions0 = lv.obj()

ui_Screen1 = lv.obj()
SetFlag(ui_Screen1, lv.obj.FLAG.SCROLLABLE, False)

ui_Button1 = lv.button(ui_Screen1)
ui_Button1.set_width(100)
ui_Button1.set_height(50)
ui_Button1.set_align( lv.ALIGN.CENTER)
SetFlag(ui_Button1, lv.obj.FLAG.SCROLLABLE, False)
SetFlag(ui_Button1, lv.obj.FLAG.SCROLL_ON_FOCUS, True)
ui_Button1.set_style_bg_color(lv.color_hex(0x00FF50), lv.PART.MAIN | lv.STATE.PRESSED )
ui_Button1.set_style_bg_opa(255, lv.PART.MAIN| lv.STATE.PRESSED )

ui_Label1 = lv.label(ui_Screen1)
ui_Label1.set_text("text")
ui_Label1.set_width(lv.SIZE_CONTENT)	# 1
ui_Label1.set_height(lv.SIZE_CONTENT)   # 1
ui_Label1.set_align( lv.ALIGN.CENTER)

lv.screen_load(ui_Screen1)
```

closed 2025-07-20 by quantrpeter