lvgl-micropython/lvgl_micropython #412: Build not working

chicco-carone, 2025-07-19:
I was following [this](https://gist.github.com/invisiblek/08fc71de9ffc7af3575036102f5b7c76) guide to try and set up lvgl and micropython for my waveshare esp32 c6 1.47 in but I can't manage to build micropython. I tried both on my main pc with arch and on a clean ubuntu lxc on proxmox without success.

<details>
  <summary>Here are the full logs from the error</summary>

```text
-- Check for working C compiler: /root/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /root/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c6
"idf_component.yml" file was found for components:
/root/lvgl_micropython/lib/micropython/ports/esp32/main
However, the component manager is not enabled.
-- Project sdkconfig file /root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/sdkconfig
Loading defaults file /root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/sdkconfig.combined...
/root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/sdkconfig.combined:104 CONFIG_ESP32_WIFI_IRAM_OPT was replaced with CONFIG_ESP_WIFI_IRAM_OPT
/root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/sdkconfig.combined:105 CONFIG_ESP32_WIFI_RX_IRAM_OPT was replaced with CONFIG_ESP_WIFI_RX_IRAM_OPT
/root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/sdkconfig.combined:165 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
warning: unknown kconfig symbol 'XTAL_FREQ_AUTO' assigned to 'y' in /root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/sdkconfig.combined
-- Adding linker script /root/lvgl_micropython/lib/esp-idf/components/riscv/ld/rom.api.ld
-- Configuring incomplete, errors occurred!
CMake Error at /root/lvgl_micropython/lib/esp-idf/tools/cmake/component.cmake:317 (message):
Include directory
'/root/lvgl_micropython/lib/esp-idf/components/mbedtls/mbedtls/include' is
not a directory.
Call Stack (most recent call first):
/root/lvgl_micropython/lib/esp-idf/tools/cmake/component.cmake:500 (__component_add_include_dirs)
/root/lvgl_micropython/lib/esp-idf/components/mbedtls/CMakeLists.txt:24 (idf_component_register)
make: Leaving directory '/root/lvgl_micropython/lib/micropython/ports/esp32'
cmake failed with exit code 1, output of the command is in the /root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/log/idf_py_stderr_output_5702 and /root/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C6/submodules/log/idf_py_stdout_output_5702
make: *** [Makefile:115: submodules] Error 2
```
</details>

I have a clean install of esp-idf following the official docs and the os is completely clean so i'm not sure what could cause this issue.

kdschlosser, 2025-07-19:
No the build is working as it should. You didn't read the directions in that link.  No where in the directions did it tell you to install the ESP-IDF. It says to clone the repo and run the build command.

chicco-carone, 2025-07-19:
I removed the ~/esp folder where esp-idf is installed and tried to run it again and it gave the same error.

kdschlosser, 2025-07-20:
you also need to remove the `~/.espressif` folder as well as clean any environment variables that the ESP-IDF may have set.

chicco-carone, 2025-07-20:
I recreated the lxc with only python make cmake and gcc and it compiled. Thanks for the help

closed 2025-07-20 by chicco-carone

tetopik, 2025-07-28:
Hi, @kdschlosser. 

Since you're making compile mp as easier as possible, how to include another module like this [umodbus](https://github.com/brainelectronics/micropython-modbus) to bake it into the firmware? Is it the same as vanilla mp to copy it manually to ports/esp32/modules folder?