lvgl-micropython/lvgl_micropython #413: SemiBlock UI Builder

quantrpeter, 2025-07-20:
I am building the LVGL UI Builder. I will release the platform here after I finish, so everyone can comment.

[image]

[image]

quantrpeter, 2025-08-27:
Dear Vincent
      Semiblock Micropython part is published in https://semiblock.hkprog.org, but it is only for limited guess, give me your gmail i can allow you access. For the LVGL Builder, we are working on it.
thanks
Peter

Vincent1-python, 2025-08-27:
lllzzzyyy1586@gmail.com Thank you very much.