lvgl-micropython/lvgl_micropython #415: Failed to generate - No such file

PeeTeeBee, 2025-07-23:
### Followed instructions per readme

1. git clone https://github.com/lvgl-micropython/lvgl_micropython
2. cd lvgl_micropython
3. python3 make.py rp2 BOARD=**RPI_PICO2_W** DISPLAY=ILI9488 INDEV=XPT2046 clean**

### Error Output

Pico W Wi-Fi build support available.
Including User C Module(s) from ../../../../ext_mod/micropython.cmake
/usr/bin/python3: can't open file '/home/pee/Pico/lvgl_micropython/gen/_api_gen_mpy.py': [Errno 2] No such file or directory
OUTPUT:
RESULT: 2
CMake Error at /home/pee/Pico/lvgl_micropython/ext_mod/lvgl/micropython.cmake:45 (message):
Failed to generate
/home/pee/Pico/lvgl_micropython/lib/micropython/ports/rp2/build-RPI_PICO2_W/submodules/lv_mp.c
Call Stack (most recent call first):
/home/pee/Pico/lvgl_micropython/ext_mod/micropython.cmake:14 (include)
/home/pee/Pico/lvgl_micropython/lib/micropython/py/usermod.cmake:56 (include)
CMakeLists.txt:92 (include)
-- Configuring incomplete, errors occurred!
make: Leaving directory '/home/pee/Pico/lvgl_micropython/lib/micropython/ports/rp2'
make: *** [Makefile:77: submodules] Error 1

### Comment:

I did note the board is not directly described in the readme however I do see it is included in vgl_micropython/lib/micropython/ports/rp2/

kdschlosser, 2025-07-23:
Support for the RP2 is something that will hopefully happen in the future. I have not had anyone that is knowledgeable about the RP2 and is also willing to help write the drivers that are needed to get it up and running.

PeeTeeBee, 2025-07-24:
Much appreciated, but no I am good, I have something working, not planning an elaborate UI.

I would like to have a play with the graphics library at some stage. Is there a specific board with wifi/bluetooth and display that is well supported. Thanks.