lvgl-micropython/lvgl_micropython #421: SSD1306 monochrome display supports

tetopik, 2025-07-29:
Hi, Kevin. 

I can see that the ssd1306 driver already there in the display option list, so I guess I can give it a try since there are likely no one has ever tested it out yet.

Could you please, Kevin give me an initialization example for the specific display so I can run the setup and giving back whatever the results would be.

Let's say the display connected via i2c bus with `addr=0x3c`.

Any help would be so much be appreciated.

kdschlosser, 2025-07-30:
OK I have updated the display driver to fix some things..

here is some sample code that you put into the main.py file and then upload the file to the ESP32 using a script like mpremote or ampy. mpremote is included in MicroPython and that is located in the `lib/micropython/tools/mpremote` folder. 

You will need to set `_LCD_SDA` and `_LCD_SCL` as well as `_LCD_BKL` if you have a backlight pin. everything else should be good to go.

Give it a try and see if it works. 

```python
from micropython import const
import lcd_bus

_LCD_WIDTH = const(128)
_LCD_HEIGHT = const(64)

_LCD_SDA = const(0)
_LCD_SCL = const(0)
_LCD_ADDR = const(0x3C)
_LCD_HOST = const(1)
_LCD_FREQ = const(400000)
_LCD_DC_OFFSET = const(6)
_LCD_DC_LOW = const(1)
_LCD_BKL = const(-1)

data_bus = lcd_bus.I2CBus(sda=_LCD_SDA, scl=_LCD_SCL, addr=_LCD_ADDR, host=_LCD_HOST, 
                     dc_bit_offset=_LCD_DC_OFFSET, freq=_LCD_FREQ, 
                     dc_low_on_data=_LCD_DC_LOW)

buf = data_bus.allocate_framebuffer(int(_LCD_WIDTH * _LCD_HEIGHT / 8), lcd_bus.MEMORY_INTERNAL)

import ssd1306

display = ssd1306.SSD1306(data_bus=data_bus, display_width=_LCD_WIDTH, 
                          display_height=_LCD_HEIGHT, frame_buffer1=buf, 
                          backlight_pin=_LCD_BKL)

display.set_power(True)
display.init()
display.set_backlight(100.0)

import task_handler

task_handler.TaskHandler()

import lvgl as lv

label = lv.Label(lv.screen_active())
label.set_text('Hello World')
label.center()
```

tetopik, 2025-07-30:
i'm deleting the old one actually, then clonning the repo again.
Turns out it compiled successfully on generic esp32, but fails on c3 s2 and s3

tetopik, 2025-07-30:
I did make a mistake to put a wrong path to my added `USER_C_MODULE=`, my bad, i'm sorry.

Now I got this from the test code:
```
  File "ssd1306.py", line 39, in SSD1306
NameError: name 'LV' isn't defined
```

Manually change it to `lv` got me:
```
  File "ssd1306.py", line 96, in __init__
  File "display_driver_framework.py", line 168, in __init__
ValueError: invalid pin
```

tetopik, 2025-07-30:
- `_LCD_BKL = const(-1)` to `None` for `ValueError: invalid pin` from `display` init.

- Adding this line before `display.init()` for the following error:
```
display.width=_LCD_WIDTH
display.height=_LCD_HEIGHT
```
```
  File "<stdin>", line 31, in <module>
  File "display_driver_framework.py", line 460, in init
  File "_ssd1306_init.py", line 38, in init
AttributeError: 'SSD1306' object has no attribute 'height'
```

- And now the oled stays blank

kdschlosser, 2025-07-31:
i am not using the built in micropython frame buffer to handle the rendering stuff.

the -1 for the backlight pin should stay -1 if you do not have a backlight pin.

the LV thing is my bad..

i am not sure why you are doing this...

```python
display.width=_LCD_WIDTH
display.height=_LCD_HEIGHT
```

the width and height are already passed to the constructor of the display driver.

```python
display = ssd1306.SSD1306(data_bus=data_bus, display_width=_LCD_WIDTH, 
                          display_height=_LCD_HEIGHT, frame_buffer1=buf, 
                          backlight_pin=_LCD_BKL)
```

if you don't have a backlight pin you can remove it all together by using this for the display driver..

```python
display = ssd1306.SSD1306(data_bus=data_bus, display_width=_LCD_WIDTH, 
                          display_height=_LCD_HEIGHT, frame_buffer1=buf)
```

everything is taken care of for you with respects to the display getting updated. all you need to do is modify things for the different UI elements. create widgets, position them. change their color. The rendering all happens without you needing to do anything.

tetopik, 2025-07-31:
> the width and height are already passed to the constructor of the display driver.

I got following error without that line of code:
```
  File "<stdin>", line 31, in <module>
  File "display_driver_framework.py", line 460, in init
  File "_ssd1306_init.py", line 38, in init
AttributeError: 'SSD1306' object has no attribute 'height'
```
It says that `'SSD1306' object has no attribute 'height'` and also `'SSD1306' object has no attribute 'width'`.

What am I suppose to do then?

kdschlosser, 2025-07-31:
I just updated the driver again. give it a try and see if it works now.

tetopik, 2025-08-01:
The previous error persist:
```
Traceback (most recent call last):
  File "<stdin>", line 33, in <module>
  File "display_driver_framework.py", line 460, in init
  File "_ssd1306_init.py", line 46, in init
AttributeError: 'SSD1306' object has no attribute 'width'
```

This is my current setup:
```py
from micropython import const

_LCD_WIDTH  = const(128)
_LCD_HEIGHT = const(32)
_LCD_SDA  = const(33)
_LCD_SCL  = const(18)
_LCD_ADDR = const(0x3C)
_LCD_HOST = const(1)
_LCD_FREQ = const(400_000)
_LCD_DC_OFFSET = const(6)
_LCD_DC_LOW    = const(1)

from machine import Pin
Pin(16, Pin.OUT, value=1) # turn on the oled using custom pin

from time import sleep
sleep(1) # wait the oled to stabilized

from machine import I2C
i2c_bus = I2C(_LCD_HOST, scl=Pin(_LCD_SCL), sda=Pin(_LCD_SDA), freq=_LCD_FREQ)

from lcd_bus import I2CBus
data_bus = I2CBus(i2c_bus=i2c_bus, addr=_LCD_ADDR, dc_bit_offset=_LCD_DC_OFFSET, dc_low_on_data=_LCD_DC_LOW)

from lcd_bus import MEMORY_INTERNAL
buf = data_bus.allocate_framebuffer(int(_LCD_WIDTH * _LCD_HEIGHT / 8), MEMORY_INTERNAL)

from ssd1306 import SSD1306
display = SSD1306(data_bus=data_bus, display_width=_LCD_WIDTH, display_height=_LCD_HEIGHT, frame_buffer1=buf)

display.width = _LCD_WIDTH
display.set_power(True)
display.init()
display.set_backlight(100.0)

from task_handler import TaskHandler
TaskHandler()

import lvgl as lv
label = lv.label(lv.screen_active())
label.set_text('Hello World')
label.center()
```

The oled is still doing nothing.

tetopik, 2025-08-02:
That's a lot man, I know almost nothing about anything that you're doing. So, it sounds scary for me. LOL

Let me contribute later by testing the final code from the perspective of end user. Just update this thread. Great job, Kevin.

kdschlosser, 2025-08-02:
It's involved not because of the writing of the code so much as it is reading through the MicroPython source code to figure out how to collect the information that I need or can use so I am able to tweak how the I2C works in MicroPython. This way the classes stay in the same modules instead of having another module to handle just the I2C and another module to handle the SPI. I would rather override the current I2C and SPI to get them to play nice nice with the LCD screens. I have to make some changes to the SPI and then I can compile and iron out any compile bugs and then it's onto testing it out to see if it works properly.

anotherdoesnm, 2025-11-20:
i found ssd1306 driver here, may be useful?
https://github.com/rsxtrix/heltec_lora_32_v3-micropython_examples/blob/main/OLED%20Test/ssd1306.py