lvgl-micropython/lvgl_micropython #428: mbedtls compilation failure on MacOS build

thatbudakguy, 2025-08-13:
During a macOS build, compilation halts due to a failure when compiling mbedtls.

```shell
CC ../../lib/mbedtls/library/ssl_msg.c
CC ../../lib/mbedtls/library/ssl_ticket.c
CC ../../lib/mbedtls/library/ssl_tls.c
CC ../../lib/mbedtls/library/ssl_tls12_client.c
CC ../../lib/mbedtls/library/ssl_tls12_server.c
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
CC ../../lib/mbedtls/library/timing.c
../../lib/mbedtls/library/ssl_msg.c:4971:28: error: no member named 'badmac_seen' in 'struct mbedtls_ssl_context'
4971 |                     ++ssl->badmac_seen >= ssl->conf->badmac_limit) {
|                       ~~~  ^
1 error generated.
```

Invoked using `python make.py macOS` after a fresh clone, using an Apple M3 Pro (arm64) running macOS 15.6.

kdschlosser, 2025-08-13:
you can disable mbedtls by editing `lib/micropython/ports/unix/mpconfigport.mk` and changing `MICROPY_SSL_MBEDTLS = 1`  to `MICROPY_SSL_MBEDTLS= 0` and then compiling it again.

kdschlosser, 2025-08-13:
also you are going to need to add a couple more things to your build command..

```
python3 make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer
```

kdschlosser, 2025-08-13:
you can install SSL and use that in place of mbedtls

stevenmcastano, 2025-08-13:
I just ran into the exact same issue with mbedtls and even after disabling as suggested by @kdschlosser above and running `python3 make.py clean DISPLAY=sdl-display INDEV=sdl_pointer` it's still failing. This time with a new error:

```
MPY task_handler.py
MPY sdl_pointer.py
MPY sdl_display.py
GEN build-standard/frozen_content.c
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
../../extmod/modcryptolib.c:84:5: error: unknown type name 'AES_CTX_IMPL'
84 |     AES_CTX_IMPL ctx;
|     ^
../../extmod/modcryptolib.c:258:5: error: call to undeclared function 'aes_initial_set_key_impl'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
258 |     aes_initial_set_key_impl(&o->ctx, keyinfo.buf, keyinfo.len, ivinfo.buf);
|     ^
../../extmod/modcryptolib.c:297:9: error: call to undeclared function 'aes_final_set_key_impl'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
297 |         aes_final_set_key_impl(&self->ctx, encrypt_mode);
|         ^
../../extmod/modcryptolib.c:312:17: error: call to undeclared function 'aes_process_ecb_impl'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
312 |                 aes_process_ecb_impl(&self->ctx, in, out, encrypt);
|                 ^
../../extmod/modcryptolib.c:318:13: error: call to undeclared function 'aes_process_cbc_impl'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
318 |             aes_process_cbc_impl(&self->ctx, in_bufinfo.buf, out_buf_ptr, in_bufinfo.len, encrypt);
|             ^
5 errors generated.
make: *** [build-standard/extmod/modcryptolib.o] Error 1
make: *** Waiting for unfinished jobs....
../../extmod/modhashlib.c:246:59: error: use of undeclared identifier 'hashlib_sha1_digest'; did you mean 'hashlib_sha256_digest'?
246 | static MP_DEFINE_CONST_FUN_OBJ_1(hashlib_sha1_digest_obj, hashlib_sha1_digest);
|                                                           ^~~~~~~~~~~~~~~~~~~
|                                                           hashlib_sha256_digest
../../py/obj.h:377:42: note: expanded from macro 'MP_DEFINE_CONST_FUN_OBJ_1'
377 |     {{&mp_type_fun_builtin_1}, .fun._1 = fun_name}
|                                          ^
../../extmod/modhashlib.c:139:17: note: 'hashlib_sha256_digest' declared here
139 | static mp_obj_t hashlib_sha256_digest(mp_obj_t self_in) {
|                 ^
../../extmod/modhashlib.c:258:15: error: use of undeclared identifier 'hashlib_sha1_make_new'; did you mean 'hashlib_sha256_make_new'?
258 |     make_new, hashlib_sha1_make_new,
|               ^~~~~~~~~~~~~~~~~~~~~
|               hashlib_sha256_make_new
../../py/obj.h:802:250: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE'
802 | #define MP_DEFINE_CONST_OBJ_TYPE(...) MP_DEFINE_CONST_OBJ_TYPE_EXPAND(MP_DEFINE_CONST_OBJ_TYPE_NARGS(__VA_ARGS__, _INV, 12, _INV, 11, _INV, 10, _INV, 9, _INV, 8, _INV, 7, _INV, 6, _INV, 5, _INV, 4, _INV, 3, _INV, 2, _INV, 1, _INV, 0)(mp_obj_type_t, __VA_ARGS__))
|                                                                                                                                                                                                                                                          ^
../../py/obj.h:763:245: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE_NARGS_2'
763 | #define MP_DEFINE_CONST_OBJ_TYPE_NARGS_2(_struct_type, _typename, _name, _flags, f1, v1, f2, v2) const _struct_type _typename = { .base = { &mp_type_type }, .flags = _flags, .name = _name, .slot_index_##f1 = 1, .slot_index_##f2 = 2, .slots = { v1, v2, } }
|                                                                                                                                                                                                                                                     ^
../../py/obj.h:789:44: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE_EXPAND'
789 | #define MP_DEFINE_CONST_OBJ_TYPE_EXPAND(x) x
|                                            ^
../../extmod/modhashlib.c:119:17: note: 'hashlib_sha256_make_new' declared here
119 | static mp_obj_t hashlib_sha256_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) {
|                 ^
../../extmod/modhashlib.c:340:58: error: use of undeclared identifier 'hashlib_md5_digest'; did you mean 'hashlib_md5_digest_obj'?
340 | static MP_DEFINE_CONST_FUN_OBJ_1(hashlib_md5_digest_obj, hashlib_md5_digest);
|                                                          ^~~~~~~~~~~~~~~~~~
|                                                          hashlib_md5_digest_obj
../../py/obj.h:377:42: note: expanded from macro 'MP_DEFINE_CONST_FUN_OBJ_1'
377 |     {{&mp_type_fun_builtin_1}, .fun._1 = fun_name}
|                                          ^
../../extmod/modhashlib.c:340:34: note: 'hashlib_md5_digest_obj' declared here
340 | static MP_DEFINE_CONST_FUN_OBJ_1(hashlib_md5_digest_obj, hashlib_md5_digest);
|                                  ^
../../extmod/modhashlib.c:340:58: error: initializing 'mp_fun_1_t' (aka 'void *(*)(void *)') with an expression of incompatible type 'const mp_obj_fun_builtin_fixed_t' (aka 'const struct _mp_obj_fun_builtin_fixed_t')
340 | static MP_DEFINE_CONST_FUN_OBJ_1(hashlib_md5_digest_obj, hashlib_md5_digest);
[... cut ...]               ^
../../py/obj.h:789:44: note: expanded from macro 'MP_DEFINE_CONST_OBJ_TYPE_EXPAND'
789 | #define MP_DEFINE_CONST_OBJ_TYPE_EXPAND(x) x
|                                            ^
../../extmod/modhashlib.c:119:17: note: 'hashlib_sha256_make_new' declared here
119 | static mp_obj_t hashlib_sha256_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) {
|                 ^
5 errors generated.
make: *** [build-standard/extmod/modhashlib.o] Error 1
```

kdschlosser, 2025-08-13:
the problem is not macOS. The problem is the ARM64 processor. There is something that mbedtls doesn't like about that processor and I am not sure what it is. I don't have a way of testing building on the ARM64 processor.

kdschlosser, 2025-08-13:
The actual problem is in mbedtls and they used an incorrect field name.

The problem line in mbedtls is this...

```c
++ssl->badmac_seen >= ssl->conf->badmac_limit) {
```

It should read

```c
++ssl->MBEDTLS_PRIVATE(badmac_seen) >= ssl->conf->badmac_limit) {
```

Xyzzzy, 2025-10-07:
> The actual problem is in mbedtls and they used an incorrect field name.
> 
> The problem line in mbedtls is this...
> 
> ++ssl->badmac_seen >= ssl->conf->badmac_limit) {
> It should read
> 
> ++ssl->MBEDTLS_PRIVATE(badmac_seen) >= ssl->conf->badmac_limit) {

Did not work for me:

`../../lib/mbedtls/library/ssl_msg.c:4971:44: error: no member named 'badmac_seen' in 'struct mbedtls_ssl_context'
4971 |                     ++ssl->MBEDTLS_PRIVATE(badmac_seen) >= ssl->conf->badmac_limit) {
|                       ~~~                  ^
/opt/homebrew/include/mbedtls/private_access.h:17:33: note: expanded from macro 'MBEDTLS_PRIVATE'
17 | #define MBEDTLS_PRIVATE(member) member
|                                 ^
1 error generated.`

kdschlosser, 2025-10-08:
what is causing this issue is a development version of mbedtls is installed on the system. That needs to be uninstalled in order to compile it. The problem lies upstream in mbedtls and they are not going to fix the problem. their fix is to remove the one that is installed.