lvgl-micropython/lvgl_micropython #429: Use of .toml files

hemonu, 2025-08-13:
I discovered two issues with the creation and use of .toml files: 

a) I first created for my ESP32S3 display a custom board directory and created all the files in line with your MY_CUSTOM_BOARD example. That compiled and worked as expected. After that I created a .toml file in same directory, including all the pin configuration etc. 
Compiling again with  
`python3 make.py esp32 --custom-board-path=../../esp32/ELECROW_DLC35010R DISPLAY=ili9488 INDEV=ft6x36`
ends with an error:

```
Compiling....
Getting ESP-IDF build Environment
Traceback (most recent call last):
  File "/home/hermoers/projects/esp32/lvgl_micropython/make.py", line 330, in <module>
    mod.compile(*extra_args)
  File "/home/hermoers/projects/esp32/lvgl_micropython/builder/esp32.py", line 1452, in compile
    cmd_.extend(args)
TypeError: 'NoneType' object is not iterable

```

I think the problem is in esp32.py, function build_sdkconfig(*args), line 1304, called from compile(*args), line 1429. As there is no return value, all the arguments which were created from the toml file get lost. 

b) I moved the .toml file into a separate directory and compile with 
`python3 make.py --toml=../../esp32/ELECROW_DLC35010R_TOML/elecrow_dlc35010r.toml
`
That does compile without an error, but the module touch (ft6x36) is not included. 

my .toml file is as follows: 

```
[MCU.esp32]
BOARD = "ESP32_GENERIC_S3"
BOARD_VARIANT = "SPIRAM_OCT"
flash_size = 16
enable_jtag_repl = 'n'
enable_cdc_repl = 'n'
enable_uart_repl = 'y'
uart_repl_bitrate = 115200

[I80Bus.display_bus]
data0 = 47
data1 = 21
data2 = 14
data3 = 13
data4 = 12
data5 = 11
data6 = 10
data7 = 9
data8 = 3
data9 = 8
data10 = 16
data11 = 15
data12 = 7
data13 = 6
data14 = 5
data15 = 4
dc = 45
wr = 18
cs = -1
freq = 40_000_000

[I2C.Bus.i2c_bus]
host = 0
scl = 39
sda = 38
freq = 100_000

[I2C.Device.indev_device]
bus = "i2c_bus"
dev_id = "ft6x36.I2C_ADDR"
reg_bits = "ft6x36.BITS"

[FT6x36.indev]
device = "indev_device"

[ILI9488.display]
data_bus = "display_bus"
display_width = 320
display_height = 480
backlight_pin = 46
color_byte_order = "ili9488.BYTE_ORDER_BGR"
color_space = "lv.COLOR_FORMAT.RGB565"

[ILI9488.display.init]
params = []

[display.set_color_inversion]
params = [false]

[display.set_rotation]
params = ["lv.DISPLAY_ROTATION._0"]

[display.set_backlight]
params = [100]

[task_handler.TaskHandler]
params=[]

```
I think the problem is in toml_reader.py lines 268/269. "ft6x36.I2C_ADDR" in my .toml fle is parsed there, the import statement ist written, 'ft6x36' is appened to the list used_imports. But it is not appended to the list self.imports, which ist scanned in line 500! If I change dev_id and reg_bits to numeric values, the module is included as expected.
In lines 222/223 and 243/244 potentially the same problem would show up. 

Thanks for your fantastic job. The TOML file makes life much easier. Maybe you can mention in the README, that importing the display module ist the trick to get access to the TOML entries, It took me quite a while to figure that out ;) 

Herbert

kdschlosser, 2025-08-13:
attach a complete copy of your custom board zipped up and add ".txt" to the end so you can upload it to github. It will be easier for me to see what is going on that way.

There are some things that get skipped as the result of using a custom board. This is because you have the ability to set those things elsewhere in the custom board config files so there is no reason to even check the command line for those things. building the sdkconfig file is one of those things that gets skipped. That's because you can create your own sdkconfig file in the custom board directory. 

The other thing is you don't pass the toml file when the toml file is apart of a custom board. it will check the directory for the toml file and if one is found then it will use it and if not then it won't. When using a custom board without a TOML file you would use all of the usual build commands and the custom board build option. If there is a toml file on the custom board then don't supply anything other than the custom board path. 

make sense?

kdschlosser, 2025-08-13:
your toml file is also incorrect..

```toml
[MCU.esp32]
BOARD = "ESP32_GENERIC_S3"
BOARD_VARIANT = "SPIRAM_OCT"
flash_size = 16
enable_jtag_repl = 'n'
enable_cdc_repl = 'n'
enable_uart_repl = 'y'
uart_repl_bitrate = 115200

[I80Bus.display_bus]
data0 = 47
data1 = 21
data2 = 14
data3 = 13
data4 = 12
data5 = 11
data6 = 10
data7 = 9
data8 = 3
data9 = 8
data10 = 16
data11 = 15
data12 = 7
data13 = 6
data14 = 5
data15 = 4
dc = 45
wr = 18
cs = -1
freq = 40000000

[I2C.Bus.i2c_bus]
host = 0
scl = 39
sda = 38
freq = 100000

[I2C.Device.indev_device]
bus = "i2c_bus"
dev_id = "ft6x36.I2C_ADDR"
reg_bits = "ft6x36.BITS"

[ILI9488.display]
data_bus = "display_bus"
display_width = 320
display_height = 480
backlight_pin = 46
color_byte_order = "ili9488.BYTE_ORDER_BGR"
color_space = "lv.COLOR_FORMAT.RGB565"

[disdplay.set_power]
params = [true]

[display.init]
params = []

[FT6x36.indev]
device = "indev_device"

[display.set_color_inversion]
params = [false]

[display.set_backlight]
params = [100]

[task_handler.TaskHandler]
params=[]
```

hemonu, 2025-08-14:
> your toml file is also incorrect..
> 
> [MCU.esp32]
> BOARD = "ESP32_GENERIC_S3"
> BOARD_VARIANT = "SPIRAM_OCT"
> flash_size = 16
> enable_jtag_repl = 'n'
> enable_cdc_repl = 'n'
> enable_uart_repl = 'y'
> uart_repl_bitrate = 115200
> 
> [I80Bus.display_bus]
> data0 = 47
> data1 = 21
> data2 = 14
> data3 = 13
> data4 = 12
> data5 = 11
> data6 = 10
> data7 = 9
> data8 = 3
> data9 = 8
> data10 = 16
> data11 = 15
> data12 = 7
> data13 = 6
> data14 = 5
> data15 = 4
> dc = 45
> wr = 18
> cs = -1
> freq = 40000000
> 
> [I2C.Bus.i2c_bus]
> host = 0
> scl = 39
> sda = 38
> freq = 100000
> 
> [I2C.Device.indev_device]
> bus = "i2c_bus"
> dev_id = "ft6x36.I2C_ADDR"
> reg_bits = "ft6x36.BITS"
> 
> [ILI9488.display]
> data_bus = "display_bus"
> display_width = 320
> display_height = 480
> backlight_pin = 46
> color_byte_order = "ili9488.BYTE_ORDER_BGR"
> color_space = "lv.COLOR_FORMAT.RGB565"
> 
> [disdplay.set_power]
> params = [true]
> 
> [display.init]
> params = []
> 
> [FT6x36.indev]
> device = "indev_device"
> 
> [display.set_color_inversion]
> params = [false]
> 
> [display.set_backlight]
> params = [100]
> 
> [task_handler.TaskHandler]
> params=[]

Doesn't make a difference in the behaviour ...