lvgl-micropython/lvgl_micropython #440: Show CPU usage and FPS count

Vincent1-python, 2025-08-27:
I # defined LV_USE_PERF_MONITOR 1 in the project.
In my project, both fps and CPU are displayed, but when I click a button or switch tabs, FPS changes. But it's always 0%CPU.

Vincent1-python, 2025-08-28:
Now the fps can reach an astonishing 100fps, but the cpu display is still 0%. @kdschlosser

kdschlosser, 2025-08-28:
You cannot use the performance monitor that is built into LVGL. It is flawed. It is impossible for it to calculate CPU use because there are more than a single task that is running on a single core. I can put together a function that you can call from MicroPython that will give you a correct CPU use if you want...

The FPS calculation that LVGL uses is flawed. It doesn't give an accurate count because of DMA memory use and also because partial buffers do not represent the update of a "frame". It could take multiple partial buffers to update a single frame. 

The 100FPS you are seeing is more than likely being caused by not taking into account the time it takes to send the last buffer. the actual call made to the driver takes nanoseconds to execute when using DMA memory due to the non blocking call. There is also the fact that LVGL only has millisecond resolution so to time anything faster than 1 millisecond in LVGL is impossible to do. 

Another flaw which is a huge one is when the rendering process starts but then it turns out that there is nothing to render. LVGL counts this as a frame update when it shouldn't because nothing has been updated. The FPS should hold the last count if there is a run of updates like this that occur.

The last flaw is the way the calculations are done. It is done using LVGL timers which are only as accurate as when lv.tick_inc is called and what value is passed to it but also the interval in which lv.task_handler is called. LVGL doesn't use a rolling type of time stamp that holds the timestamp values associated with a specific frame update continually appending to it and when the last timestamp - the first timestamp is >= 1 second then append the next stamp and drop the first one so you have a rolling calculation.

we cannot start the timing by getting a timestamp just before calling lv.task_handler because that function does a heap load of other tasks like reading the indev and updating widgets for animations. and those things you do not want mucking up the calculations.

I would need to look at the code in both LVGL and in this binding to determine the best way to go about adding the ability to do a proper FPS calculation. LVGL did add the ability to check to see if the buffer that is being flushed is actually the last one of a series of updates. when that function returns `True` that would constitute a single frame as being updated. 

let me do some looking to see what the best way is going to be to get the most accurate count.

Vincent1-python, 2025-08-28:
But I seem to find that your file for generating lvgl module seems to ignore lv_conf.h, which makes me unable to change something to make it effective.

kdschlosser, 2025-08-28:
it doesn't ignore it. you are probably modifying the wrong one. The one you want to modify is `lib/lv_conf.h`

Vincent1-python, 2025-08-28:
For example, I recently transplanted C5, but the space is not enough, so setting all the components of the picture class to 0 still generates an lvgl with gif and picture functions, which leads to an error.