lvgl-micropython/lvgl_micropython #445: Neopixels do not work correctly

ste7anste7an, 2025-08-30:
Using the lvgl_micropython firmware build (idf 5.4) for ESP32 wth SPIRAM, NeoPixels do not work properly. I see that some pixels are not lighting up, and thers are the wrong color. It seems that the timing (RMT) is slightly off. 

Comparing to the official micropython 1.25 build (idf 5.2.2) When I compile the lvgl_micropython.

Would it be possible to use idf 5.2 for building lvgl_micropythin, to see wther that solves the problem?

```
from machine import Pin
from neopixel import NeoPixel
np=NeoPixel(Pin(19),9)  # 9 pixels
np.fill((100,100,0))
np.write()
```

kdschlosser, 2025-08-31:
OK I have been looking at the driver and it looks to be pretty easy to write up a new driver that would be non blocking when transferring data so it would be faster. You can allocate multiple frame buffers to handle this. The frame buffers would be num_pixels * 3 in size. I would write the driver so you can allocate as many frame buffers as you like and that would be the maximum number of buffers you could queue up to be transmitted. There would be a callback that gets called so you can track which buffers are being transmitted and which ones can be written to. I would not have it erase the pixel data in the buffer after a transmit was successful

kdschlosser, 2025-08-31:
The neopixel library is set up for the ws2812 IC's so if you are not using a strip that uses those specific IC's then some additional things need to be done to get it to work properly.

kdschlosser, 2025-08-31:
I am getting close to being finished writing a new driver. I still have to write the python side of the driver but that is not going to be too difficult to do.

kdschlosser, 2025-08-31:
OK so the driver isn't defaulted to the timings for the WS2812B  IC. That is a different set of timings.

The timings for that ws2812 are 
```python
(350, 800, 700, 600)
```

and the timings for you IC are

```python
(400, 850, 800, 450)
```

you see how close they are?? which could explain why the PSRAM access slows things down a bit which mucks up how the timings are. 

Try setting the timings to the above and see if it works better.

kdschlosser, 2025-08-31:
You can give it a try now. I updated the driver in MicroPython so it should hopefully work now. I did not try to compile it so the build may fail. If it does paste the exact error(s) so I can fix them.

ste7anste7an, 2025-08-31:
You did some impressive revisions in the esp32_bitstream.c file. It seems that the decorator ` RMT_ENCODER_FUNC_ATTR` is not recognised.

<details>
<summary> Error, click to expand </summary>

```c

/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:122:22: error: expected ';' before 'static'
122 | RMT_ENCODER_FUNC_ATTR
|                      ^
|                      ;
123 | static size_t rmt_encode_led_strip(rmt_encoder_t *encoder, rmt_channel_handle_t channel, const void *primary_data, size_t data_size, rmt_encode_state_t *ret_state)
| ~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:170:22: error: expected ';' before 'static'
170 | RMT_ENCODER_FUNC_ATTR
|                      ^
|                      ;
171 | static esp_err_t rmt_led_strip_encoder_reset(rmt_encoder_t *encoder)
| ~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c: In function 'rmt_new_led_strip_encoder':
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:185:5: error: implicit declaration of function 'ESP_GOTO_ON_FALSE' [-Wimplicit-function-declaration]
185 |     ESP_GOTO_ON_FALSE(config && ret_encoder, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
|     ^~~~~~~~~~~~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:185:67: error: 'err' undeclared (first use in this function)
185 |     ESP_GOTO_ON_FALSE(config && ret_encoder, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
|                                                                   ^~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:185:67: note: each undeclared identifier is reported only once for each function it appears in
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:185:72: error: 'TAG' undeclared (first use in this function)
185 |     ESP_GOTO_ON_FALSE(config && ret_encoder, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
|                                                                        ^~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:207:37: error: 'bit1_duration1' undeclared (first use in this function); did you mean 'bit2_duration1'?
207 |     if (config->bit1_duration1 < 0) bit1_duration1 = (uint32_t)-config->bit1_duration1;
|                                     ^~~~~~~~~~~~~~
|                                     bit2_duration1
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:227:5: error: implicit declaration of function 'ESP_GOTO_ON_ERROR' [-Wimplicit-function-declaration]
227 |     ESP_GOTO_ON_ERROR(rmt_new_bytes_encoder(&bytes_encoder_config, &led_encoder->bytes_encoder), err, TAG, "create bytes encoder failed");
|     ^~~~~~~~~~~~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:259:13: error: invalid storage class for function 'machine_bitstream_high_low_rmt'
259 | static void machine_bitstream_high_low_rmt(mp_hal_pin_obj_t pin, uint32_t *timing_ns, const uint8_t *buf, size_t len, uint8_t channel_id) {
|             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c: In function 'machine_bitstream_high_low_rmt':
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:289:86: error: expected ';' before ')' token
289 |     check_esp_err(rmt_transmit(channel_handle, encoder_handle, buf, len, &tx_config)));
|                                                                                      ^
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:289:86: error: expected statement before ')' token
In file included from /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/py/mphal.h:35,
from /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:28:
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:293:46: error: 'config' is a pointer; did you mean to use '->'?
293 |     check_esp_err(rmt_tx_wait_all_done(config.channel, pdMS_TO_TICKS(timeout_ms)));
|                                              ^
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/mphalport.h:64:44: note: in definition of macro 'check_esp_err'
64 | #define check_esp_err(code) check_esp_err_(code)
|                                            ^~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c: In function 'rmt_new_led_strip_encoder':
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:312:1: error: expected declaration or statement at end of input
312 | }
| ^
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:245:1: error: label 'err' defined but not used [-Werror=unused-label]
245 | err:
| ^~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:196:14: warning: unused variable 'bit2_duration1' [-Wunused-variable]
196 |     uint32_t bit2_duration1;
|              ^~~~~~~~~~~~~~
At top level:
[... cut ...]
make failed with exit code 2, output of the command is in the /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/log/idf_py_stderr_output_3970665 and /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/log/idf_py_stdout_output_3970665
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:77: all] Error 1

cp: cannot stat 'build/lvgl_micropy_ESP32_GENERIC-SPIRAM-4.bin': No such file or directory
```

</detail>

kdschlosser, 2025-08-31:
OK I removed that from the code. give it a try again.

ste7anste7an, 2025-09-01:
Looks like a similar error.

<detail>

<summary> Errors are here</summary>

```
[ 58%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_dac.c.obj
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:169:22: error: expected ';' before 'static'
169 | RMT_ENCODER_FUNC_ATTR
|                      ^
|                      ;
170 | static esp_err_t rmt_led_strip_encoder_reset(rmt_encoder_t *encoder)
| ~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c: In function 'rmt_new_led_strip_encoder':
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:184:5: error: implicit declaration of function 'ESP_GOTO_ON_FALSE' [-Wimplicit-function-declaration]
184 |     ESP_GOTO_ON_FALSE(config && ret_encoder, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
|     ^~~~~~~~~~~~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:184:67: error: 'err' undeclared (first use in this function)
184 |     ESP_GOTO_ON_FALSE(config && ret_encoder, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
|                                                                   ^~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:184:67: note: each undeclared identifier is reported only once for each function it appears in
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:184:72: error: 'TAG' undeclared (first use in this function)
184 |     ESP_GOTO_ON_FALSE(config && ret_encoder, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
|                                                                        ^~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:206:37: error: 'bit1_duration1' undeclared (first use in this function); did you mean 'bit2_duration1'?
206 |     if (config->bit1_duration1 < 0) bit1_duration1 = (uint32_t)-config->bit1_duration1;
|                                     ^~~~~~~~~~~~~~
|                                     bit2_duration1
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:226:5: error: implicit declaration of function 'ESP_GOTO_ON_ERROR' [-Wimplicit-function-declaration]
226 |     ESP_GOTO_ON_ERROR(rmt_new_bytes_encoder(&bytes_encoder_config, &led_encoder->bytes_encoder), err, TAG, "create bytes encoder failed");
|     ^~~~~~~~~~~~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:258:13: error: invalid storage class for function 'machine_bitstream_high_low_rmt'
258 | static void machine_bitstream_high_low_rmt(mp_hal_pin_obj_t pin, uint32_t *timing_ns, const uint8_t *buf, size_t len, uint8_t channel_id) {
|             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c: In function 'machine_bitstream_high_low_rmt':
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:288:86: error: expected ';' before ')' token
288 |     check_esp_err(rmt_transmit(channel_handle, encoder_handle, buf, len, &tx_config)));
|                                                                                      ^
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:288:86: error: expected statement before ')' token
In file included from /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/py/mphal.h:35,
from /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:28:
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:292:46: error: 'config' is a pointer; did you mean to use '->'?
292 |     check_esp_err(rmt_tx_wait_all_done(config.channel, pdMS_TO_TICKS(timeout_ms)));
|                                              ^
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/mphalport.h:64:44: note: in definition of macro 'check_esp_err'
64 | #define check_esp_err(code) check_esp_err_(code)
|                                            ^~~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c: In function 'rmt_new_led_strip_encoder':
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:311:1: error: expected declaration or statement at end of input
311 | }
| ^
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:244:1: error: label 'err' defined but not used [-Werror=unused-label]
244 | err:
| ^~~
/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/machine_bitstream.c:195:14: warning: unused variable 'bit2_duration1' [-Wunused-variable]
195 |     uint32_t bit2_duration1;
|              ^~~~~~~~~~~~~~
At top level:
cc1: note: unrecognized command-line option '-Wno-unterminated-string-initialization' may have been intended to silence earlier diagnostics
cc1: some warnings being treated as errors
make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:3896: esp-idf/main/CMakeFiles/__idf_main.dir/__/machine_bitstream.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM'
[... cut ...]
make failed with exit code 2, output of the command is in the /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/log/idf_py_stderr_output_4106064 and /home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/log/idf_py_stdout_output_4106064
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/stefan/esp32/lvgl_mp/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:77: all] Error 1

```

</detail>

kdschlosser, 2025-09-03:
I added the extension module to the repo, I still have yet to fully add it so it will compile and to run the compilation tests on it.