lvgl-micropython/lvgl_micropython #446: CONFLICT! driver_ng is not allowed to be used with the legacy driver

aliasnameless, 2025-09-02:
I compiled and flashed the firmware without problems. However, I get a bootloop
```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4112
load:0x40078000,len:15052
load:0x40080400,len:4
load:0x40080404,len:3332
entry 0x400805ac
E (973) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: [github.com/micropython/micropython/wiki/ESP32-debugging](https://github.com/micropython/micropython/wiki/ESP32-debugging)

LVGL MicroPython
IDF version : 67c1de1e
Machine     : Generic ESP32 module with ESP32

abort() was called at PC 0x40228cb0 on core 0

Backtrace: 0x400813dd:0x3ffe3b40 0x40095de5:0x3ffe3b60 0x4009b32e:0x3ffe3b80 0x40228cb0:0x3ffe3bf0 0x4018ca46:0x3ffe3c10 0x400817e2:0x3ffe3c40 0x40079666:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 603a97685
```

My compilation options is: 
`python3 make.py esp32 clean --flash-size=4 BOARD=ESP32_GENERIC DISPLAY=ili9341 INDEV=xpt2046`

My board is: sunton esp32 2432s024c aka CYD

kdschlosser, 2025-09-02:
I am aware of the problem and I am working on fixing it. You can compile the code form several commits ago and it should compile fine.

kdschlosser, 2025-09-05:
soft reboots don't work with this project regardless of the error you are getting. It usually causes a core panic and it has to do something with the root pointers to the lvgl objects.

so if you didn't get this error you more than likely would have ended up with a core panic anyhow.

howardtechnical, 2025-09-06:
@kdschlosser Does that mean we can never trigger a reset from a Micropython application?  My project requires the ability to reset the device.  I'm currently using machine.reset().

kdschlosser, 2025-09-06:
That's a hard reset so you are fine. you don't want to use soft_reset()

closed 2025-11-12 by kdschlosser