lvgl-micropython/lvgl_micropython #456: JC3248W535 - board not recognized after flashing firmware

de-dh, 2025-09-17:
**Describe the bug**

I tried to build the firmware for the JC3248W535 today following [@straga instructions](https://github.com/straga/micropython_lcd/blob/master/device/JC3248W535/README.md). I used WSL for building.
The building and flashing was successfull. However, the board is not recognized correctly after flashing the firmware and I can't access it (see first image).

**Images**
Board not recognized correctly (the board can not be accessed by clicking on an entry, it produces an error):

[image]

Board is correctly recognized after flashing an older firmware build (02.06.2025).
[image]

**Build Command**

```shell
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16
```

**Flash**

```shell
python -m esptool --chip esp32s3 --port COM7 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m --erase-all 0x0 lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-16.bin
```

Any ideas?

kdschlosser, 2025-09-17:
Does your board have 2 USB ports?

If it does have you tried plugging into the other USB port?

If it doesn't and only have one USB port is that USB port is connected to a UART to USB bridge IC then you need to add a couple of extra parameters to your build command.

```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --enable-cdc-repl=n --enable-uart-rep=y --enable-jtag-repl=n --flash-size=16
```

de-dh, 2025-09-18:
Thank you, I will try this. What I don't understand is why the board is recognized properly when it's flashed with earlier builds of the firmware. I've also tested another JC3248W535 board. It's also not recognized properly when flashed with the current firmware build.

The board is shown in the device manager and I flash it using windows cmd + esptool. I never had problems with this setup until now. Same goes for Thonny. I know that most people don't like it, but I'm not that skilled with software development and I like it's simple usage.
I'll try to flash from WSL and use putty.

kdschlosser, 2025-09-18:
earlier versions might also be using an earlier version of micropython. Not too long ago there were a lot of changes that were made to how micropython handles the usb connection.

de-dh, 2025-09-20:
I think thats the problem. I'm trying to use older versions of the binding. Thanks anyway.

ctan10, 2025-09-24:
You are welcome. Glad it work.